MichaIng/DietPi #1287: v159 Release

Fourdee, 2017-12-10:
### ToDo:
- 🈯️ Switch to RPi Stretch image on dietpi.com https://github.com/Fourdee/DietPi/issues/475#issuecomment-350519670
- Create/upload Allo GUI RPi Stretch image, to replace Jessie version
-- Test updates of existing installations
-- update forum links
- Merge
- Twitter post for update

Fourdee, 2017-12-10:
Release will now be done Monday evening, ran out of time today.

k-plan, 2017-12-10:
> - Xmas Tree? 🎄

Yes, please .... :laughing:

:speak_no_evil:

Fourdee, 2017-12-10:
@k-plan 
[image]

k-plan, 2017-12-10:
@Fourdee 

Very nice! 👍 

Good to see, you don't lose one's your humour! 😄

Fourdee, 2017-12-10:
@k-plan 

> Good to see, you don't lose one's your humour!

Trying, although its really salty recently lol

Fourdee, 2017-12-10:
Completed.

Thanks @k-plan and @MichaIng for all your contributions this release, really appreciate it 👍

closed 2017-12-10 by Fourdee