MichaIng/DietPi #1288: DietPi-Services | stop/start dietpi-wifi-monitor drops connection

Fourdee, 2017-12-10:


Master92, 2017-12-11:
Had the same issue when running software installation with `dietpi-software`.

Last output is:
```
 DietPi-Services
─────────────────────────────────────────────────────
 Mode: stop
 Please wait...

 [Ok] cron
 [Ok] nginx
 [Ok] octoprint
dietpi-wifi-monitor
```
After that, the connection drops and won't automatically reestablish.

#### Required information:
- DietPi Version: latest (can't find out right now, due to offline device)
- SBC Device: Pine A64

Yury-MonZon, 2017-12-12:
It is weird, but for dietpi-update it didn't work.
Info here: http://dietpi.com/phpbb/viewtopic.php?f=11&t=2547&p=9918#p9918

Master92, 2017-12-13:
Did you update your system after commenting out ```dietpi-wifi-manager``` and before updating again?
Seems like the ```/DietPi/dietpi/dietpi-services``` file gets replaced by each version-upgrade.

Yury-MonZon, 2017-12-13:
Nope. I was on v158, commented it out and tried to update to v159. After it disconnected I had to power cycle it to reboot. Then checked that it is still commented out and tried to run the update again with the same results.

Fourdee, 2017-12-13:
@Master92 

> Seems like the /DietPi/dietpi/dietpi-services file gets replaced by each version-upgrade.

Correct, all DietPi scripts are replaced during the upgrade.

> monzon@MZ-Cloud:~ $ cat /DietPi/dietpi/dietpi-services | grep wifi
#'dietpi-wifi-monitor'

@Yury-MonZon 

Bizarre,

Does the same occur when you run:
```
/DietPi/dietpi/dietpi-services stop
```

Yury-MonZon, 2017-12-13:
No, it doesn't disconnect this way.
`monzon@MZ-Cloud:~ $ sudo /DietPi/dietpi/dietpi-services stop
[sudo] password for monzon:

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: stop
 Please wait...

 [Ok] cron
 [Ok] proftpd
 [Ok] nginx
 [Ok] php5-fpm
 [Ok] mysql
 [Ok] fail2ban
 [Ok] avahi-daemon
 [Ok] rpimonitor

`

Yury-MonZon, 2017-12-13:
It does disconnect in the middle of installation,
not at the beginning.
[code]
monzon@MZ-Cloud:~ $ sudo dietpi-update 1
 [Info] Checking RootFS R/W access. Please wait...
 [Ok] RootFS R/W access.

 DietPi-Update
─────────────────────────────────────────────────────
 Mode: Checking for DietPi updates
 Please wait...

 [Info] Checking Mirror : http://dietpi.com/downloads/dietpi-update_mirror/master/server_version
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100     6  100     6    0     0     26      0 --:--:-- --:--:-- --:--:--    26
 [Ok] Using update server: http://dietpi.com/downloads/dietpi-update_mirror/master/server_version

 [Ok] Update available
 [Info] Current Version : 158
 [Info] Server Version  : 159
 [Info] Detecting drives, please wait...
 [Info] Detected /dev mount: /dev/mmcblk0p1 > /boot
 [Info] Detected UUID mount: /dev/mmcblk0p2 > /mnt/6e11ba37-3d23-4d99-8ad6-b6a7a5fb372d
 [Info] Detected UUID mount: /dev/sda1 > /
 [Info] Processing drive information, please wait...
 [Info] Checking available free space on RootFS, please wait...
 [Ok] 402975 MB available, 500 MB required
 [Ok] Updates have been found and are being applied, please wait...
 [Info] Current Version : 158
 [Info] Server Version  : 159
 #               processing | D i e t  P i             #

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: stop
 Please wait...

 [Ok] cron
 [Ok] proftpd
 [Ok] nginx
 [Ok] php5-fpm
 [Ok] mysql
 [Ok] fail2ban
 [Ok] avahi-daemon
 [Ok] rpimonitor

  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  384k  100  384k    0     0   700k      0 --:--:-- --:--:-- --:--:--  699k
Archive:  /tmp/dietpi-update/update.zip
5cc4e59a128ff3998802e4a80fa5751d11f7a88f
   creating: /tmp/dietpi-update/DietPi-master/
  inflating: /tmp/dietpi-update/DietPi-master/.gitattributes
   creating: /tmp/dietpi-update/DietPi-master/.github/
  inflating: /tmp/dietpi-update/DietPi-master/.github/ISSUE_TEMPLATE.md
  inflating: /tmp/dietpi-update/DietPi-master/.github/PULL_REQUEST_TEMPLATE.md
  inflating: /tmp/dietpi-update/DietPi-master/.gitignore
  inflating: /tmp/dietpi-update/DietPi-master/CHANGELOG.txt
  inflating: /tmp/dietpi-update/DietPi-master/PREP_SYSTEM_FOR_DIETPI.sh
  inflating: /tmp/dietpi-update/DietPi-master/README.md
  inflating: /tmp/dietpi-update/DietPi-master/TESTING-BRANCH.md
  inflating: /tmp/dietpi-update/DietPi-master/boot_c1.ini
  inflating: /tmp/dietpi-update/DietPi-master/boot_c2.ini
  inflating: /tmp/dietpi-update/DietPi-master/boot_xu4.ini
  inflating: /tmp/dietpi-update/DietPi-master/config.txt
  inflating: /tmp/dietpi-update/DietPi-master/dietpi.txt
   creating: /tmp/dietpi-update/DietPi-master/dietpi/
 extracting: /tmp/dietpi-update/DietPi-master/dietpi/.dietpi-autostart_index
 extracting: /tmp/dietpi-update/DietPi-master/dietpi/.install_stage
[... 133 lines cut ...]
 [Info] Detected UUID mount: /dev/sda1 > /
 [Info] Processing drive information, please wait...
 [Info] Checking available free space on RootFS, please wait...
 [Ok] 402975 MB available, 500 MB required
 [Info] Testing connection to http://mirror.ox.ac.uk/sites/archive.raspbian.org/archive/raspbian
 [Info] Max duration of 20 seconds, please wait...
 [Ok] Connection test | Completed

 [Ok] NTPD: time sync | Completed

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: stop
 Please wait...

 [Ok] stop : cron
 [Ok] stop : proftpd
 [Ok] stop : nginx
 [Ok] stop : php5-fpm
 [Ok] stop : mysql
 [Ok] stop : fail2ban
 [Ok] stop : avahi-daemon
 [Ok] stop : rpimonitor
dietpi-wifi-monitor
[/code]

Fourdee, 2017-12-14:
@Yury-MonZon 

> DietPi-Services
─────────────────────────────────────────────────────
Mode: stop
Please wait...

[Ok] stop : cron
[Ok] stop : proftpd
[Ok] stop : nginx
[Ok] stop : php5-fpm
[Ok] stop : mysql
[Ok] stop : fail2ban
[Ok] stop : avahi-daemon
[Ok] stop : rpimonitor
dietpi-wifi-monitor

Ah yes, thats because the updated script is now running, which replaces your dietpi-wifi-monitor changes.

I've disabled control of ```dietpi-wifi-monitor ``` for v160 onwards:
- https://github.com/Fourdee/DietPi/commit/793a923f266c4b1c77dd0d274ff4206f2836c59a

**You'll need to wait for v160 release, then**:
- Disable ```dietpi-wifi-monitor ``` as previously
- Update
- New script will have ```dietpi-wifi-monitor ``` disabled by default.

Completed, albeit waiting for v160 release.

closed 2017-12-14 by Fourdee