MichaIng/DietPi #1290: Plexmediaserver installed but launches only as Plex Web

Dinesh3219, 2017-12-11:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```?
- SBC Device (EG: RPi 3)?
- Power supply used (EG: 5v 1A RAVpower)?
- SD card used (EG: Sandisk ultra)?
- Distro (EG: Jessie) | ```uname -a```?

#### Additional Information (if applicable):
- Software title?
- Can this issue be replicated on a fresh installation of DietPi?

#### Expected behaviour:

#### Actual behaviour:

#### Steps to reproduce:

#### Did you submit a ```dietpi-bugreport```?

#### Extra details:

Fourdee, 2017-12-16:
@Dinesh3219 

Hi,

You need to add ```/web``` after the port, eg:
http://192.168.0.100:32400/web

> http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=1949#p1949

Dinesh3219, 2017-12-16:
Hi Fudree,

Thank you for replying.. yes I have added web at the end.. I login with ipaddress:32400/web and it takes me through the plex log in page and everything seems normal, but the server section is missing under the users and I cannot link any of my folders to PLEXSERVER. I have my windows computer on the same network and when I go under the networks tab in My Computer then I can see plexmediaserver - dietpi. So I think the installation of plexmediaserver from dietpi-software is fine but I still cannot see the server section under users tab. I have re-installed my plexmediaserver few times but I keep hitting into the same wall.
If there is anything else that I should do then please help me out.

Thanks a lot.

Fourdee, 2018-02-22:
Outdated, marking as closed.

closed 2018-02-22 by Fourdee