MichaIng/DietPi #1292: Network Problems by silent installation

tatoosh, 2017-12-11:
Test on Raspberry B+ and Raspberry Pi Zero W

Image: v158 - DietPi_v158_RPi-armv6-(Stretch)
Problem:
Pi Zero W: my WLAN isn't working so i tried with PiB+ with plugged ethernet adapter.
Ethernet is also not working. I have to enable the Ethernet Adapter in dietpi-config.

k-plan, 2017-12-11:
@tatoosh 

> Raspberry B+ and Raspberry Pi Zero W
> Image: v158 - DietPi_v158_RPi-armv6-(Stretch)
> PiB+ with plugged ethernet adapter
> I have to enable the Ethernet Adapter in dietpi-config.

Ethernet is enable by default in `dietpi.txt`, Nothing is to do.

```
# >> Networking Options -----------------------------
#If both Ethernet and Wifi are enabled, Wifi will take priority and Ethernet will be disabled.
#   1=enabled
Ethernet_Enabled=1
Wifi_Enabled=0
```

Extra testing now:

### RPi Zero W (armv6l)

- [ ] Automated installation - Wifi only - DHCP  - (works)

- [ ] Manual Installation - Wifi only - static IP address - (works)

- [ ] Automated installation - Ethernet USB Adapter - DHCP  - (works)

- [ ] Automated installation - Ethernet USB Adapter - static IP address  - (works)

```
───────────────────────────────────────
 DietPi     | 16:08 | Mon 11/12/17
 ───────────────────────────────────────
 V159       | RPi Zero W (armv6l)
 ───────────────────────────────────────
 IP Address | 192.16.1.100
 ───────────────────────────────────────

 Created by : Daniel Knight
 Web        : http://DietPi.com
 Twitter    : http://twitter.com/dietpi_
 Donate     : http://goo.gl/pzISt9
 DietPi's web hosting is powered by: MyVirtualServer.com

 dietpi-launcher  = All the DietPi programs in one place.
 dietpi-config    = Feature rich configuration tool for your device.
 dietpi-software  = Select optimized software for installation.
 htop             = Resource monitor.
 cpu              = Shows CPU information and stats.

        ^
       *#*
      *###*           Lightweight Xmas Justice Tips:
     *#####*          - Be good, Amazon do sell coal.
    *#######*         - Cherish every moment, even the awkward family gatherings ;)
   *#########*        - 'Ho Ho Ho, Green Giant' swaps his job!
  *Merry  Xmas*
 *#############*      All the best from everyone at DietPi =))

root@RPi-Zero-W:~#

```
[image]

[image]

---------------

### RPi B+ (armv6l)

- [ ] Automated installation - onboard Ethernet Adapter - DHCP  - (works)

- [ ] Manual installation - onboard Ethernet Adapter - static IP address  - (works)

```
───────────────────────────────────────
 DietPi     | 16:44 | Mo 11.12.2017
 ───────────────────────────────────────
 V159       | RPi B+ (armv6l)
 ───────────────────────────────────────
 IP Address | 192.168.10.222
 ───────────────────────────────────────

 Created by : Daniel Knight
 Web        : http://DietPi.com
 Twitter    : http://twitter.com/dietpi_
 Donate     : http://goo.gl/pzISt9
 DietPi's web hosting is powered by: MyVirtualServer.com

 dietpi-launcher  = All the DietPi programs in one place.
 dietpi-config    = Feature rich configuration tool for your device.
 dietpi-software  = Select optimized software for installation.
 htop             = Resource monitor.
 cpu              = Shows CPU information and stats.

        ^
       *#*
      *###*           Lightweight Xmas Justice Tips:
     *#####*          - Be good, Amazon do sell coal.
    *#######*         - Cherish every moment, even the awkward family gatherings ;)
   *#########*        - 'Ho Ho Ho, Green Giant' swaps his job!
  *Merry  Xmas*
 *#############*      All the best from everyone at DietPi =))

root@RPi-B:~#
```

[image]

--------------

Can not replicate this issue.

For your RPi Zero W you need a [USB On-The-Go (USB-OTG)  micro-B USB to USB A female cable](https://www.rasppishop.de/Raspberry-Pi-Zero-USB-Adapter-Kabel) and a working USB 2.0 dongle with a supported chipset.

```
root@RPi-Zero-W:~# lsusb
Bus 001 Device 002: ID 0bda:8153 Realtek Semiconductor Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

k-plan, 2017-12-11:
@Fourdee 

BTW: (and nothing to do with this issue!)
 What I personally really dislike with `DietPi_v158_RPi-armv6-(Stretch).img` why to update `dietpi_rpi_kernel_4.9.zip	07-Dec-2017 17:36	   77M` on a image `DietPi_v158_RPi-armv6-(Stretch).img   10-Dec-2017 02:47	 631,2M` at inital boot up ❓ 

- Yes, I have a flat-rate
- Yes, my internet connection is fast
1. But network, especially WiFi, is slow

2. SBC processor are slow (especally RPi Zero and RPi B)

3. write speed on sd card is slow

 -  if you tested this 6 times or more now, like I do, you fly into a frenzy!
- and yes I'm old and my time is runnning out ... 😥 

😃

tatoosh, 2017-12-11:
thx @k-plan for your answer and time. i think i started with an old version of dietpi.txt - sorry for that.
Now i am testing with new dietpi.txt on Pi3 Wifi only - worked very nice!
Good work for v159! Thanks a lot

k-plan, 2017-12-11:
> Good work for v159! Thanks a lot

This congratulations must be directed to @Fourdee  and  @MichaIng , not to me. 😄 

Nice to hear it will work for you now. 👍

Fourdee, 2017-12-11:
@k-plan 

> What I personally really dislike with DietPi_v158_RPi-armv6-(Stretch).img :
Why to update dietpi_rpi_kernel_4.9.zip 07-Dec-2017 17:36 77M on a image DietPi_v158_RPi-armv6-(Stretch).img 10-Dec-2017 02:47 631,2M at inital boot up ⁉️

Kernel should not be updated during 1st run of this image, as it should match the current, i'll run tests to verify.

Fourdee, 2017-12-11:
@tatoosh 

> thx @k-plan for your answer and time. i think i started with an old version of dietpi.txt 

That would explain it. 

We patch dietpi.txt incrementally during dietpi-update, if your using a older dietpi.txt than the image version, the patching may miss some critical additions.

Solution is to always use the dietpi.txt, sourced from the image you are writing. This will ensure patching is correct.

I'll see if we can improve this to patch previous versions.

Fourdee, 2017-12-12:
Runs during ```dietpi-update``` to add missing ```dietpi.txt``` entries:
[image]

Fourdee, 2017-12-12:
Completed.

closed 2017-12-12 by Fourdee

tatoosh, 2017-12-13:
Thx @Fourdee