MichaIng/DietPi #1294: Nodered service runs as root

TerryE, 2017-12-11:
#### Required Information:
- DietPi Version: 159
- SBC Device: RPi 3
- Power supply used: RPI 3 standard 5V 2.5A
- SD card used:  Sandisk ultra 8Gb + USB attached Kingston UV400 120Gb SSD
- Distro: 4.9.62-v7+ armv7l GNU/Linux

#### Additional Information (if applicable):
- Node-red
- Standard DietPi install

#### Expected behaviour:

Node red is intended to be run from an unprivileged service account (much in the same way that Apache2 uses typically `www-data`), and hence node red flow developers do not have direct access to root services.  Linux provides standard sandboxing patterns (e.g. using `/etc/sudoers`) if specific privileged functions need to be made accessible to flows.

#### Actual behaviour:

Node-red runs as root, and this means that any flow has full root access to the processor.  

#### Steps to reproduce:
Construct a minimal flow `timestamp` -> `exec` -> `debug` with the exec script as `ps -u` and you get a `ps -u root` listing.  Alternatively replace this by a `shutdown -H now ` to get a better idea of the vunlerability.

#### Did you submit a ```dietpi-bugreport```?
No for this one.

#### Extra details:
I will post my suggested workaround here when done.

k-plan, 2017-12-11:
@TerryE 

Hi,
can you please share with us:

`cat /etc/debian_version`

TerryE, 2017-12-12:
> can you please share with us: `cat /etc/debian_version`

8.0

Fourdee, 2017-12-12:
@TerryE 

Thanks for the information, confirmed we currently use ```root``` for this installation:
https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-software#L9742-L9764

DarkElvenAngel, 2017-12-12:
I Have two installations where I use node-red my main NAS server and a Kodi box.  It was a simple task to add a node-red user and setup sudo on my NAS. However it can be very convenient to have node-red running under root as it does on my Kodi box where I use it to shutdown and reboot the system.

I would suggest having an option to have node-red run under either it's own user or root

TerryE, 2017-12-12:
> However it can be very convenient to have node-red running under root as it does on my Kodi box where I use it to shutdown and reboot the system.

@DarkElvenAngel , you do *not* need to run Node red as root to do this, but you do need to learn how to use sudoers.

TerryE, 2017-12-12:
@Fourdee Dan, the DietPi installation sometimes uses a convention which isn't aligned to normal Linux practices. I've raised this as a separate issue #1297

DarkElvenAngel, 2017-12-12:
@TerryE your exactly right I don't need to run as root I just never bothered to set my Kodi box to run as it's own user

DarkElvenAngel, 2017-12-12:
@Fourdee I have already set up and run node-red as it's own user would there be a good place to post my setup?

Fourdee, 2017-12-13:
@DarkElvenAngel 

> @Fourdee I have already set up and run node-red as it's own user would there be a good place to post my setup?

Thanks for the offer, by all means post if you are willing to share 👍

DarkElvenAngel, 2017-12-14:
I'm attaching my setup for node-red that runs under it's own user
Stop node-red service

``` service node-red stop ```

Next edit the  **/etc/systemd/system/node-red.service** change the User=root to User=node-red

```nano /etc/systemd/system/node-red.service```

```[Unit]
Description=Node-Red

[Service]
Type=simple
User=node-red
ExecStart=/usr/local/bin/node-red -u /mnt/dietpi_userdata/node-red

[Install]
WantedBy=multi-user.target
```
Created a user and group for node-red 
```addgroup --gid 55 node-red```
```adduser --home /mnt/dietpi_userdata/node-red --no-create-home --shell /bin/false --uid 55 --gid 55 node-red```

Give the user node-red the ability to write to the it directory
```
chmod ug+rwx -R  /mnt/dietpi_userdata/node-red
chown node-red:node-red -R  /mnt/dietpi_userdata/node-red
```

Since we have change the service run
```systemctl daemon-reload```

Then we start node-red service with
```service node-red start```

Now node-red is running under the node-red user

If you want node-red to run commands with sudo we run
```echo "node-red ALL=NOPASSWD: ALL" >> /etc/sudoers```

DarkElvenAngel, 2017-12-14:
@TerryE for your application you can try using the www-data user but I haven't tested that. I know this setup works been running for 8+ months

TerryE, 2017-12-14:
@DarkElvenAngel, as a general principle you should use separate service accounts for separate services.  That way you have the option of using UID / GID-based access control to isolate web apps from Node RED ones, etc.  Such dynamically allocated service accounts are usually allocated in the 100-500 ranges<sup>[(1)](https://refspecs.linuxfoundation.org/LSB_3.0.0/LSB-Core-generic/LSB-Core-generic/uidrange.html)</sup>.

DarkElvenAngel, 2017-12-14:
@TerryE I choose 55 because I need a static UID that is available on all nodes on the network. 55 is not used.

On FreeBSD, porters who need a UID for their package can pick a free one from the range 50 to 999 and then register the static allocation.

DarkElvenAngel, 2017-12-14:
@TerryE so I will update to UID:gid 155:155 and that should satisfy all requirements, yes?  If so I'll update my setup post.  Thanks for the insight.

TerryE, 2017-12-16:
@Fourdee perhaps you can unlock #1297 so I can respond in the correct issue.  Re logs, the services still log to the RAMdisk `/var/log` hierarchy. Dietpi optionally preserves these statically prior to shutdown.  The correct place to do this is in the `/var/lib/dietpi` hierarchy, though there is no reason why this shouldn't optionally be symlinked to a node in another FS.

TerryE, 2017-12-28:
@Fourdee Dan, BTW I think that the title is wrong: it should be **Run nodered under its own service account**.  There are good reasons why this should be separate from `www-data`.

@DarkElvenAngel, you typically don't statically allocated service account UIDs.  I created my version by
```bash
adduser --system --group --no-create-home --shell=/bin/nologin nodered
```

Fourdee, 2017-12-29:
NodeRed will run under user ```nodered``` in v160, thanks @TerryE 👍 

Test install passed.
Completed.

closed 2017-12-29 by Fourdee