MichaIng/DietPi #1295: Kodi crashes at boot

DarkElvenAngel, 2017-12-12:
## Creating a bug report/issue:

#### Required Information:
- DietPi 159
- Raspberry Pi 0w
- Power Supply 5v 2.5Amp Belkin
- SD card used NOOBS 8Gb
- Distro Stretch

#### Additional Information (if applicable):
- Kodi
- Can this issue be replicated on a fresh installation yes

#### Expected behaviour:
Boot to Kodi menu

#### Actual behaviour:
ERROR MESSAGE
Illegal command (core dumped)
Then stuck in a loop forever
#### Steps to reproduce:
Fresh installation
Install Kodi
Set automatic boot to kodi

#### Did you submit a ```dietpi-bugreport```?

#### Extra details:
I don't know if this is a DietPi issue however hen swapping the SD card to a RPI2 or RPI3 it boots as expected.

Fourdee, 2017-12-16:
@DarkElvenAngel 

Thanks for the report 👍 

Notes:
- Possibly Stretch with ARMv7 only binary in Raspbian repo?

DarkElvenAngel, 2017-12-16:
@Fourdee I just checked that out with a fresh Raspbian-Stretch-Lite install and it worked without problem, I then rebooted my DietPi card and ran ```apt-get update``` ```apt-get upgrade``` The was a update for kodi and now it's working.

I suspect there was a missing ARMv6 binary as I read it was suppose to auto-select between the two based on which would be best. Ref https://www.raspberrypi.org/forums/viewtopic.php?t=192499

Just to confirm this was setup with the ```dietpi-software``` install option I want to be sure this just preforms a ```apt-get install kodi``` or similar and not something else that may have caused this issue.

If everything is good on the DietPi end I am confirming this issue is solved and can be marked closed.
Thank you for DietPi!

Fourdee, 2017-12-17:
@DarkElvenAngel 

Thanks for confirming cause of issue 👍 

I'll check the sourcecode and see why this is failing in our installer.

closed 2017-12-17 by Fourdee