MichaIng/DietPi #1299: DietPi-Software | Wifi Hotspot  RPI3.v.159  not working out of the box.

yuret, 2017-12-12:
## Creating a bug report/issue:
hostapd wont start and the log is:

error while loading shared libraries: libssl.so.1.0.0

 
#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```?
- SBC Device (EG: RPi 3)?
- Power supply used (EG: 5v 1A RAVpower)?
- SD card used (EG: Sandisk ultra)?
- Distro (EG: Jessie) | ```uname -a```?

root@DietPi:~# cat /DietPi/dietpi/.version
159
RPi 3
Transcend UHS-I 16 Gb
#### Additional Information (if applicable):
- Software title?
- Can this issue be replicated on a fresh installation of DietPi?
Yes

#### Expected behaviour:

Wi-Fi Hotspot should work
#### Actual behaviour:

Wi-Fi Hotspot should not working
#### Steps to reproduce:

#### Did you submit a ```dietpi-bugreport```?

No
#### Extra details:

Fourdee, 2017-12-14:
@yuret 

Thanks for the report 👍 

If possible, please could you test following (copy and paste all into term):
```
wget http://dietpi.com/downloads/binaries/all/libssl1.0.0_1.0.1t-1+deb8u7_armhf.deb -O package.deb
dpkg -i package.deb
dietpi-services restart
dietpi-services status
```

Fourdee, 2017-12-14:
https://github.com/Fourdee/DietPi/issues/1303#issue-281684601

yuret, 2017-12-14:
Thank you! Your solutions works:
DietPi-Services
─────────────────────────────────────────────────────
 Mode: status
 Please wait...

 [Ok] cron      active (running) since Thu 2017-12-14 08:14:18 GMT; 11s ago
 [Ok] isc-dhcp-server   active (running) since Thu 2017-12-14 08:14:20 GMT; 8s ago
 [Ok] hostapd   active (running) since Thu 2017-12-14 08:14:21 GMT; 8s ago

Fourdee, 2017-12-15:
@yuret 

Legend 👍 Thanks for confirming.

I've applied the fix for v160. Completed.

closed 2017-12-15 by Fourdee