MichaIng/DietPi #1308: DietPi-Software | homeassistant: Add a soft link for config folder at install

borpin, 2017-12-16:
## Feature Request

### Background
Same as issue [#1256](https://github.com/Fourdee/DietPi/issues/1256)
Homeassistant users expect to find their configurations etc in a folder ~/.homeassistant.  DietPi uses the folder /mnt/dietpi_userdata/homeassistant.

### Solution
Add a soft link at install time from ~/.homeassistant to /mnt/dietpi_userdata/homeassistant

borpin, 2018-09-04:
This is really old - was it before V6?

I think it is likely that the user would use HA from the same user account as they installed it from (unless unattended).  Adding it to that users home folder would help new users.

Perhaps an item in the install instructions would be sufficient.

borpin, 2018-09-04:
By configuring HA to use ```/mnt/dietpi_userdata/homeassistant``` as the configuration dir, is there a risk if 2 users use HA on the same device that they will both point to the same config files?  Would they both have the same permissions?

Should ```/mnt/dietpi_userdata/homeassistant``` actaully be ```/mnt/dietpi_userdata/<user>/homeassistant```?

Again, if this is a specific design/setup decision (not saying it is the wrong decision), that should be explained to a user in the notes.  Ultimately, could you offer the choice between per user or global setup?

Do the same sort of issues appear with node-red (not tried it recently)?

MichaIng, 2018-09-04:
@borpin 
Indeed from the ground up DietPi was designed as a single user (let's say single login) system. We just recently adjusted some installs to place data and config folders inside `/mnt/dietpi_userdata/`, on the one hand because this can be easily placed onto external drives (to reduce SDcard I/O and/or speed up file transfer), on the other hand because it makes it easier to backup and restore the system with application data.

But this does not include all software titles or packages. E.g. `htop` places a config dir/file into the users home dir right on execution and many other packages/software title will do it similar. Only for software that is either started via systemd unit/init.d service or allows to change config/data dir via command argument/settings file, this is possible.

> By configuring HA to use /mnt/dietpi_userdata/homeassistant as the configuration dir, is there a risk if 2 users use HA on the same device that they will both point to the same config files? Would they both have the same permissions?

Installing Homeassistant via dietpi-software places a single systemd unit with a separate run user to start it on boot. Thus there is just one instance for all users and nothing else makes sense. The instance runs as it's own separate unix user `homeassistant` with it's own config dir, symlinked to dietpi_userdata. Other unix users need to be in `dietpi` group to have direct access to this config folder.
But just to clarify: This does not necessarily has to do something with the users that can login the homeassistant application UI. Sorry as I never used it, I would need to check the UI etc. how this is handled, but different end users can use the same homeassistant instance that runs under a separate unix user. Per-HAuser configuration should be stored all within the same config folder/file. But again, I need to check.

Fourdee, 2018-09-09:
@borpin 

@MichaIng Has pretty much covered the original questions, however, i'll add a little to this aswell, try to clarify.

> By configuring HA to use /mnt/dietpi_userdata/homeassistant as the configuration dir, is there a risk if 2 users use HA on the same device that they will both point to the same config files? Would they both have the same permissions?

The config file is a single item required by the program, regardless of the amount of users on HA. To prevent additional users from making changes to the HA config files, it would require HA to implement a user login system. Its not something we can do our end, as running multiple HA servers, via multiple unique install directories, is not a solution.

> Again, if this is a specific design/setup decision (not saying it is the wrong decision), that should be explained to a user in the notes. Ultimately, could you offer the choice between per user or global setup?

DietPi follows the standard rules of process execution, in that:
- A service is created to run the application
- That service has a user and group run level (which are under privileged)
- Past this point, the program itself would need to implement its own user account system. EG: I believe NextCloud allows for adding additional users.

I hope this answers your questions, I'll mark this as closed, however, please reopen if required.

closed 2018-09-09 by Fourdee