MichaIng/DietPi #1310: [Aria2][Stretch+] libssl1.0.0 missing, binary needs update for Stretch+

MichaIng, 2017-12-17:
http://dietpi.com/phpbb/viewtopic.php?f=11&t=2573

Our Aria2 binary expects [libssl1.0.0](https://packages.debian.org/jessie/libssl1.0.0) to be installed. But since Stretch, just [libssl1.0.2](https://packages.debian.org/stretch/libssl1.0.2) is shipped.

We use our own binary for DietPi: https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-software#L6548, but why not going with the debian repo package? https://packages.debian.org/stretch/aria2
Should automatically solve dependency issues.

Fourdee, 2017-12-17:
@MichaIng 

Thanks 👍 

Same issues as:
- https://github.com/Fourdee/DietPi/issues/1306
- https://github.com/Fourdee/DietPi/issues/1299#issuecomment-351623366

There is a few more already in ```dietpi-software```
We should probably have a linux install option for this, so we can pull in shared install as needed.

Fourdee, 2017-12-17:
@MichaIng 

Yep, lets go for the APT install for stretch, sending commit in a few.

Fourdee, 2017-12-17:
Tested install and web interface:
- 🈯️ RPi 3 Stretch

Completed.

closed 2017-12-17 by Fourdee