MichaIng/DietPi #1311: DietPi-Globals | Replace duplicate function/var code in our scripts

Fourdee, 2017-12-17:
By simply loading this script on demand.

Fourdee, 2017-12-17:
@MichaIng 

```
root@DietPi:~# AGI thispackagedoesnotexist_Test
 [Info] APT is processing, please wait...

E: Unable to locate package thispackagedoesnotexist_Test
```

MichaIng, 2017-12-17:
Nice! 👍 
Just some minor questions/hints:
- Why not include the error check/output to base AGI etc. directly? Would be helpful in every script and terminal, I think.
- Since -qq includes -y, we can remove the latter: https://wiki.ubuntuusers.de/apt/apt-get/ Why ubuntu wiki is so much better/more detailed than debian ones? Totally missing complete command overviews with all sub commands and options there! 😛.

Fourdee, 2017-12-19:
ToDo:
- 🈯️ Global dietpi-notify
- Replace existing ```apt-get``` commands in dietpi scripts with ```AGI``` etc
- Send my commit
- Manually fix conflicts on outstanding PR's
-- + using global dietpi-notify
-- + using global AGX commands

Fourdee, 2017-12-19:
@MichaIng 

> Why not include the error check/output to base AGI etc. directly? Would be helpful in every script and terminal, I think.

We can provide a notify for failure, but if we want to handle the end result, we'll need to do this outside the global script.
eg: in ```dietpi-software```, we need to handle the error code, then we can match it to software title etc. If we simply ```exit``` in ```dietpi-global```, we lose the ability to do that.

Fourdee, 2017-12-19:
This is now completed as intended, however, we'll continue to add/improve it as needed in the future.

Fourdee, 2017-12-20:
dietpi-drive_manager move userdata:
[image]

Fourdee, 2017-12-20:
Completed (again).

Fourdee, 2017-12-20:
- use of ```G_``` at start of all global commands/vars to clearly define global status, and, avoid variable interaction
```
root@DietPi:~# G_
G_AGA              G_AGI              G_AGUP             G_CHECK_URL
G_AGDUG            G_AGP              G_CHECK_ROOTFS_RW  G_DIETPI-NOTIFY
G_AGF              G_AGUG             G_CHECK_ROOT_USER
```

Fourdee, 2017-12-20:
@MichaIng 

Globals are now finalized, using ```G_format``` https://github.com/Fourdee/DietPi/issues/1311#issuecomment-352999485

Completed (for now) 😄

Fourdee, 2017-12-22:
Check and automatically install required packages:
[image]

Fourdee, 2017-12-22:
Note to self:
- Allo GUI replace ```$(/DietPi/dietpi/dietpi-cpuinfo 1)``` with ```G_OBTAIN_CPU_TEMP```

Fourdee, 2017-12-23:
### Improved error handling:
[image]

MichaIng, 2017-12-28:
@Fourdee 
Maybe we can include the option the directly send the error as dietpi-bugreport (together with all the other system data)?
_______________
About bug reporting generally, although this would belong into an own topic, it would be nice to have an option or by default make an output that one can copy&paste directly into github preformatted.
Nextcloud e.g. offers this via an app and by this all information is together directly: https://apps.nextcloud.com/apps/issuetemplate

MichaIng, 2017-12-29:
I played around a bid with global error handling and info echo during installation. It would be nice to reduce the overall amount of echo lines without reducing the information content.

I loved the installation script of PiHole and had a look into their method:
- `echo -n 'test...'` skips new line
- `"\\r\\033[K"` at the beginning of string, will overwrite existing line.
- Thus you can use:
```
echo -n 'Command is running...'
if ( command succeded ); then
  echo -e "\\r\\033[KCommand succeded"
else
  echo -e "\\r\\033[KCommand failed"
end
```
... to just show the final result of a command, after it's finished.
- Maybe we can use this for `G_RUN_CMD`?

Another issue I found with `G_RUN_CMD` is, if you want to forward output into a file, e.g. `G_RUN_CMD echo "blabla" > /path/to/file`, the echo output of error handler will also be found in the file. I'm not sure how in this case it is set, where the input for `G_RUN_CMD` ends and in case it's output could be treated. Is it possible to somehow parenthesise the error handlers input?

Fourdee, 2017-12-30:
@MichaIng 

> Maybe we can include the option the directly send the error as dietpi-bugreport (together with all the other system data)?

Love the idea 👍 

The main issue we have is structure, currently bug reports are simply uploaded to an FTP server. Its really not that much fun to work with bug reports in that format.

Maybe we could add a simple error send on failures, however, we'd need a better system to log them than FTP. SQL db on server, some kind of API on github that allows us to automatically generate a ticket? All options need research.

Fourdee, 2017-12-30:
@MichaIng 

> Another issue I found with G_RUN_CMD is, if you want to forward output into a file, e.g. G_RUN_CMD echo "blabla" > /path/to/file, the echo output of error handler will also be found in the file. I'm [...]

Yep, had the same issue with ```cat << _EOF_ > file```. I'll take a look, but we may need another command for this.

Fourdee, 2017-12-30:
@MichaIng 

> I loved the installation script of PiHole and had a look into their method:

Yep, i agree, very clean.

Currently, i've simply enabled as much info as possible during dev, using existing ```dietpi-notify```. We can always trim/clean/refine it at a later date.
Notify may have an issue with ```echo -n 'test...```, would need testing.

Fourdee, 2017-12-30:
@MichaIng 
having a little play with your idea 👍 
[image]

Fourdee, 2017-12-30:
test:
#### Required Information:
- DietPi Version | 160
- SBC Device     | oDroid C2 (aarch64) (index=12)
- Distro         | buster (index=5)
- Command        | G_AGI: nevergonnawork
- Error Handler  | G_ERROR_HANDLER

#### Additional Information (if applicable):
- Software title | DietPi-Software

#### Expected behaviour:

#### Actual behaviour:

#### Steps to reproduce:

#### Additional logs:
Log file contents:
E: Unable to locate package nevergonnawork

MichaIng, 2017-12-30:
Nice so far 👍.
- On $G_USER_INPUT => whiptail, we could add an user choice at the bottom to create the guthub ready output on demand (if user has/agrees to create github account e.g.).
- If it is possible to then create this within whiptail and automark the lines (at least on SSH it marks all parts from outside the whiptail "window" as well), we could easily add more information to the output and replace the existing dietpi-bugtracker partly with this. But yeah it should stay for exceptional hard issues, as I know there are muuuch files/configs/logs gathered and copied to your server.

MichaIng, 2017-12-30:
@Fourdee 
I played around also a bid with process messages and error handling. Basically I merged some status messages to reduce the output lines and established a new G_DIETPI-NOTIFY option "-2" == process. This process messages will not produce a newline and will be overwritten by "ok" and "failure" outputs. Thus in the end you will only see the final results and still perhaps interesting/important "2" info messages in between.

I also made the brackets all same size. As "Processing" is a long word, this might need tuning, anyway is just an offer/idea, how we could handle our output:

After adding some "2" notifications to ownCloud installation and running some of it's commands as "G_RUN_CMD" it looks like this:
[image]

During (longer) G_AGX/G_CMD_RUN commands you will see:
[image]

Fourdee, 2017-12-30:
@MichaIng 

> During (longer) G_AGX/G_CMD_RUN commands you will see:

Love it, looks brilliant, very nice work! 👍

k-plan, 2017-12-30:
I want to play around as well: 😄 

[image]

#### Required Information:
- DietPi Version | v
- SBC Device     | oDroid XU3/4 (armv7l) (index=11)
- Distro         | stretch (index=4)
- Command        | G_AGA:
- Error Handler  | G_ERROR_HANDLER

#### Additional Information (if applicable):
- Software title | DietPi-PREP_SYSTEM_FOR_DIETPI

#### Expected behaviour:

- Build a new DietPi image for my Odroid HC1

#### Actual behaviour:

- ` [Error] Unable to continue, the program will now terminate.` 😢 

#### Steps to reproduce:

- set: `GIT_BRANCH=testing`
- run: `:~# ./PREP_SYSTEM_FOR_DIETPI.sh`

#### Additional logs:
```
Log file contents:
Removing libapt-inst1.5:armhf (1.0.9.8.4) ...
Removing libapt-pkg4.12:armhf (1.0.9.8.4) ...
Removing libcap2-bin (1:2.25-1) ...
Removing libirs-export91 (1:9.9.5.dfsg-9+deb8u14) ...
Removing libdns-export100 (1:9.9.5.dfsg-9+deb8u14) ...
Removing libestr0 (0.1.10-2) ...
Removing perl (5.24.1-3+deb9u2) ...
Removing libperl5.24:armhf (5.24.1-3+deb9u2) ...
Removing libgdbm3:armhf (1.8.3-14) ...
Removing libgnutls-deb0-28:armhf (3.3.8-6+deb8u7) ...
Removing libgpm2:armhf (1.20.4-6.2+b1) ...
Removing libhogweed2:armhf (2.7.1-5+deb8u2) ...
Removing libpsl0:armhf (0.5.1-1) ...
Removing libicu52:armhf (52.1-8+deb8u6) ...
Removing libisccfg-export90 (1:9.9.5.dfsg-9+deb8u14) ...
Removing libisc-export95 (1:9.9.5.dfsg-9+deb8u14) ...
Removing liblogging-stdlog0:armhf (1.0.5-2+b2) ...
Removing nfacct (1.0.1-1.1+b1) ...
Removing libnetfilter-acct1:armhf (1.0.2-1.1) ...
Removing libnettle4:armhf (2.7.1-5+deb8u2) ...
Removing libnfnetlink0:armhf (1.0.1-3) ...
Removing libpipeline1:armhf (1.4.1-2) ...
Removing libprocps3:armhf (2:3.3.9-9) ...
Removing libreadline6:armhf (6.3-8+b3) ...
Removing libssl1.0.0:armhf (1.0.1t-1+deb8u7) ...
Removing libusb-0.1-4:armhf (2:0.1.12-30) ...
Removing linux-image-4.9-armhf-odroid-xu3 (4.9.61-1) ...
Removing linux-image-4.9.61+ (4.9.61-20171118-XU3) ...
dpkg: error processing package linux-image-4.9.61+ (--remove):
 subprocess installed pre-removal script returned error exit status 1
Removing perl-modules-5.24 (5.24.1-3+deb9u2) ...
Removing xz-utils (5.2.2-1.2+b1) ...
dpkg: u-boot-tools: dependency problems, but removing anyway as you requested:
 linux-image-4.9.61+ depends on uboot-mkimage | u-boot-tools; however:
  Package uboot-mkimage is not installed.
  Package u-boot-tools is to be removed.

Removing u-boot-tools (2016.11+dfsg1-4) ...
dpkg: uboot: dependency problems, but removing anyway as you requested:
 linux-image-4.9.61+ depends on uboot; however:
  Package uboot is to be removed.
  Package uboot which provides uboot is to be removed.
 linux-image-4.9.61+ depends on uboot; however:
  Package uboot is to be removed.
  Package uboot which provides uboot is to be removed.

Removing uboot (20170824-9+deb8u1) ...
Errors were encountered while processing:
 linux-image-4.9.61+
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

k-plan, 2017-12-30:
Next run testing fun: 😃 

[image]

#### Required Information:
- DietPi Version | v
- SBC Device     | NULL (index=11)
- Distro         | stretch (index=4)
- Command        | G_AGI: apt-transport-https apt-utils bash-completion bc bzip2 ca-certificates console-setup cron curl dbus debconf dosfstools dphys-swapfile dropbear ethtool fake-hwclock fbset firmware-realtek gnupg hdparm hfsplus htop initramfs-tools iputils-ping isc-dhcp-client locales nano net-tools ntfs-3g ntp p7zip-full parted psmisc resolvconf sudo tzdata unzip usbutils wget whiptail wput zip linux-image-armhf-odroid-xu3 crda firmware-atheros firmware-brcm80211 firmware-ralink iw rfkill wireless-tools wpasupplicant
- Error Handler  | G_ERROR_HANDLER

#### Additional Information (if applicable):
- Software title | DietPi-PREP_SYSTEM_FOR_DIETPI

#### Expected behaviour:

[Info] Non-DietPi system

- Build a new DietPi image for my Odroid HC1 from https://oph.mdrjr.net/meveric/images/Jessie/Debian-Jessie-1.1.4-20171121-XU3+XU4.img.xz

#### Actual behaviour:

- after 1h of running and updating files .... 🕛  🕐  😢  

[Error] Unable to continue, the program will now terminate.

- after this, system is f... up, `apt-get update` won't run any more.

#### Steps to reproduce:

- download the image, write it to sd card and after booting up run: `:~# ./PREP_SYSTEM_FOR_DIETPI.sh`

#### Additional logs:
```
Log file contents:
E: Method https has died unexpectedly!
E: Sub-process https received a segmentation fault.
E: Method /usr/lib/apt/methods/https did not start correctly
```

-----------------

### General Notes:

- run `apt-get update` as first step because `a_MIN_APT_PREREQS` will fail elsewhere on a fresh image

- first run and set `dpkg-reconfigure locales` to `en_GB.UTF-8` to avoid hundert messeges like this:
```
warning: Setting locale failed.
 warning: Please check that your locale settings:
        LANGUAGE = (unset),
        LC_ALL = (unset),
        LANG = "en_GB.UTF-8"
    are supported and installed on your system.
warning: Falling back to the standard locale ("C").
locale: Cannot set LC_CTYPE to default locale: No such file or directory
locale: Cannot set LC_MESSAGES to default locale: No such file or directory
locale: Cannot set LC_ALL to default locale: No such file or directory
```

- set: `passwd` to a known password like "dietpi" for user "root"

------------------

NB: - how will we handle differrent disk layouts? 
Not two partitions which first is "boot" (FAT) / second is "/" (ext4), eg one big BTRFS ?

-------------

Just like that, the fun's over for today.  😃

MichaIng, 2017-12-30:
@k-plan 
Ah just found this one: http://www.hardkernel.com/main/products/prdt_info.php?g_code=G150229074080
> Any OS for XU4 is fully compatible with the HC1.

And as I see the packages, the image seems to be based on Meverics?

- For this device we choose the default kernel meta package `linux-image-armhf-odroid-xu3`: https://github.com/Fourdee/DietPi/blob/testing/PREP_SYSTEM_FOR_DIETPI.sh#L553
- But it depends on ancient `linux-image-3.10.106+`: https://oph.mdrjr.net/meveric/dists/all/xu3/binary-armhf/Packages Thus the existing kernel package will removed first.
- On official Debian images, `linux-image*` packages are blocked from autoremove by `/etc/apt/apt.conf.d/01autoremove(-kernels)`, which seems not the case for Meverics image.
- Removing the only installed kernel package seems to need failing, which causes the dpkg error?
- On HC1 product page they promote the use of newest 4.14 kernel on their official linux (Ubuntu based) image, so I assume `linux-image-4.14-armhf-odroid-xu4` to be stable enough also for us? Otherwise at least we should use `linux-image-4.9-armhf-odroid-xu3`.
- xu3/xu4 naming mixed together there, both are meta packages depending on linux-image-4.XX.XX packages where no device naming appears any more, thus I think they are compatible vice versa and for HC1 as well.
- **To prevent any kernel removal issues during image preparation: Install the chosen kernel package before `G_AGA`, or identify and choose already installed kernel package. I would vote for first solution, as we then keep control over which kernel is used with DietPi.**

k-plan, 2017-12-30:
@MichaIng 

sorry @Fourdee , no english, to time-consuming

Ja, die Odroid Welt ist etwas schwieriger ....

- Die wollen kompatibel bleiben für: XU4/XU3/HC1/MC1 und nur ein Image (+ Quellen) pflegen

> We guarantee the production of ODROID XU4/XU3/HC1/MC1 to the middle of 2020, but expect to continue production long after.

Hardkernels Images (Ubuntu) sind schon zur Zeit auf 4.14.x , da gibt es auch die Kernel source für:
https://forum.odroid.com/viewtopic.php?f=146&t=28895

Meveric bastelt aus der Kernel Source ein Debian draus und pflegt das schon seit Jahren. Sein Schwerpunkt ist die Gaming Welt ( https://forum.odroid.com/viewtopic.php?f=11&t=2684 ) daher auch die gute Unterstützung für Kodi, XBMC, Grafikausgabe über HDMI.
Der ist aber auch eine "One-Man-Show" wie Dan bisher. Daher hängt er immer etwas hinterher.
Hatte aber die gesamte Mali-T628 MP6 Hardware Grafik Unterstützung als erster laufen und lange Zeit die Nase vorn damit.

Mit dem 4.x Kernel ist es zum "Bruch" gekommen, so wie ich das bis jetzt verstanden habe.
Bis 3.x nutzen alle ODROID XU4/XU3/HC1/MC1 nur `exynos5422-odroidxu3.dtb`
Seit 4.x wir es spezifischer und man muss das richtige beim booten laden je nach verwendeten Board: 
```
exynos5422-odroidxu3-lite.dtb
exynos5422-odroidxu3.dtb
exynos5422-odroidxu4-kvm.dtb
exynos5422-odroidxu4.dtb
```

Da gibt es zwar ein paar nette Tricks für, wie man das über die `boot.ini` steuern könnte, nur haben wir das im Moment (noch) nicht (_Code Schnippsel entliehen_):

```
if test "${board_name}" = "xu4"; then setenv fdtfile "exynos5422-odroidxu4.dtb"; fi
if test "${board_name}" = "xu3"; then setenv fdtfile "exynos5422-odroidxu3.dtb"; fi
if test "${board_name}" = "xu3l"; then setenv fdtfile "exynos5422-odroidxu3-lite.dtb"; fi

if ext4load mmc 0:1 0x00000000 "/boot/.next" || fatload mmc 0:1 0x00000000 ".next"  || ext4load mmc 0:1 0x00000000 ".next"; then echo "Found mainline kernel configuration"; else setenv fdtfile "exynos5422-odroidxu3.dtb"; fi
ext4load mmc 0:1 0x44000000 /boot/dtb/${fdtfile} || fatload mmc 0:1 0x44000000 dtb/${fdtfile} || ext4load mmc 0:1 0x44000000 dtb/${fdtfile}
```

Weitere Schwierigkeit, Fourdee will es "schlank" haben und will eine FAT boot Partition, in der man einfach Konfigs editieren kann.
Beides finde ich sehr gut. Aber herzlichen Glückwunsch, wenn das Gerät von SD-Karte, eMMC, USB und HDD/SSD booten kann. Das wird dann mit der `/etc/fstab` sehr schnell unschön und die Tricksereien fangen an.

Meine persönlichen "Erfahrungen" mit den verschiedenen Kernel:

- 3.x hat eine schlechtere RAM Performance, warum auch immer.

- 4.9.x ist mit dem RAM der schnellste

- 4.14.x ist etwas langsamer bei meinen Tests

Schreib/Lesegeschwindigkeit (SD-Karte), RAM Größe und Schreib/Lesegeschwindigkeit, machen bei den Devices, neben Prozessortakt und Kerne Anzahl den größten Unterschied aus.

Klar haben neuere Kernel meist bessere Unterstützung für "neue Features", doch persönlich brauche ich die wenigsten davon. Würde also vorerst zu 4.9.x bei XU4 tendieren.
Da wird aber von Seiten Hardkernel erst einmal nichts mehr kommen.

Ordoid C1/C1+/C2 schenke ich mir an dieser Stelle erst einmal. Das sind noch ganze andere Nummern mit anderen Schwierigkeiten.

k-plan, 2017-12-30:
@Fourdee 
@MichaIng 

Only for clarification: 
I like your work so much. You are on a good trip and I think and hope you have finally success.
Only difficulties arising from a new way.
Keep one's tail up boys and girls. 😃

MichaIng, 2017-12-30:
https://github.com/Fourdee/DietPi/issues/1357

Fourdee, 2018-01-01:
Testing/notes: https://github.com/Fourdee/DietPi/pull/1364
[image]

Fourdee, 2018-01-02:
Notes: https://github.com/Fourdee/DietPi/pull/1365
- Looks much better 👍 
- ```G_CHECK_URL http://google.comxxxx``` hangs, stuck in BG.

Fourdee, 2018-01-08:
Notes:
- Incorrect usermesg (not cleared from previous)
- Missing dietpi-software info 
```
 [  Error   ] DietPi-Software: G_AGI: mosquitto
 [   Info   ] exit_code = 100
 [   Info   ] VERSION:v6.-1 | HW_MODEL:3 | HW_ARCH:2 | DISTRO:4
 [   Info   ] Log file contents:
E: Failed to fetch http://repo.mosquitto.org/debian/pool/main/m/mosquitto/mosquitto_1.4.12-0mosquitto1_armhf.deb  404  Not Found
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
 [   Info   ] User message:
Connection failed while trying to access: http://dietpi.com/downloads/binaries/all/libwebsockets3_1.2.2-1_armhf.deb. The URL may be offline, or invalid.
 [   Info   ] If problems persist, please report this to DietPi for investigation, including a screenshot of this error! (https://github.com/Fourdee/DietPi/issues).

```

Fourdee, 2018-01-14:
I'll mark this as completed. Will reopen if required.

closed 2018-01-14 by Fourdee