MichaIng/DietPi #1313: DietPi-Software | Pi-Hole: Use the updated block page?

Fourdee, 2017-12-18:
> [image]
https://github.com/Fourdee/DietPi/issues/1298#issuecomment-352314748

k-plan, 2017-12-18:
@Fourdee 

Nice, thanks! 👍 

[image]

```
root@RPi-Zero_W-Stretch:~# ls -lah /var/www/
total 16K
drwxr-xr-x  4 root     root     4.0K Dec 18 12:45 .
drwxr-xr-x 12 root     root     4.0K Dec 18 12:44 ..
drwxr-xr-x  2 root     root     4.0K Jan 14  2017 cgi-bin
drwxrwxr-x  4 www-data www-data 4.0K Dec 18 13:24 html
```

```
root@RPi-Zero_W-Stretch:~# ls -lah /var/www/html/
total 20K
drwxrwxr-x 4 www-data www-data 4.0K Dec 18 13:24 .
drwxr-xr-x 4 root     root     4.0K Dec 18 12:45 ..
drwxr-xr-x 7 root     root     4.0K Dec 18 12:44 admin
-rw-r--r-- 1 root     root     3.3K Dec 18 12:46 index.lighttpd.orig
drwxr-xr-x 2 root     root     4.0K Dec 18 13:24 pihole
```

```
root@RPi-Zero_W-Stretch:~# ls -lah /var/www/html/pihole/
total 40K
drwxr-xr-x 2 root     root     4.0K Dec 18 13:24 .
drwxrwxr-x 4 www-data www-data 4.0K Dec 18 13:24 ..
-rwxr-xr-x 1 root     root      13K Dec 18 13:24 blockingpage.css
-rwxr-xr-x 1 root     root      14K Dec 18 13:24 index.php
```

-----------------------

[image]

```
root@NanoPi-Neo:~# ls -lah /var/www/ | grep admin
lrwxrwxrwx  1 www-data www-data   19 Dez 11 20:11 admin -> /var/www/html/admin
lrwxrwxrwx  1 www-data www-data   19 Dez 11 20:11 pihole -> /var/www/html/admin
```

-------------------------

Can I help you, eg. provide additional informations or config files ?

Think, where is/will a problem, to get:

`pihole -r`

and

`pihole -up`

to run proper on all different web stacks.

LexiconCode, 2017-12-24:
Would there be a possibility to specify the port that Pi Hole uses for the web administration panel? That way other services can utilize port 80.

MichaIng, 2018-04-09:
@LexiconCode 
The current implementation should allow other services to use port 80 as well, as PiHole web ui ist just available via `.../pihole` sub location. But currently PiHole reconfiguration and update breaks this, as it reapplies it's own Lighttpd config, at least if there is an update for the web ui available.

Hui, we have now 3 issues opened about basically the same topic:
- https://github.com/Fourdee/DietPi/issues/1313
- https://github.com/Fourdee/DietPi/issues/1375
- https://github.com/Fourdee/DietPi/issues/1623

Need to merge this, collecting all helpful information...

As PiHole is great stuff that is widely used by our users, I wan't to focus a bid to get this solved, e.g. via pihole service hooks, as mentioned within the other topic. For practical reasons just need a quick possibility to fake a lower version for `pihole -up` 😉.

MichaIng, 2018-04-10:
Close in favour of https://github.com/Fourdee/DietPi/issues/1696

closed 2018-04-10 by MichaIng