MichaIng/DietPi #1314: Nextcloud install fails - only a limited number of commands

guny74, 2017-12-18:
## Creating a bug report/issue:
Hello,

I update to dietpi V159 on BPI m2+. 

then I tried to (new) install Nextcloud via "Dietpi-software", the installation stops with error 

> Nextcloud is not installed - only a limited number of commands are available

#### Required Information:
- DietPi Version | V159
- BPI M2+
- Distro| 
Linux DietPi 3.4.113-sun8i #4 SMP PREEMPT Wed Nov 22 13:45:28 CET 2017 armv7l GNU/Linux

#### Additional Information (if applicable):
- nextcloud
- Problem also on V158 an can be replicated

Fourdee, 2017-12-18:
@guny74 

Thanks for the report, we'll try to replicate 👍

MichaIng, 2017-12-18:
@guny74 
This massage can only occur, if the `nextcloud/occ` is used but `config.php` does not have `'installed' => 'true',` inside. I'm at the moment not sure, if for no reason the message also appears on running occ for installation itself. Will try later. Otherwise we run the occ command one time later at the end of installation script to set system cron.

Maybe command line installation failed. But you should have seen an additional error, as we do not suppress them.
- Check command: `ncc`, if it shows just `occ maintenance:install` as option or full list of commands (which would mean it is actually installed!), not that we installed `ncc` as a shortcut for `sudo -u www-data php /var/www/nextcloud/occ` with v159 😉.
- Otherwise try manually run installation commands (including generation of temporary MySQL admin user to allow www-data access:
```
mysql -uroot -pdietpi -e "grant all privileges on *.* to 'tmp_root'@'localhost' identified by 'dietpi' with grant option"
ncc maintenance:install --no-interaction --database "mysql" --database-name "nextcloud" --database-user "tmp_root" --database-pass "dietpi" --admin-user "admin" --admin-pass "dietpi" --data-dir "/mnt/dietpi_userdata/nextcloud_data"
mysql -uroot -pdietpi -e "drop user 'tmp_root'@'localhost'"
```

guny74, 2017-12-18:
Hello,

I had tried to install it with the dietpi-software and the message occurs. So I had no deeper look, if it would be installed. 
Checking ncc says : Installed !!! - How can I check if all elements/configuration steps necessary is installed ?

Nextcloud is  not accessible via browser...here I installed "lighthttp"

Should I uninstall it and install it with the manueal command ?

Regards

MichaIng, 2017-12-18:
@guny74
Hmm, does dietpi-software show Nextcloud as installed (listed as uninstallable)? In that case all the install steps should be gone through.

Could you post content of your /var/www/nextcloud/config/config.php?
In case error logs of mysql + lighttpd could be helpful: /var/log/...
and output of journalctl -p err.

We do no special configs to lighttpd, as it runs out o the box on our tests. I will invetigate again, and also see if there is a way to suppress the non installed occ message, on occ install command in case.

MichaIng, 2017-12-19:
@guny74 
- I further investigated: Yes the message always shows up, if you use occ without installed Nextcloud, even if you run occ for installation. A bid confusing indeed 😉. I added an issue to their guthub: https://github.com/nextcloud/server/issues/7563
- Do you use MySQL or MariaDB? I just found a problem on Jessie + MariaDB when using mysql command with password option: It blocks access. I will provide solution, if this is indeed your case.

guny74, 2017-12-19:
Hello,

I use mysql.  I worked on the configuration (add USB Stick for Data and changed ist in the config.php).
After the first success "Login Page", the failure "UI seems to be destroyed"
[image]

Known issue of the installation or issue of nextcloud ?

Thanks for help

MichaIng, 2017-12-19:
@guny74 
Hmm, looks like some kind of caching issue.

> add USB Stick for Data and changed ist in the config.php

Note, that this might cause problems, read here how to securely change your Nextcloud data directory after installation: https://help.nextcloud.com/t/howto-change-move-data-directory-after-installation/17170

But it is a good point, we do not yet support:
- I will add an option within dietpi.txt to change location of ownCloud/Nextcloud data directory pre installation, same as it allows to adjust admin user name (also ownCloud hint missing here): https://github.com/Fourdee/DietPi/blob/master/dietpi.txt#L211
- Self note: Add notes about that to Nextcloud/ownCloud forum pages.

MichaIng, 2017-12-20:
@guny74 
Added the possibility to set ownCloud/Nextcloud data directory before installation. That way possible issues with moving it afterwards can be avoided: https://github.com/Fourdee/DietPi/pull/1319

This will be introduced with v160 and announced on the forum, e.g. http://dietpi.com/phpbb/viewtopic.php?f=9&t=2564

Could you solve your issue in between? As generally the page loads, did you try to play around with theming and different browsers? Otherwise I would recommend a new installation (as you want to start, stay with v159), do NOT change the data directory inside config.php, instead follow "`Solution 2`" of my howto on the nextcloud forum: https://help.nextcloud.com/t/howto-change-move-data-directory-after-installation/17170
The default data directory is `/mnt/dietpi_userdata/nextcloud_data` so move this folder/content to your desired location and replace it by symlink. That way the database entries and stuff like within Nextcloud stay correct.

guny74, 2017-12-21:
Hello,

I could not solve the problem, I tried other Browser, theming....
I did an unsinstall and I will do an new Installation maybe with  V160...or with your guide to change data directory. 

Thank you for your help and development

regards
  Thorsten

guny74, 2017-12-21:
BTW: The first prio is my heating problem of the BPI m2+ ....

MichaIng, 2017-12-22:
@guny74 

Maybe this is somehow related to **BPI m2+**, as I can't reproduce it on VM.

@Fourdee 

Do you have some **BPI m2+** to test Jessie+default Lighttpd stack+Nextcloud? In case manual changing datadir by editing `nextcloud/config/config.php`?

k-plan, 2017-12-23:
@MichaIng 

> Do you have some BPI m2+ to test Jessie+default Lighttpd stack+Nextcloud?

Sorry, don't own a BananaPi M2+

> BananaPi M2+
> CPU:Allwinner H3 1.2 GHz quad core
> RAM:1GB DDR3

But have a OrangePi One (same CPU):

> OrangePi One
> CPU:Allwinner H3 1.2 GHz quad core
> RAM:512MB DDR3

As mentioned [here](https://github.com/Fourdee/DietPi/issues/1281#issuecomment-350553709) I like to give you this for free, if you sent me your contact details.

> ... to test Jessie+default Lighttpd stack+Nextcloud

Yes, I can set this up as clean manual install on OrangePi One DietPi v159.

Or do you like to do this with automated installation with v160 (testing) and:

```
#Optional username for ownCloud/Nextcloud admin account, default is 'admin'. Applied during installation.
dietpi_ownnextcloud_username=admin

#Optional data directory for ownCloud, default is '/mnt/dietpi_userdata/owncloud_data'. Applied during installation.
#	This option is for advanced users. For full compatibility, please keep this options defaults, and, use dietpi-drive_manager to move the DietPi user data location.
dietpi_owncloud_datadir=/mnt/dietpi_userdata/owncloud_data

#Optional data directory for Nextcloud, default is '/mnt/dietpi_userdata/nextcloud_data'. Applied during installation.
#	This option is for advanced users. For full compatibility, please keep this options defaults, and, use dietpi-drive_manager to move the DietPi user data location.
dietpi_nextcloud_datadir=/mnt/dietpi_userdata/nextcloud_data

```

Please let me know. Thanks.

MichaIng, 2017-12-31:
@k-plan 
If you find time, a test on clean v159 would be nice, if it works a quick test by adjusting config.php also, just to see if the issue is the same. The little message in the middle of @guny74 's picture points already into the direction: "Sie haben keine Berechtigung, hier Dateien hochzuladen oder zu erstellen", which shows up if `chown www-data:www-data <datadir>` is missing, e.g. also on single unchowned folders inside <datadir>. I would have expected an error on earlier stage, about missing `.ocdata` file inside, but maybe guny74 did manually create it.

About v160/6.0 Fourdee made already a successful quick test, so generally the procedure works.

k-plan, 2018-01-01:
@MichaIng 

> a test on clean v159 would be nice

fresh image, all settings are default v159 (` LLMP: lighttpd | mysql   | php` ) before starting nextcloud installation :

```
....
Setting up mysql-server-5.5 (5.5.58-0+deb8u1) ...
180101 14:13:08 [Warning] Using unique option prefix key_buffer instead of key_buffer_size is deprecated and will be removed in a future release. Please use the full name instead.

...

Setting up redis-server (2:2.8.17-1+deb8u5) ...
adduser: Warning: The home directory `/var/lib/redis' does not belong to the user you are currently creating.

...

 [Ok] stop : cron
 [Ok] stop : lighttpd
 [Ok] stop : php5-fpm
 [Ok] stop : mysql
 [Ok] stop : redis-server

...

2018-01-01 14:20:02 (1.99 MB/s) - ‘/var/www/apc.php’ saved [38125/38125]

Nextcloud is not installed - only a limited number of commands are available
Nextcloud was successfully installed
Set mode for background jobs to 'cron'
Maintenance mode enabled
```

After auto-reboot open nextcloud web interface (this take a long time, but finally it works)

[image]

```
root@OrangePi-One:~# cat /var/www/nextcloud/config/config.php
<?php
$CONFIG = array (
  'passwordsalt' => 'YsDCRyLmR4JK8N1zmODM7L7GsvXH0J',
  'secret' => 'sxX0KNbGp0obbPm0gSMnGAnN8Xx+kYudoZxpI5XsNJ7/5g/V',
  'trusted_domains' =>
  array (
    0 => 'localhost',
    1 => '*',
  ),
  'datadirectory' => '/mnt/dietpi_userdata/nextcloud_data',
  'overwrite.cli.url' => 'http://localhost/nextcloud',
  'dbtype' => 'mysql',
  'version' => '12.0.4.3',
  'memcache.local' => '\\OC\\Memcache\\APCu',
  'filelocking.enabled' => true,
  'memcache.locking' => '\\OC\\Memcache\\Redis',
  'redis' =>
  array (
    'host' => '/var/run/redis/redis.sock',
    'port' => 0,
  ),
  'dbname' => 'nextcloud',
  'dbhost' => 'localhost',
  'dbport' => '',
  'dbtableprefix' => 'oc_',
  'mysql.utf8mb4' => true,
  'dbuser' => 'oc_admin',
  'dbpassword' => 'xQ48cNI7acYiZggG4yYsanM7s4WPtF',
  'installed' => true,
  'instanceid' => 'oc87gvmn1agt',
  'maintenance' => false,
);
```

```
root@OrangePi-One:~# ls -lah /var/www/nextcloud/data/
total 12K
drwxrwxr-x  2 www-data www-data 4.0K Jan  1 14:20 .
drwxrwxr-x 15 www-data www-data 4.0K Jan  1 14:21 ..
-rwxrwxr-x  1 www-data www-data  324 Jan  1 14:20 .htaccess
-rwxrwxr-x  1 www-data www-data    0 Jan  1 14:20 index.html
```

k-plan, 2018-01-01:
```
root@OrangePi-One:~# lsblk -f
NAME        FSTYPE LABEL UUID                                 MOUNTPOINT
sda
└─sda1      vfat   USB   56D0-EBBC
```

```
root@OrangePi-One:~# mkdir /mnt/usb-stick
root@OrangePi-One:~# mount /dev/sda1 /mnt/usb-stick/
```

```
root@OrangePi-One:~# ls -lah /mnt/usb-stick/
total 12K
drwxr-xr-x 3 root root 4.0K Jan  1 16:32 .
drwxr-xr-x 7 root root 4.0K Jan  1 16:30 ..
drwxr-xr-x 2 root root 4.0K Jan  1 16:32 nextcloud_data

```

```
root@OrangePi-One:~# nano /var/www/nextcloud/config/config.php

root@OrangePi-One:~# cat /var/www/nextcloud/config/config.php | grep datadirectory
  'datadirectory' => '/mnt/usb-stick/nextcloud_data',
```
  
  
  
  
```
root@OrangePi-One:~# dietpi-services restart

Please wait...

Your data directory is invalid
Ensure there is a file called ".ocdata" in the root of the data directory.

Your data directory is not writable
Permissions can usually be fixed by giving the webserver write access to the root directory. See https://docs.nextcloud.com/server/12/go.php?to=admin-dir_permissions.

An unhandled exception has been thrown:
exception 'Exception' with message 'Environment not properly prepared.' in /var/www/nextcloud/lib/private/Console/Application.php:145
Stack trace:
#0 /var/www/nextcloud/console.php(99): OC\Console\Application->loadCommands(Object(Symfony\Component\Console\Input\ArgvInput), Object(Symfony\Component\Console\Output\ConsoleOutput))
#1 /var/www/nextcloud/occ(11): require_once('/var/www/nextcl...')
```

k-plan, 2018-01-01:
Testing:

- uninstall and reinstall `nextcloud` via `dietpi-software`
- after reboot testing new nextcloud installation => works
- attached USB stick and move `User Data Location`  over to USB via `dietpi-software`

[image]

```
root@OrangePi-One:~# dietpi-services status

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: status
 Please wait...

 [Ok] cron      active (running) since Mon 2018-01-01 17:12:32 GMT; 1min 44s ago
 [Ok] lighttpd  active (running) since Mon 2018-01-01 17:12:33 GMT; 1min 44s ago
 [Ok] php5-fpm  active (running) since Mon 2018-01-01 17:12:34 GMT; 1min 42s ago
 [Failed] mysql failed (Result: exit-code) since Mon 2018-01-01 17:13:07 GMT; 1min 10s ago
● mysql.service - LSB: Start and stop the mysql database server daemon
   Loaded: loaded (/etc/init.d/mysql)
   Active: failed (Result: exit-code) since Mon 2018-01-01 17:13:07 GMT; 1min 10s ago
  Process: 4554 ExecStart=/etc/init.d/mysql start (code=exited, status=1/FAILURE)

Jan 01 17:13:07 OrangePi-One mysql[4554]: Starting MySQL database server: mysqld . . . . . . . . . . . . . . . . . . . . . . . . . . . . . . failed!
Jan 01 17:13:07 OrangePi-One systemd[1]: mysql.service: control process exited, code=exited status=1
Jan 01 17:13:07 OrangePi-One systemd[1]: Failed to start LSB: Start and stop the mysql database server daemon.
Jan 01 17:13:07 OrangePi-One systemd[1]: Unit mysql.service entered failed state.
 [Ok] redis-server      active (running) since Mon 2018-01-01 17:13:07 GMT; 1min 9s ago
```

```
An unhandled exception has been thrown:
exception 'Doctrine\DBAL\DBALException' with message 'Failed to connect to the database: An exception occured in driver: SQLSTATE[HY000] [2002] Can't connect to local MySQL server through socket '/var/run/mysqld/mysqld.sock' (2)' in /var/www/nextcloud/lib/private/DB/Connection.php:61
Stack trace:
#0 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(429): OC\DB\Connection->connect()
#1 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(389): Doctrine\DBAL\Connection->getDatabasePlatformVersion()
#2 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(328): Doctrine\DBAL\Connection->detectDatabasePlatform()
#3 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(623): Doctrine\DBAL\Connection->getDatabasePlatform()
#4 /var/www/nextcloud/lib/private/DB/Connection.php(148): Doctrine\DBAL\Connection->setTransactionIsolation(2)
#5 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/DriverManager.php(172): OC\DB\Connection->__construct(Array, Object(Doctrine\DBAL\Driver\PDOMySql\Driver), Object(Doctrine\DBAL\Configuration), Object(Doctrine\Common\EventManager))
#6 /var/www/nextcloud/lib/private/DB/ConnectionFactory.php(151): Doctrine\DBAL\DriverManager::getConnection(Array, Object(Doctrine\DBAL\Configuration), Object(Doctrine\Common\EventManager))
#7 /var/www/nextcloud/lib/private/Server.php(588): OC\DB\ConnectionFactory->getConnection('mysql', Array)
#8 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(113): OC\Server->OC\{closure}(Object(OC\Server))
#9 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(115): Pimple\Container->offsetGet('OCP\\IDBConnecti...')
#10 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('OCP\\IDBConnecti...')
#11 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(163): OC\ServerContainer->query('OCP\\IDBConnecti...')
#12 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(109): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}(Object(OC\Server))
#13 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(115): Pimple\Container->offsetGet('DatabaseConnect...')
#14 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('DatabaseConnect...')
#15 /var/www/nextcloud/lib/private/Server.php(1345): OC\ServerContainer->query('DatabaseConnect...')
#16 /var/www/nextcloud/lib/private/Server.php(309): OC\Server->getDatabaseConnection()
#17 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(113): OC\Server->OC\{closure}(Object(OC\Server))
#18 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(115): Pimple\Container->offsetGet('OC\\Authenticati...')
#19 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('OC\\Authenticati...')
#20 /var/www/nextcloud/lib/private/Server.php(313): OC\ServerContainer->query('OC\\Authenticati...')
#21 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(113): OC\Server->OC\{closure}(Object(OC\Server))
#22 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(115): Pimple\Container->offsetGet('OC\\Authenticati...')
#23 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('OC\\Authenticati...')
#24 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(163): OC\ServerContainer->query('OC\\Authenticati...')
#25 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(109): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}(Object(OC\Server))
#26 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(115): Pimple\Container->offsetGet('OC\\Authenticati...')
[... cut ...]
#63 /var/www/nextcloud/lib/private/Server.php(1274): OC\ServerContainer->query('L10NFactory')
#64 /var/www/nextcloud/lib/private/Server.php(1285): OC\Server->getL10NFactory()
#65 /var/www/nextcloud/lib/base.php(225): OC\Server->getL10N('lib')
#66 /var/www/nextcloud/lib/base.php(678): OC::checkConfig()
#67 /var/www/nextcloud/lib/base.php(1086): OC::init()
#68 /var/www/nextcloud/console.php(56): require_once('/var/www/nextcl...')
#69 /var/www/nextcloud/occ(11): require_once('/var/www/nextcl...')
```

MichaIng, 2018-01-01:
@k-plan 
Many thanks for testing: As I thought there is an error about missing `.ocdata` file after just editing `config.php`. I guess @guny74 already had ncdata at the stick, so `.ocdata` was already there, or created it manually. Would be another test, but not necessary 😉.

> uninstall and reinstall nextcloud via dietpi-software

Did you adjust `dietpi.txt` then?

About mysql error: Seems we do not update the symlink of mysql folder (correctly), but services were stopped in between, right?

k-plan, 2018-01-01:
@MichaIng 

> Did you adjust dietpi.txt then?

No, simply did, what user will/can do. Open `dietpi-software` and change `User Data Location`.

> services were stopped in between

Yes, services will be stop automatically while `dietpi-drive_manager` is moving data in the background.

> symlink of mysql folder (correctly)

have to look on it later, because at the moment your `GIT_BRANCH='741035212a314ae58534cab62fcdb591408ccf4a'` testing is running on Odroid HC1. 😄 

Will start OrangePi One up, if Odroid HC1 has finished.

k-plan, 2018-01-01:
@MichaIng 

```
root@OrangePi-One:~# ls -l /mnt/
total 20
drwxr-xr-x 4 root   root   4096 Jan  1  1970 56D0-EBBC
lrwxrwxrwx 1 dietpi dietpi   30 Jan  1 17:12 dietpi_userdata -> /mnt/56D0-EBBC/dietpi_userdata
drwxr-xr-x 2 root   root   4096 Dec  7 21:23 ftp_client
drwxr-xr-x 2 root   root   4096 Jan 27  2017 nfs_client
drwxr-xr-x 2 root   root   4096 Dec  7 21:23 samba
```

```
root@OrangePi-One:~# ls -l /mnt/dietpi_userdata/
total 24
drwxr-xr-x 2 root root 4096 Jan  1 17:11 downloads
drwxr-xr-x 2 root root 4096 Jan  1 17:11 Music
drwxr-xr-x 5 root root 4096 Jan  1 17:43 mysql
drwxr-xr-x 6 root root 4096 Jan  1 17:11 nextcloud_data
drwxr-xr-x 2 root root 4096 Jan  1 17:11 Pictures
drwxr-xr-x 2 root root 4096 Jan  1 17:11 Video
```

```
root@OrangePi-One:~# ls -l /mnt/dietpi_userdata/mysql/
total 28700
-rwxr-xr-x 1 root root        0 Jan  1 17:11 debian-5.5.flag
-rwxr-xr-x 1 root root 18874368 Jan  1 17:11 ibdata1
-rwxr-xr-x 1 root root  5242880 Jan  1 17:11 ib_logfile0
-rwxr-xr-x 1 root root  5242880 Jan  1 17:11 ib_logfile1
drwxr-xr-x 2 root root     8192 Jan  1 17:11 mysql
-rwxr-xr-x 1 root root        6 Jan  1 17:11 mysql_upgrade_info
drwxr-xr-x 2 root root    12288 Jan  1 17:11 nextcloud
drwxr-xr-x 2 root root     4096 Jan  1 17:11 performance_schema
```

```
root@OrangePi-One:~# ls -l /mnt/dietpi_userdata/nextcloud_data/
total 104
drwxr-xr-x 5 root root  4096 Jan  1 17:11 admin
drwxr-xr-x 8 root root  4096 Jan  1 17:11 appdata_oc87gvmn1agt
drwxr-xr-x 7 root root  4096 Jan  1 17:11 appdata_ocgotkbd55ar
drwxr-xr-x 2 root root  4096 Jan  1 17:11 files_external
-rwxr-xr-x 1 root root     0 Jan  1 17:11 index.html
-rwxr-xr-x 1 root root 87249 Jan  1 17:11 mysql_backup.sql
-rwxr-xr-x 1 root root     0 Jan  1 17:11 nextcloud.log

```

Do you need something else?

MichaIng, 2018-01-01:
@k-plan 
I will test a bid by myself, thanks so far.

I forgot, that /var/lib/mysql (where MySQL/MariaDB) looks for it's database files, links to /mnt/dietpi_userdata/mysql, thus after userdata move, no other links need to be adjusted.

Have to investigate then, why mysqld fails to start up afterwards.

Fourdee, 2018-01-19:
@MichaIng 

> €: Just tested on v6.0 to move datadir just into a new folder /mnt/userdata with Nextcloud + LLAP stack on VM Stretch, and didn't face any issue. Maybe have to test with USB stick again.

Tested fine here:
- XU4 jessie v6.0 + Owncloud + userdata on rootFS
- 🈯️ Moved userdata to USB drive
- 🈯️ Moved userdata back to rootFS

MichaIng, 2018-01-22:
Mark as closed. Issue is high likely related to unsupported manual (not via DietPi) moving of Nextcloud datadir post installation. Ability to move datadir pre installation implemented with v6.0.

closed 2018-01-22 by MichaIng

BristolPiggy, 2022-07-14:
Out of interest, what was the solution to this situation? As 5 years later in 2022 I am experiencing the same situation at gyny74 and am finding it hard to understand what the resolution to the issue was.

Joulinar, 2022-07-14:
maybe better to open a new own issue and to fill the respective bug report (including full error message)