MichaIng/DietPi #1315: mpd old version in dietpi v159

tatoosh, 2017-12-18:
I updates my virtualbox and rpi images to dietpi v159.
mpd still on version 0.19.1
>mpd --version
>Music Player Daemon 0.19.1

Changelog fromhttps://github.com/Fourdee/DietPi/blob/master/CHANGELOG.txt says:
**DietPi-Software | MPD (Stretch): Install updated to 0.20.11. **

k-plan, 2017-12-18:
@tatoosh 

> I updates my virtualbox and rpi images to dietpi v159

Can you please provide terminal output from RPi and VM:

`cat /etc/debian_version` 

Thanks

tatoosh, 2017-12-18:
>cat /etc/debian_version
>8.10

k-plan, 2017-12-18:
@tatoosh 

what's easy:

```
root@RPi-Zero_W-Stretch:~# cat /etc/debian_version
9.1
```

```
root@NanoPi-Neo-Jessie:~# cat /etc/debian_version
8.10
```

Debian Version 8.x   ==>>` Jessie`

Debian Version 9.x ==>> `Stretch`

You have `DietPi_VirtualBox-x86_64-(Jessie)` and `DietPi_RPi-armv6-(Jessie)`

> DietPi-Software | MPD **(Stretch)**: Install updated to 0.20.1

```
root@NanoPi-Neo-Jessie:~# apt-get -s install mpd
...
mpd (0.19.1-1.1 Debian:8.10/oldstable [armhf])
```

You need Stretch:

[DietPi_RPi-armv6-(Stretch)](http://dietpi.com/downloads/images/DietPi_RPi-armv6-(Stretch).7z)

[DietPi_VirtualBox-x86_64-(Stretch)](DietPi_VirtualBox-x86_64-(Stretch).7z)

tatoosh, 2017-12-18:
Thanks @k-plan.
I thought dietpi-update is enough. Should i habe to change image or is there a dist-upgrade option?

k-plan, 2017-12-18:
@tatoosh 

> Should i habe to change image ...

For my part, I do this. It's quick and easy (thanks to `dietpi.txt`) and it works 💯 

> .. or is there a dist-upgrade option?

Simple `apt-get dist-upgrade` can't work.

But at DietPi forum are some people, who have tried this: http://dietpi.com/phpbb/viewtopic.php?f=11&t=2566

I personally do not or will not try it out.
If this fail, a `dietpi-backup` will not help you out.

tatoosh, 2017-12-18:
@k-plan 
ok i think my error it solved by using new images.
i will also look at dietpi-backup to backup all my files.

closed 2017-12-18 by tatoosh