MichaIng/DietPi #1323: Loosing WIFI by installing software

tatoosh, 2017-12-19:
#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```? - 159
- SBC Device (EG: RPi 3)? - RPi Zero
- Power supply used (EG: 5v 1A RAVpower)? - 2A Powerbank
- SD card used (EG: Sandisk ultra)? - Sandisk ultra 32GB
- Distro (EG: Jessie) | ```uname -a```?

#### Can this issue be replicated on a fresh installation of DietPi?
I am using two WLAN SSIDs in wpa_suppicant and by installing software with dietpi-software my ssh connection will be dropped.

tatoosh, 2017-12-19:
after long waiting the session comes back!

closed 2017-12-19 by tatoosh

Fourdee, 2017-12-20:
@tatoosh 

Fixed for v160, however, existing users will need to disable the service control, before updating:
https://github.com/Fourdee/DietPi/issues/1288#issuecomment-351613165

tatoosh, 2017-12-20:
Great, thx for issue linking @Fourdee