MichaIng/DietPi #1324: DietPi-Software | MySQL/MariaDB: mysqld fails after update to v160

MichaIng, 2017-12-19:
Seems to be related to core environment creation:
`[ERROR] mysqld: Can't create/write to file '/var/lib/mysql/aria_log_control' (Errcode: 13 "Permission denied")`

`chmod +x /mnt/dietpi_userdata` solved the problem for me. We should verify that the folder was executable before, so we do not change something after all, but I guess, as this seems to be necessary for mysqld to access it's mysql subfolder.

MichaIng, 2017-12-20:
Jep `dietpi_userdata`, it is made executable during dietpi-software execution: https://github.com/Fourdee/DietPi/blob/master/dietpi/dietpi-software#L8599

Fourdee, 2017-12-20:
@MichaIng 
Thanks 👍 

I'll set +x during set_env aswell

Fourdee, 2017-12-20:
@MichaIng 

https://github.com/Fourdee/DietPi/commit/39fb3fdc29da46f93c09ce03522a4134ccebe8f7 should do it, but if we can verify with another test, i'd be grateful.

I've not been able to replicate it since with testing https://github.com/Fourdee/DietPi/pull/1319#pullrequestreview-84664248, although thats mariaDB default install.

MichaIng, 2017-12-20:
@Fourdee 
Jep looks good, will test later, have to rewind first my VM to v159.

MichaIng, 2017-12-20:
Works like a charm, tested on VM Stretch with LAAP stack. 👍

closed 2017-12-20 by MichaIng