MichaIng/DietPi #1327: dietpi shutdown/reboot very slow

tatoosh, 2017-12-20:
Updates to strecth dietpi 159 my Rpi3.
Booting is very fast but a reboot tooks very long.
Screen says "Dietpi Ramdisk shutting down". 

What can i do to improve that?

Fourdee, 2017-12-20:
@tatoosh 

Should not take longer than 1 second to save RAMdisk to disk. We'll try to replicate.

Which SD card/PSU are you using?

tatoosh, 2017-12-20:
I am using this sd-card https://www.amazon.de/gp/product/B073S9SFK2/ref=oh_aui_detailpage_o01_s00?ie=UTF8&psc=1

tatoosh, 2017-12-20:
I could not reproduce it on my other Rpi3. at the moment ...

tatoosh, 2017-12-20:
Reinstalled everything, no problem. 
Justed change from 16GB SanDisk to 32GB Sandisk Card.

tatoosh, 2017-12-20:
Oh after some reboots the error comes back :(

tatoosh, 2017-12-20:
Also ```service mpd stop``` is delayed. I think i have to kill the process !?

tatoosh, 2017-12-21:
Issue solved

Problem was my change of /etc/mpd.conf for the playlist folder. There was a file system permission issue. 
But u can not see this in mod.log :(

closed 2017-12-21 by tatoosh