MichaIng/DietPi #1328: DietPi-Software | Universal Media Server

LexiconCode, 2017-12-20:
[Universal Media Server](http://www.universalmediaserver.com/) is an open source DLNA-compliant UPnP Media Server. Minimal configuration for more than 50 different devices.

Universal Media Server is similar to Plex and Emby except the premium features that you pay for in the latter two programs are free in Universal Media Server. Granted it doesn't have cloud storage services..who cares?

WolfganP, 2017-12-20:
I can't speak for Emby, but I moved from UMS to Plex a while ago and not planning to return. 
UMS dependency in Java kills any installation in small devices like Raspberry, and Plex DLNA services are included for free as in UMS and allow for custom profiles, so no advantage in there either. 
Besides, the development pace of Plex is really high, and the ability to add many active developed plugins to the server is also a plus for me.

LexiconCode, 2017-12-20:
Do pay for DVR using plex? @WolfganP

WolfganP, 2017-12-20:
Not a DVR user, just using my media server to distribute media inside my LAN with many diff clients (from android boxes to PCs to smartTVs), which I want to offer the richest experience.
Not using DVR features, and if I decide to go there, most probably will use a targeted project like https://github.com/tvheadend/tvheadend and look for integrations with my media server (via native hookup or plugin alternative)

Fourdee, 2017-12-30:
General Cleanup:
Marking all software requests as closed. Once interest peaks, we'll reopen once work starts.

closed 2017-12-30 by Fourdee

echable9999, 2019-11-24:
Support UMS on Odroid.