MichaIng/DietPi #1329: DietPi-System | Apt upgrade fails

LexiconCode, 2017-12-20:
#### Required Information:
- DietPi Version | 159
- SBC Device (EG: RPi 3) Odroid C2
- Power supply used (EG: 5v 1A RAVpower)? official odroid power adapter
- SD card used (EG: Sandisk ultra)? scan disk ultra
- Distro (EG: Jessie) | Jessie

#### Additional Information (if applicable):
- Software title? not applicable
- Can this issue be replicated on a fresh installation of DietPi? yes

Council output recommended
```
 [Info] Please try:
 [Info] - Fixing apt package errors with 'apt-get install -f'
 [Info] - Change apt mirror: 'dietpi-config' > 'Networking: NAS/MISC' > 'Apt Mirror'
 [Info] - Clear and refresh apt 'apt-get update'
 [Info] - If problems persist, please create a Git ticket so DietPi can investigate the issue:
 - https://github.com/Fourdee/DietPi/issues
```
That did not fix the issue

**Recommendations**
manually executing "sudo apt-get dist-upgrade" seems to be a fix.

#### Steps to reproduce:

Try 'apt-get upgrade' on Odroid C2 fresh install. libc6 package fails 

**output of 'apt-get upgrade'**
`root@DietPi:~# apt-get upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  ca-certificates curl debconf debconf-utils debian-archive-keyring dropbear gpgv initramfs-tools libc6 libcurl3 libcurl3-gnutls
  libdb5.3 libdns-export100 libexpat1 libffi6 libgcrypt20 libgnutls-deb0-28 libgnutls-openssl27 libgssapi-krb5-2 libicu52
  libirs-export91 libisc-export95 libisccfg-export90 libk5crypto3 libkrb5-3 libkrb5support0 libldap-2.4-2 libncurses5 libncursesw5
  libpam-modules libpam-modules-bin libpam0g librtmp1 libssl1.0.0 libsystemd0 libtasn1-6 libtinfo5 libudev1 linux-image-3.14.79+
  locales multiarch-support openssl passwd setup-odroid sudo systemd systemd-sysv tzdata udev unzip wget wpasupplicant
52 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/42.5 MB of archives.
After this operation, 50.2 kB disk space will be freed.
Do you want to continue? [Y/n] y
Extracting templates from packages: 100%
Preconfiguring packages ...
(Reading database ... 20588 files and directories currently installed.)
Preparing to unpack .../libc6_2.19-18+deb8u10_arm64.deb ...

LD_LIBRARY_PATH contains the traditional /lib directory,
but not the multiarch directory /lib/aarch64-linux-gnu.
It is not safe to upgrade the C library in this situation;
please remove the /lib/directory from LD_LIBRARY_PATH and
try again.

dpkg: error processing archive /var/cache/apt/archives/libc6_2.19-18+deb8u10_arm64.deb (--unpack):
 subprocess new pre-installation script returned error exit status 1
Errors were encountered while processing:
 /var/cache/apt/archives/libc6_2.19-18+deb8u10_arm64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

Fourdee, 2017-12-21:
@LexiconCode 

Thanks for the report 👍 i'll investigate.

Fourdee, 2017-12-21:
Additional reports from forums:
http://dietpi.com/phpbb/viewtopic.php?f=11&t=2603

Fourdee, 2017-12-21:
Notes:
- check default ```LD_LIBRARY_PATH```
- ```LD_LIBRARY_PATH=$LD_LIBRARY_PATH:xxxx:xxxx```?

MichaIng, 2017-12-22:
Oh, this error showed up after I upgraded via preparation script from Stretch to Buster. Thought it would have something to do with this.

I solved it by `unset LD_LIBRARY_PATH`.

k-plan, 2017-12-22:
```
echo $LD_LIBRARY_PATH

export LD_LIBRARY_PATH=/lib/arm-linux-gnueabihf:/lib:/usr/lib:/usr/local/lib:/opt/vc/lib

ldconfig -v
```

Fourdee, 2017-12-22:
@k-plan @MichaIng 

Thanks 👍 

Due to following commit: https://github.com/Fourdee/DietPi/commit/fd4610589f9aeaab3d23a8e45285c906f01da600

I'll tweak it.

Fourdee, 2017-12-22:
Completed.

closed 2017-12-22 by Fourdee