MichaIng/DietPi #1330: DietPi-Boot | "[FAILED] Failed to start Daily apt download activities."

MichaIng, 2017-12-22:
See this error on every VirtualBox VM (Jessie+Stretch+Buster) and if I remember right also on my RPi:
`[FAILED] Failed to start Daily apt download activities.`

I remember a commit where we disabled daily apt update, but maybe we can somehow disable the error message?
________________________
Different error on reboot/shutdown, one of the last messages:
`Failed to start Shutdown all ssh sessions before network.`
________________________
Both minor, but just to have it mentioned somwhere

DarkElvenAngel, 2017-12-22:
@MichaIng  The file /etc/systemd/system/apt-daily.service is a link that points to /dev/null this is what is giving you the error.

I deleted the link and replace the apt-daily.service with this
```Description=Apt-daily

[Service]
Type=simple
User=root
ExecStart=/bin/false

[Install]
WantedBy=multi-user.target
```

That will get rid of the error on the bootup.

Fourdee, 2017-12-22:
Reason for mask:
- the service instigates a apt-cache/dpkg lock during updates, which will break ```dietpi-software``` and our apt installations.
- DietPi runs G_AGUP as needed.

MichaIng, 2017-12-22:
@DarkElvenAngel 

Thanks for solution. Simple and clean by dummy service file.

Just for interest, I will try to have a look where the error is produced, maybe we can stop the execution/error msg at it's root.

DarkElvenAngel, 2017-12-22:
@MichaIng 

I'm not sure I already tried finding the source. I was looking for symlinks to apt-daily.system but I may have already deleted them. but they would be in /etc/systemd/ somewhere if they exists otherwise some other part of the boot process is trying to start the service in another way. 

Good luck

MichaIng, 2017-12-27:
@DarkElvenAngel 
Replacing `/etc/systemd/system/apt-daily.service` by the dummy service as mentioned above just worked fine, error is gone 🙂!

MichaIng, 2017-12-27:
About the ssd stop error:
Found the commit, that was meant to stop error messages: https://github.com/Fourdee/DietPi/commit/0e9806dff2628a41e652a9dc7d11a618b8142aa9#diff-e9e0c6c64b4937739e13ffac58f4a888R290
However systemd seems to have it's own error handling that does not rely on command outputs. 

I trief `ExecStart=/bin/bash -c '( ps -A | grep sshd ) && killall sshd; ( ps -A | grep dropbear ) && killall dropbear'` to do killall only, if process does exist, but the error is just the same. Maybe something else with the service is not right, I am not too deep in this, or simply this commands cannot be run/are not available anymore on shutdown?

@Fourdee 
Maybe you have an idea how to suppress this?

Fourdee, 2017-12-27:
@MichaIng 

> Maybe you have an idea how to suppress this?

Not sure, probably needs a full bash script that we can control, check for process/installed service and control it that way. Or ```systemctl stop dropbear``` in the service, or something similar.

I'd need to take a stab at it.

Fourdee, 2017-12-28:
@MichaIng 

Just a quick look over this:
- Dropbear is a sysinit service ```/etc/init.d/dropbear```. We could probably create our own SystemD one with the additional ```killall``` command to kill SSH connections during shutdown. 
-- This all being that OpenSSH no longer requires the the additional ```killall```, nope, it still requires the custom killall.

@MichaIng 
I think a bash script to cover both killall of Dropbear and OpenSSH, that we can control error output is the solution here?

MichaIng, 2017-12-28:
@Fourdee 
Works on my VMs on Jessie+Stretch+Buster 👍.

@apt-daily.service
- There are three other services: `apt-daily.timer`, `apt-daily-upgrade.service`, `apt-daily-upgrade.timer` and the last two wait for the first and our masked service to start. I guess this the reason, why systemd wants to start them/see them start and even knows the masked services description text.
- So simple solution is to mask all 4 services, as apt upgrade anyway does not make any sense without apt update ;).

MichaIng, 2017-12-28:
Fixed and merged changelog: https://github.com/Fourdee/DietPi/pull/1348

Fourdee, 2017-12-30:
@MichaIng 

> Fixed and merged changelog: #1348

Thanks 👍 

I'll mark this as closed.

closed 2017-12-30 by Fourdee