MichaIng/DietPi #1333: no space,how to use all available disk space?

xnathanh, 2017-12-22:
i have 64gb sdcard,but i only have <10gb,
```
root@DietPi:~# df -lh
Filesystem      Size  Used Avail Use% Mounted on
udev            2.0G     0  2.0G   0% /dev
tmpfs           393M   11M  383M   3% /run
/dev/mmcblk1p7  476M  452M     0 100% /
tmpfs           2.0G     0  2.0G   0% /dev/shm
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           2.0G     0  2.0G   0% /sys/fs/cgroup
tmpfs           2.0G  4.0K  2.0G   1% /tmp
tmpfs            10M  1.5M  8.6M  15% /DietPi
tmpfs            20M   16K   20M   1% /var/log
/dev/mmcblk1p6  100M   42M   59M  42% /boot/efi
tmpfs           393M     0  393M   0% /run/user/0
```

Fourdee, 2017-12-22:
@xnathanh 

Thanks for the report 👍 

Looks like the rootFS re-partition/expansion has failed. Which SBC device please?

MichaIng, 2017-12-22:
@xnathanh 

Ui, jep partitioning went totally wrong. Just to sort issues out: You did flash image from dietpi.com just by Etcher onto your SD, which was then formatted to contain 2 partitions?

xnathanh, 2017-12-22:
@Fourdee is rock64 4g with 64gb sd card(SanDisk SDSQUSC-064G-APCIA Ultra Plus 64GB MicroSDXC UHS-1 Class10)

Fourdee, 2017-12-23:
@xnathanh 

Thanks 👍 

I'll try to replicate on my system.

> which was then formatted to contain 2 partitions?

@MichaIng 

Rock image has 7 partitions, although, will be a good test for our new PREP_SYSTEM 😄

Fourdee, 2017-12-26:
Unable to replicate with current image:
```DietPi_v156_Rock64-arm64-(Stretch).img```

Putting this down to a possible instability issue with SD card and/or PSU used.

I'll mark as closed, please reopen if required.

closed 2017-12-26 by Fourdee