MichaIng/DietPi #1334: Not able to install MOSQUITTO

profra, 2017-12-22:
HW: Raspberry Pi 3
OS: Dietpi V159

I have tried all suggestions from your side... no success. I don't know what to do. Any help from your side?

[image]

[image]

```
root@RPI3A:~# apt-cache search mosquitto
libmosquitto-dev - MQTT version 3.1/3.1.1 client library, development files
libmosquitto1 - MQTT version 3.1/3.1.1 client library
libmosquitto1-dbg - debugging symbols for libmosquitto binaries
libmosquittopp-dev - MQTT version 3.1 client C++ library, development files
libmosquittopp1 - MQTT version 3.1/3.1.1 client C++ library
libmosquittopp1-dbg - debugging symbols for libmosquittopp binaries
mosquitto - MQTT version 3.1/3.1.1 compatible message broker
mosquitto-auth-plugin - Authentication plugin for Mosquitto with multiple back-ends
mosquitto-clients - Mosquitto command line MQTT clients
mosquitto-dbg - debugging symbols for mosquitto binaries
mosquitto-dev - Development files for Mosquitto
root@RPI3A:~# apt-get install mosquitto
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 mosquitto : Depends: libssl1.0.0 (>= 1.0.1) but it is not installable
             Depends: libwebsockets3 (>= 1.2) but it is not installable
E: Unable to correct problems, you have held broken packages.
```

## Creating a bug report/issue:

#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```?
- SBC Device (EG: RPi 3)?
- Power supply used (EG: 5v 1A RAVpower)?
- SD card used (EG: Sandisk ultra)?
- Distro (EG: Jessie) | ```uname -a```?

#### Additional Information (if applicable):
- Software title?
- Can this issue be replicated on a fresh installation of DietPi?

#### Expected behaviour:

#### Actual behaviour:

#### Steps to reproduce:

#### Did you submit a ```dietpi-bugreport```?

#### Extra details:

DarkElvenAngel, 2017-12-23:
Just to help out could you please run ```cat /etc/issue``` and post the output

Fourdee, 2017-12-23:
@profra 

Thanks for the report, this is fixed for v160. 

Please see below for info + fix now:
https://github.com/Fourdee/DietPi/issues/1306#issuecomment-352167350

I'll mark this as closed/dupe. Please reopen if issues persist.

closed 2017-12-23 by Fourdee