MichaIng/DietPi #1336: DietPi-Software | Directory where UrBackup will save temporary files is inaccessible

LexiconCode, 2017-12-25:
#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```? 159
- SBC Device (EG: RPi 3)? Odroid C2
- Power supply used (EG: 5v 1A RAVpower)?
- SD card used (EG: Sandisk ultra)?
- Distro (EG: Jessie) | ```uname -a```? Jessie

#### Additional Information (if applicable):
- Software title?
UrBackup 
- Can this issue be replicated on a fresh installation of DietPi? yes

#### Expected behaviour:

Modifying the folder in the settings directory to Backup storage path: /mnt/dietpi_userdata/urbackup should be sufficient. 

#### Actual behaviour:

Web UI reports "The directory where UrBackup will save temporary files is inaccessible. Please fix that by modifying this folder in 'Settings' or by giving UrBackup rights to access this directory."

#### Steps to reproduce:

Install UrBackup 
Once you have the web interface open:
    Select Settings
    Change the Backup Storage Path. /mnt/dietpi_userdata/urbackup is recommended.
    Click Save
    Restart service with dietpi-services restart
    Then click "status" 
#### Did you submit a ```dietpi-bugreport```?

#### Extra details:

The permissions seem correct
[image]

[Further reading 1]([url](https://forums.urbackup.org/t/the-directory-where-urbackup-will-save-temporary-files-is-inaccessible/886)). [Further reading 2](https://forums.urbackup.org/t/server-cannot-access-where-backups-are-to-be-saved/702/6) I've tried enabling Temporary files as file backup buffer and Temporary files as image backup buffer. For some reason it slows the interface down to a crawl. File permissions don't seem to issue.

Fourdee, 2017-12-25:
@LexiconCode 

Merry xmas! 🎄 

Thanks for the report 👍 

We'll try to replicate and resolve, after xmas period.

Fourdee, 2017-12-26:
@LexiconCode 

Appears to be a bug with UrBackup itself, pre 2.1.20.

Installed 2.1.20 = fine
Installed 2.1.19 = error confirmed on main screen.

I'll mark this as closed, due to https://github.com/Fourdee/DietPi/issues/1335#issuecomment-353970229

closed 2017-12-26 by Fourdee

Yury-MonZon, 2021-04-03:
The bug is still there. I've updated everything to the latest versions. Moved my urbackup folder and changed the /etc/urbackup/backupfolder file as well as the path in the web interface.
```
cat /etc/urbackup/backupfolder
/mnt/<UUID>/urbackup/
```

Temporary files as file backup buffer and Temporary files as image backup buffer are off as my /tmp is in the RAM:
tmpfs on /tmp type tmpfs (rw,nosuid,nodev,noatime,lazytime,size=4103168k)

Still getting this error:` The directory where UrBackup will save temporary files is inaccessible. Please fix that by modifying this folder in 'Settings' or by giving UrBackup rights to access this directory.`

And in the log I see this:
`04/03/21 13:01 | ERROR | Error creating temporary file at "/mnt/dietpi_userdata/urbackup/tmp/cps.InKQMo"`

Is there a place where I need to change the path to tmp folders?

MichaIng, 2021-04-03:
Please assure that the urbackup user has write permissions to do directory:
```
chown -R urbackup:urbackup /mnt/<UUID>/urbackup
```

Yury-MonZon, 2021-04-03:
My permissions are correct. But I've found the config I need to edit:  
`/etc/default/urbackupsrv`

NoahDar, 2021-09-26:
> My permissions are correct. But I've found the config I need to edit:
> `/etc/default/urbackupsrv`

If this config file can't be edited in the WebGUI maybe a little note on the settings page?

MichaIng, 2021-09-29:
We will add the info to our documentation: https://github.com/MichaIng/DietPi-Docs/issues/583

StephanStS, 2022-12-31:
See PR https://github.com/MichaIng/DietPi-Docs/pull/818