MichaIng/DietPi #1338: DietPi-Software | pihole no IP addresses found.

LexiconCode, 2017-12-26:
#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```? 159
- SBC Device (EG: RPi 3)? Odroid C2
- Power supply used (EG: 5v 1A RAVpower)? Odroid official power supply
- SD card used (EG: Sandisk ultra)?  Sandisk ultra
- Distro (EG: Jessie) | ```uname -a```?  aarch64 **Jessie and Stretch**

#### Additional Information (if applicable):
- Software title? pihole
- Can this issue be replicated on a fresh installation of DietPi? yes

#### Expected behaviour:
pihole should install by DietPi and be accessible through the web UI.

#### Actual behaviour:
Upon Restart after install pihole does not show in the optimized software as installed and is not accessible through the web UI. Nor does it function correctly as a DNS through the router.

#### Steps to reproduce:
Install pihole as recommended through DietPi

#### Did you submit a ```dietpi-bugreport```? No

#### Extra details:
Initially I was having issues with nginx being the preferred Web server along with Lighttpd installed by dietpi. This proved not to be the case on a fresh install Lighttpd as the preferred Web server with just pihole.

local_Ip/admin/ 404 Not Found
local_Ip/pihole/ 404 Not Found < This shouldn't be available but for testing purposes I looked anyway.
[image]

During the dietpi install on Jessie and Stretch
```
  [i] Preparing to run gravity.sh to refresh hosts...
  [i] Running gravity.sh
  No IP addresses found! Please run 'pihole -r' to reconfigure
 [Failed] Pi-Hole exited with code (1) and is not installed.
```
During the dietpi install
Upon the Lighttpd services restarting before the dietpi configuration it fails to restart the service.
When the dietpi controlled service Lighttpd restarts reports that it successful.
Upon reboot Lighttpd and pihole seems as successful.

Looking at all the log files (PiHole, Lighttpd and PHP) everything seems to be working like it should and no errors reported. Which doesn't really make sense... Considering the above issues described.

I was able to fix in Jessie by restoring lighttpd.conf temporarily. Jessie on first boot with pi hole working. Second reboot it would fail.
    mv /etc/lighttpd/lighttpd.conf /etc/lighttpd/lighttpd.conf.bak
    mv /etc/lighttpd/lighttpd.conf.bak /etc/lighttpd/lighttpd.conf

I believe the root of the issue can be found trying to update the gravity list with the following command.
root@DietPi:~# pihole -g
  No IP addresses found! Please run 'pihole -r' to reconfigure

pihole -r does not fix the issue either through repair or reconfiguration. Obviously this messes up DietPi intended configuration. Restoring lighttpd.conf is needed.

ipv6 is disabled in my router and dietpi.

Side note
I would recommend backing up after the dietpi configuration running  "mv /etc/lighttpd/lighttpd.conf /etc/lighttpd/lighttpd.conf.bak". with that integrated into dietpi configuration phase of installs process the lighttpd config can be restored without the user having to think about backing it up.

Anyway I'm kind of stumped to give you more detail on where to look as some of this seems impossible. For instance if local_Ip/admin/  is 404 Not Found I shouldn't see some sort relevant output lighttpd logs.

k-plan, 2017-12-27:
@LexiconCode 

can you please provide terminal output of:

`ls -lah /var/www/`

and

`cat /etc/lighttpd/lighttpd.conf | grep document-root`

LexiconCode, 2017-12-27:
ls -lah /var/www/
`root@DietPi:~# ls -lah /var/www/
total 88K
drwxrwxr-x  4 www-data www-data 4.0K Dec 27 07:36 .
drwxr-xr-x 12 root     root     4.0K Dec 26 23:06 ..
-rwxrwxr-x  1 www-data www-data  38K Dec 27 07:36 apc.php
drwxrwxr-x  2 www-data www-data 4.0K Jan 14  2017 cgi-bin
drwxrwxr-x  4 www-data www-data 4.0K Dec 27 07:35 html
-rwxrwxr-x  1 www-data www-data  612 Dec 27 07:36 index.html
-rwxrwxr-x  1 www-data www-data  23K Dec 27 07:36 opcache.php
-rwxrwxr-x  1 www-data www-data   20 Dec 27 07:36 phpinfo.php
`

cat /etc/lighttpd/lighttpd.conf | grep document-root
`root@DietPi:~# cat /etc/lighttpd/lighttpd.conf | grep document-root
server.document-root        = "/var/www/html"
`

k-plan, 2017-12-27:
On a working installation, it look like this:

```
:~# cat /etc/lighttpd/lighttpd.conf | grep document-root
server.document-root        = "/var/www"
```

```
:~# ls -lah /var/www/
insgesamt 88K
drwxrwxr-x  3 www-data www-data 4,0K Dez 11 20:11 .
drwxr-xr-x 13 root     root     4,0K Okt 21 18:45 ..
lrwxrwxrwx  1 www-data www-data   19 Dez 11 20:11 admin -> /var/www/html/admin
-rwxrwxr-x  1 www-data www-data  38K Sep 25 18:34 apc.php
drwxrwxr-x  4 www-data www-data 4,0K Dez 11 20:10 html
-rwxrwxr-x  1 www-data www-data   68 Dez 11 20:11 index.html
-rwxrwxr-x  1 www-data www-data 3,3K Sep 25 18:20 index.lighttpd.html
-rwxrwxr-x  1 www-data www-data  23K Sep 25 18:34 opcache.php
-rwxrwxr-x  1 www-data www-data   20 Sep 25 18:34 phpinfo.php
lrwxrwxrwx  1 www-data www-data   19 Dez 11 20:11 pihole -> /var/www/html/admin
```

Something went wrong with pihole installation script.

This are messages from pihole installation script:
```
[i] Preparing to run gravity.sh to refresh hosts...
[i] Running gravity.sh
  No IP addresses found! Please run 'pihole -r' to reconfigure
```
And this is a message from DietPi script:

`[Failed] Pi-Hole exited with code (1) and is not installed.`

This happens, if something is missing while installation scripts are running.

[https://github.com/Fourdee/DietPi/issues/1282#issuecomment-350490524](https://github.com/Fourdee/DietPi/issues/1282#issuecomment-350490524)

> I believe the root of the issue can be found trying to update the gravity list with the following command.
> `root@DietPi:~# pihole -g`

No the root of your problems now is you have done:

`root@DietPi:~# pihole -r`

Uninstall pihole via `dietpi-software` and start over again,

LexiconCode, 2017-12-27:
> No, root of your problems now is you have done:

` pihole -r`

@k-plan 
To emphasize this can be replicated on a fresh install just by trying to install PiHole. Sorry if I muddied the waters earlier my convoluted post. I've done a lot of testing to experiment part of which was testing out the ` pihole -r`. _Each_ test is done on a fresh install DietPi via reimaging from dietpi official images. The output in the my post of above belongs to a fresh install.  The command ` pihole -r`. was used during _one_ testing session and has no bearing on the cause of or the solution to the reported issue.

The only changes in my testing environment have been these four factors. None of which had any bearing on the success of PiHole. Thus this should be completely reproducible.
> Webserver Preference : Nginx or lighttpd
> aarch64 Jessie and Stretch

LexiconCode, 2017-12-27:
Reviewing Issue #1282 
This may have relevance

> Odroid C2 Stretch, PiHole install script is unable to obtain IP address. Due to additional ??? ??? entries and use of awk. Workaround is for user to enter the IP address manually, when asked. Other devices are fine.

> root@DietPi:~# ip route get 8.8.8.8
> 8.8.8.8 via 192.168.0.1 via ??? ??? dev eth0 src 192.168.0.112
>     cache

Fourdee, 2017-12-28:
We'll need to open this up for investigations, to dertmine if this is a DietPi, kernel and/or PiHole scrape specific issue.

Notes:
- 🈯️ RPi 3 eth:
```
root@DietPi:~# ip route get 8.8.8.8
8.8.8.8 via 192.168.0.1 dev eth0 src 192.168.0.104
    cache
```
- 🈴 C2 eth: this would fail PiHole installation, unless IP address is manually entered.
```
root@DietPi:~# ip route get 8.8.8.8
8.8.8.8 via 192.168.0.1 via ??? ??? dev eth0 src 192.168.0.112
    cache
```

Fourdee, 2018-01-16:
Re-test with v6.0 image and latest kernel:
```
root@DietPi:~# ip route get 8.8.8.8
8.8.8.8 via 192.168.0.1 dev eth0  src 192.168.0.112
    cache
root@DietPi:~# cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 8 (jessie)"
NAME="Debian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=debian
HOME_URL="http://www.debian.org/"
SUPPORT_URL="http://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

Fourdee, 2018-01-16:
I'll mark as closed for now, issue appears resolved with v6.0 C2 image.

closed 2018-01-16 by Fourdee