MichaIng/DietPi #1340: DietPi-Software | Update non-package software

MichaIng, 2017-12-27:
As there were some topics about this for some of your software, I am not going through all non-package software, check for possibly available new APT packages and updates:
- All uncritical updates made: https://github.com/Fourdee/DietPi/pull/1339
_____________
Following software has updated versions and/or APT packages available, but needs review/discussion, as more needs to be changed than version string inside URL:
- **nomachine** (index 30): https://www.nomachine.com/de/download
There is a new version 6 available on there website.  Did we just grab their binaries, thus we can link directly to them, or do we have somehow different/adjusted packages? http://dietpi.com/downloads/binaries/all
For RPi3 they have an armv8/arm64 package. Not 100% sure which one to take, but would just replace ours with their amd64, armv6hf and armhf /armv7 (they link for RPi2 and RPi3 just to the same deb download).
There are also experimental packages for other ARM devices, "e.g. BeagleBone, Radxa Rock, Banana Pi and Odroid C2", but if there is not explicit request, I think we should skip them, as long as they are listed as experimental. The armv7 devices were just tested on Ubuntu, as far as I can see.
- **mpd** (index 128): There are APT packages available, but indeed the ones until Stretch are still version 19: https://packages.debian.org/search?keywords=mpd
But the advantage of APT taking care about dependencies should be considered.
- **ympd** (index 32): https://github.com/notandy/ympd/releases
There is a never version available on github, on their webseite v1.2.3 is however also the latest: https://www.ympd.org/#downloads
- **cava** (index 119): Never version available on github: https://github.com/nikp123/cava/releases
- **mopidy** (index 118): I added Stretch and experimental Buster with the Stretch download to stick with their own repo. But actually it is available on official Debian repo for all distro versions: https://packages.debian.org/buster/mopidy
Would just take these, they are also (besides Jessie) current version.  For Jessie we could individually install their repo to stay on current version.
- **noip** (index 67): Current version 2.1.9: https://www.noip.com/download?page=linux
- **dxx-rebirth** (index 112): There is a 3rd party debian package available: http://files.crnatural.net/pool/contrib/d/
- **opentyrian** (index 51): Since Stretch there is an official debian package: https://packages.debian.org/buster/opentyrian
- **shairportsync** (index 37): Available on debian repo, but only on buster the version is current: https://packages.debian.org/buster/shairport-sync
- **squeezelite** (index 36): From Stretch on the apt package seems to be current: https://packages.debian.org/stretch/squeezelite
- **netdata** (index 65): APT package available since Buster and for Stretch as backport: https://packages.debian.org/stretch-backports/netdata
- **syncthing** (index 50): From Stretch on there is an official debian package: https://packages.debian.org/stretch/syncthing
- **sabnzbd** (index 139): 
- **nukkit** (index 164): Development seems to go on, but I don't find a new .jar download so far: https://github.com/Nukkit/Nukkit
- **gmediatender** (index 163): There would be an APT package: https://packages.debian.org/stretch/gmediarender

Fourdee, 2017-12-27:
@MichaIng 

Great stuff 👍 

Although, I hate these new git emotes, looks like a child made them, for other children.

Fourdee, 2017-12-27:
@MichaIng 

Legend 👍 

Merged: https://github.com/Fourdee/DietPi/pull/1339#issuecomment-354128584

Testing required (quick installs) to verify functional.

MichaIng, 2017-12-27:
Everything tested on VM Jessie+Stretch:
- All installations work fine, software does it's job/opens as far as I could test fast, but:
- Koel installation fails at least on only-local VMs: https://github.com/Fourdee/DietPi/issues/1343
- General issue with MariaDB: https://github.com/Fourdee/DietPi/issues/1344

Fourdee, 2017-12-29:
@MichaIng 

lol
[image]

- Koel, install is fine, but new issue with automation (cant see a solution, looks limited to Koel's source?): https://github.com/Fourdee/DietPi/issues/1343#issuecomment-354434655

- MariaDB looks like a big one, i'll take a look when I can.

MichaIng, 2017-12-29:
@Fourdee 
Huh, these icons are new for you? I have them as long as I can remember using github. Now I am interested how the old ones looked like 😆.

- Koel seems to be related to my VM network setup, see the separete topic: https://github.com/Fourdee/DietPi/issues/1343
- Yes, further discussion in separate topic also: https://github.com/Fourdee/DietPi/issues/1344

Lets concentrate here on the update method/installation source of software list above, as we find time. But as long as there are no actual issues with our current versions/sources, there is no need to hurry. Will mark this as low topic for that reason.

MichaIng, 2018-03-13:
@Fourdee 
To come back to this topic:
- **NoMachine**: version 6 available, as well as RPi3/ARMv8 version: https://www.nomachine.com/de/download/linux&id=29&s=Raspberry
  - Tested successfully on VM with: http://download.nomachine.com/download/6.0/Linux/nomachine_6.0.78_1_amd64.deb

Fourdee, 2018-03-14:
@MichaIng 

> NoMachine: version 6 available, as well as RPi3/ARMv8 version: 

Thanks.

Binaries updated, tested RPi 3 with 6.0.78:
- 🈯️

MichaIng, 2018-05-13:
@Fourdee 
I finished the list above with all our software where the newest version is not installed automatically.
By this we can easily check for new versions regularly and update our links or upload/build new binaries.
As you can see, there is already much to do 😅.

I am thinking about updating all links and binaries that do not need to be build manually, as I am not too experienced with this and do not have all required machines anyway.
But theoretically we need to retest the installation/configuration process for all affected software on all devices, which is even more work.

Maybe with above link and binary updates, we can offer a "software update" branch with just those updates based on v6.8 release and invite users to help testing their new software versions via `dietpi-software reinstall`?

Fourdee, 2018-05-14:
@MichaIng 

Great work, yep, its a long task this one 👍 

> But theoretically we need to retest the installation/configuration process for all affected software on all devices, which is even more work.

Yep, I can test ARMv6/7/8. 
Not only do we need to verify install succeeds, but also its functional via basic use testing (eg: check interface works).

> Maybe with above link and binary updates, we can offer a "software update" branch with just those updates based on v6.8 release and invite users to help testing their new software versions via dietpi-software reinstall?

Like the idea 👍 However, I believe we can request them to use the testing branch. I personally feel making multiple branches will cause havoc and confusion for us (well me at least 🤣).
They could basically do a fresh install of master, run a backup, then ```G_DEV_1```, then test.

Fourdee, 2018-05-15:
ARMv6 blynk (and all JAVA installs?)
```
Error occurred during initialization of VM
Server VM is only supported on ARMv7+ VFP
E: /etc/ca-certificates/update.d/jks-keystore exited with code 1.
done.
Errors were encountered while processing:
 ca-certificates-java
 openjdk-9-jre-headless:armhf
E: Sub-process /usr/bin/dpkg returned an error code (1)

[FAILED] DietPi-Software | G_AGI: ca-certificates-java
```

MichaIng, 2018-05-15:
> openjdk-9-jre-headless:armhf

Java 9?? Actually until Stretch `default-jre-headless` should be Java 8, or is this different on Raspbian repo? Here Raspbian is always a bid behind Debian: Raspbian Buster offers Java 9, while Debian Buster already Java 10.
- https://packages.debian.org/de/stretch/default-jre-headless
- https://github.com/Fourdee/DietPi/blob/testing/dietpi/dietpi-software#L7295

I think to assure compatibility we should install `openjdk-8-jre-headless` / `openjdk-8-jdk-headless` packages and do a controlled move to/adding of Java 9/10 by times.

MichaIng, 2018-05-15:
Blynk:
- On first install attempt I got an unlimited fast repeating error on both systems, Jessie and Stretch:
```
gyp WARN EACCES user "root" does not have permission to access the dev dir "/usr/local/lib/node_modules/onoff/node_modules/epoll/.node-gyp/10.1.0"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/usr/local/lib/node_modules/onoff/node_modules/epoll/.node-gyp"
```
- My scroll buffer was filled with this to fast, but according to error messages it should be the following within Blynk install step: `npm install -g onoff`
- Interestingly I was able to manually run the command afterwards without issue. Another dietpi-software install 131 run through then nicely.
<s>- Why do we do this node.js install etc actually? Can't find a hint about this within their install instructions: https://github.com/blynkkk/blynk-server#blynk-server</s>
- However not sure what caused this error on first attempt on both systems 🤔.

But runs fine afterwards including web UI. 🈯️

MichaIng, 2018-05-15:
HAProxy:
- It's incompatible with webservers binding to port 80 as well of course.
- If I am not mistaken , when installing a webserver in parallel, we need to configure it to listen to the haproxy backend port (9000, 9001 and 9002 by default)? Usually user should know this, when installing a load balancer, but just to have a nice installation, we could automate this.

Fourdee, 2018-05-17:
@MichaIng 

> I think to assure compatibility we should install openjdk-8-jre-headless / openjdk-8-jdk-headless packages and do a controlled move to/adding of Java 9/10 by times.

Yep agree, i'll make the change and test on systems 👍

Fourdee, 2018-05-17:
@MichaIng 

> gyp WARN EACCES user "root" does not have permission to access the dev dir "/usr/local/lib/node_modules/onoff/node_modules/epoll/.node-gyp/10.1.0"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/usr/local/lib/node_modules/onoff/node_modules/epoll/.node-gyp"

```--unsafe-perm``` will remove the root permissions warning, i'll send commit.

Fourdee, 2018-05-17:
Blynk reinstall test:

- x86_64 🈯️ 
- ARMv6 🈴  
- ARMv8 🈴 
```
Error: missing `server' JVM at `/usr/lib/jvm/java-8-openjdk-armhf/jre/lib/arm/server/libjvm.so'.
Please install or use the JRE or JDK that contains these missing components.
E: /etc/ca-certificates/update.d/jks-keystore exited with code 1.
done.
Errors were encountered while processing:
 ca-certificates-java
 openjdk-8-jre-headless:armhf
 openjdk-8-jdk-headless:armhf
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

MichaIng, 2018-05-17:
@Fourdee 
Can you check which repo is actually used on these ARMs?

On Debian it should take Java 8: https://packages.debian.org/de/stretch/ca-certificates-java
- The deps are a bid wired, but as `openjdk-7-jre-headless` is not available on Stretch, `openjdk-8-jre-headless` should be taken.

On my RPi with Raspbian Buster is shows the right deps as well:
`Depends: ca-certificates (>= 20121114), openjdk-8-jre-headless | java8-runtime-headless, libnss3 (>= 3.12.10-2~)`

Fourdee, 2018-05-17:
Raspbian Stretch:
```
root@DietPi:~# apt-cache depends ca-certificates-java
ca-certificates-java
  Depends: ca-certificates
 |Depends: <openjdk-7-jre-headless>
  Depends: <java7-runtime-headless>
    default-jre-headless
    openjdk-8-jre-headless
    openjdk-9-jre-headless
    oracle-java7-jdk
    oracle-java8-jdk
  Depends: libnss3

root@DietPi:~# apt-cache policy ca-certificates-java
ca-certificates-java:
  Installed: 20170531+nmu1
  Candidate: 20170531+nmu1
  Version table:
 *** 20170531+nmu1 500
        500 http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages
        100 /var/lib/dpkg/status

```

ARMv8 debian:
```
root@DietPi:~# apt-cache depends ca-certificates-java
ca-certificates-java
  Depends: ca-certificates
 |Depends: <openjdk-7-jre-headless>
  Depends: <java7-runtime-headless>
    default-jre-headless
    openjdk-8-jre-headless
    openjdk-9-jre-headless
  Depends: libnss3

root@DietPi:~# apt-cache policy ca-certificates-java
ca-certificates-java:
  Installed: 20170531+nmu1
  Candidate: 20170531+nmu1
  Version table:
 *** 20170531+nmu1 500
        500 https://deb.debian.org/debian stretch/main armhf Packages
        100 /var/lib/dpkg/status

```

Fourdee, 2018-05-17:
🈴 Spotify web connect:
```
[FAILED] DietPi-Services | spotify-connect-web  failed (Result: exit-code) since Thu 2018-05-17 17:13:17 BST; 21s ago
● spotify-connect-web.service - spotify-connect-web
   Loaded: loaded (/etc/systemd/system/spotify-connect-web.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2018-05-17 17:13:17 BST; 21s ago
  Process: 8422 ExecStart=/mnt/dietpi_userdata/spotify-connect-web/spotify-connect-web (code=exited, status=255)
 Main PID: 8422 (code=exited, status=255)

May 17 17:13:17 DietPi spotify-connect-web[8422]:     from connect import Connect
May 17 17:13:17 DietPi spotify-connect-web[8422]:   File "/mnt/dietpi_userdata/spotify-connect-web/connect.py", line 9, in <module>
May 17 17:13:17 DietPi spotify-connect-web[8422]:     from console_callbacks import audio_arg_parser, mixer, error_callback, connection_callbacks, debug_callbacks, playback_callbacks, playback_setup
May 17 17:13:17 DietPi spotify-connect-web[8422]:   File "/mnt/dietpi_userdata/spotify-connect-web/console_callbacks.py", line 25, in <module>
May 17 17:13:17 DietPi spotify-connect-web[8422]:     audio_arg_parser.add_argument('--mixer', '-m', help='alsa mixer name for volume control', default=alsa.mixers()[0])
May 17 17:13:17 DietPi spotify-connect-web[8422]: alsaaudio.ALSAAudioError: No such file or directory [default]
May 17 17:13:17 DietPi spotify-connect-web[8422]: Failed to execute script loader
May 17 17:13:17 DietPi systemd[1]: spotify-connect-web.service: Main process exited, code=exited, status=255/n/a
May 17 17:13:17 DietPi systemd[1]: spotify-connect-web.service: Unit entered failed state.
May 17 17:13:17 DietPi systemd[1]: spotify-connect-web.service: Failed with result 'exit-code'.
```

MichaIng, 2018-05-17:
Ah jep `apt-cache depends ca-certificates-java` shows more of the truth, on Raspbian Buster:
```
2018-05-17 17:31:28 root@micha:/var/log# apt-cache depends ca-certificates-java
ca-certificates-java
  Depends: ca-certificates
 |Depends: openjdk-8-jre-headless
  Depends: <java8-runtime-headless>
    default-jre-headless
    openjdk-10-jre-headless
    openjdk-8-jre-headless
    openjdk-9-jre-headless
    oracle-java8-jdk
  Depends: libnss3
```
- should install Java 10, which is also default-jdk on Debian Buster.
______________
ARMv8 has the same issue without installing `ca-certificates-java` even that it is Debian repo?
Otherwise if it's just a Raspbian repo issue (as it works fine on my VM to install just `openjdk-8-jdk-headless`), we could open some bug report there, to have this ugly workaround fixed with Java 9 installed as well.

Fourdee, 2018-05-17:
@MichaIng 

> ARMv8 has the same issue without installing ca-certificates-java even that it is Debian repo?

Yep, same issue, I think the cause is the actual package install script (limited to ARM?), and the order it runs in. Not much we can do about that, workaround required I believe for now.

> we could open some bug report there, to have this ugly workaround fixed with Java 9 installed as well.

Yep, ideally we need to exclude our OS from tests, if we can replicate with default Raspbian/Debian (once packages removed), we can then send this off as a bug.

Fourdee, 2018-05-18:
Kind of agree with this: https://gist.github.com/bplower/613a99156d603abac083#gistcomment-2226016

But i'll try and get it fixed.

MichaIng, 2018-05-18:
@Fourdee 
> Kind of agree with this

Yes thought the same, so much install steps, so much pre-dependencies and so much warnings about deprecated, missing, skipped etc. modules/parts whatever. But I can't really give qualified criticism here, as I don't understand much of what's actually going on. Already the build process until `make` is in detail cryptic to me, okay at least I know in theory about compiling source code into binaries of course 😄.
- At least, if after the build is fully done, if the binary is slim and fast and beautiful, everything is nice, and you can remove all the building tools, if wanted. We could think about the same, actually 😉.
- But what I don't like more is software, that calls itself slim and fast, that needs to load heavy dependencies then to run. Better a built binary, than the source code need to be compiled on every start-up again by heavy compiler. Mainly this is what I don't like about most Java software, e.g. just watched AirSonic and SubSonic and some other Software I "like" (better need) to use on Windows 😉.

**Admin info**
- https://github.com/phanan/koel/search?utf8=✓&q=ADMIN_NAME&type=
- Thus, should actually still works this way. Maybe the errors we face lead to `.env` not being read correctly? Otherwise database info is successfully read/used... But better first fix the other errors, then see if/how to automate.
- Bid similar, but not exactly the same: https://github.com/phanan/koel/issues/748
- Will try around tomorrow, maybe never versions (Buster) work, or following the install instructions precisely...

Fourdee, 2018-05-21:
@MichaIng 

Great work on Koel 👍 

Just ARMv6 to test, leaving it installing over night 😉.

---

ToDo (me):
- RPI Cam tests

Fourdee, 2018-05-22:
Koel ARMv6:
```
file_get_contents(/var/www/koel/public/mix-manifest.json): failed to open stream: No such file or directory (View: /var/www/koel/resources/views/index.blade.php)
```

Fourdee, 2018-05-25:
SABnzbd test install:
- 🈯️ 

Cava update, test installs:
- 🈯️ ARMv6
- 🈯️ ARMv7
- 🈯️ ARMv8
- 🈯️ AMD64

MichaIng, 2018-05-30:
@Fourdee 
Are the last errors (CAVA e.g.) resolved now? As v6.9 is a huge update with some deep changes, we have to take care to not forget testing/finishing everything.

Would suggest to stop new feature input for this update, just testing and bugfixing.

Fourdee, 2018-05-30:
### Cava Odroid C2
- We'll need to use the older binary, which works fine. The newer binary requires CPU instructions/extensions which are not available on the C2: https://github.com/Fourdee/DietPi/issues/1340#issuecomment-393206263
- I don't have the time + enough knowledge to edit the makefile/cava source to resolve at this time.

MichaIng, 2019-01-12:
Extended and moved table to Wiki: https://github.com/Fourdee/DietPi/wiki/DietPi-Software-list

closed 2019-01-12 by MichaIng