MichaIng/DietPi #1343: DietPi-Software | Koel: Installation fails on VMs getting composer

MichaIng, 2017-12-27:
On my only-local network VMs (Jessie and Stretch), Koel installation fails on following step: https://github.com/Fourdee/DietPi/blob/testing/dietpi/dietpi-software#L7157

Also using `php -r "copy('https://getcomposer.org/installer', 'composer-setup.php');"` like mentioned on composer download page, fails the same way: https://getcomposer.org/download/

Error: `PHP Warning:  readfile(https://getcomposer.org/installer): failed to open stream: Connection timed out in Command line code on line 1`

On my RPi2 server, which is reachable from the internet, it works well.

Might this just be a problem, if server is not reachable from outside of local network?

________________________________

On VMs after that some other warnings occur, so Koel installation with new version 3.7.0 needs to be tested on some globally reachable server again.

Fourdee, 2017-12-29:
@MichaIng 

> Might this just be a problem, if server is not reachable from outside of local network?

Was unable to replicate the errors you experienced on my network. UPnP disabled and no port forwarding to the devices.
DNS/routing issue?

>  so Koel installation with new version 3.7.0 needs to be tested on some globally reachable server again.

We get some warnings regarding npn packages, however this is what Koel requires and installs itself.

---

New user prompts during install, would be nice if we can automate these:
```
Attempting to install or upgrade Koel.
Remember, you can always install/upgrade manually following the guide here:
📙  https://koel.phanan.net/docs

Generating app key
Generating JWT secret
Migrating database
Let's create the admin account.

 Your name:
 > dietpi

 Your email address:
 > dietpi@dietpi.com

 Your desired password:
 >

 Again, just to make sure:
 >

Seeding initial data
The absolute path to your media directory. If this is skipped (left blank) now, you can set it later via the web interface.

 Media path []:
 > /mnt/dietpi_userdata/Music
```

MichaIng, 2017-12-29:
@Fourdee 
I just tested again on all VirtualBox VMs (including Buster this time) and on Apache (mod-php) as well as Lighttpd (php-fpm) and the issue is just the same:
`php -r "readfile('https://getcomposer.org/installer');"` fails with timeout mentioned above, also running just in terminal.

I ran further tests:
- The issue seems not to be consistent across web sources.
- `php -r "readfile('http://getcomposer.org/installer');"` (HTTP) fails as well.
- `php -r "readfile('https://www.w3.org/2001/06/utf-8-test/UTF-8-demo.html');"` works (HTTPS)
- `php -r "readfile('http://textfiles.com/adventure/221baker.txt');"` works as well (HTTP)
- `php -r "readfile('https://textfiles.com/adventure/221baker.txt');"` (HTTPS, but same file) gives directly an error and no timeout:
```
PHP Warning:  readfile(): SSL operation failed with code 1. OpenSSL Error messages:
error:1416F086:SSL routines:tls_process_server_certificate:certificate verify failed in Command line code on line 1
PHP Warning:  readfile(): Failed to enable crypto in Command line code on line 1
PHP Warning:  readfile(https://textfiles.com/adventure/221baker.txt): failed to open stream: operation failed in Command line code on line 1
```
... even that in browser, this textfile opens without any SSL warning.
- `curl https://getcomposer.org/installer` works without issues btw.

Behaviour is the same on all three machines, so it is not about the PHP version or run method (mod-php vs. php-fpm) and does not strictly depend on HTTPS/HTTP sources, but seems to be an issue with the kind of network access (as mentioned, works fine on production RPi2).
- My VMs are running on my laptop (desktop) on most current VirtualBox with virtio network adapters on bridges networking mode. Network established using DHCP first for router (Fritz!Box) to recognize them (also that they are running on my laptop, using it's wlan connection), then switched to fixed IP on laptop and VMs.
- Everything is working fine, SSH, internet access, local access to webserver/-services, but no port forwarding to the VMs is used.

I searched the web quite a lot, but didn't found any hint that suites my issue. Will try to change the way, VMs are connected to network by time.
But I stuck by investigating the actual problem, as I didn't find any related log entry (not /var/log nor journalctl) nor did I find more verbose mode of running this php commands.
______________________________________________
About your issues:
- Interesting, at what stage these questions were asked?
- They didn't appear on all my test installations, but might be connected to the missing composer install step?
- But I also never tried to access/start koel somehow. It's service by the way also fails to start/restart with the same readfile() timeout error on another web source 😆.

MichaIng, 2018-01-23:
@Fourdee 
Can't believe it, I should have known: The failing `readfile()` of course also is a timeout of resolving an IPv6 address... The idea driving me crazy that I must force every web access command separately to not use IPv6.

So I made some research again:
- First I found, that the old setting we used to disable IPv6 via sysctl, was solving the problem for my VMs: https://github.com/Fourdee/DietPi/issues/1119
- Wondering that no error was produced, as on my RPi with these settings, I found that `/proc/sys/net/ipv6` still exists, even that IPv6 was disabled => blacklisted via modprobe.d/.
- Thanks god, I finally found this: https://askubuntu.com/questions/309461/how-to-disable-ipv6-permanently#comment1445267_499485
> This answer assumes that the kernels ipv6 feature is a kernel module (using blacklist.local to prevent its loading). **There is a case that ipv6 is compiled into the kernel (not a module)** in which case it is necessary to boot the kernel with **ipv6.disable=1** boot paremeter (to do so one would: (1) sudo gedit /etc/default/grub (2) search for the line GRUB_CMDLINE_LINUX_DEFAULT=<args> and (3) add ipv6.disable=1 to the boot parameters ) – humanityANDpeace May 19 '17 at 10:32
- With this, no ForceIPv4/preferipfamily=ipv4/separate readfile/php adjustment is necessary anymore, finally the "system" knows, that IPv6 is not an option 😆.
- No idea how to reliable find out, which devices use IPv6 not as module, but I would now assume, that this is the case for all our amd64 devices, as they're all based on the same image and all have grub bootloader.

Of course we should not set this on system preparation, as we don't know if user needs IPv6 support for his network. But on toggle IPv6 via dietpi-config, on amd64 we need to switch, as blacklisting has simply no effect.

MichaIng, 2018-01-24:
PR is ready: https://github.com/Fourdee/DietPi/pull/1419

closed 2018-01-24 by MichaIng