MichaIng/DietPi #1345: Dietpi-Software | Mosquitto is Unable to Install

techwithjake, 2017-12-27:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | 159
- SBC Device | RPi3
- Power supply used | CanaKit 1A Power Supply
- SD card used | Sandisk ultra
- Distro (EG: Jessie) | `Stretch

#### Additional Information (if applicable):
- Software title? | Mosquitto
- Can this issue be replicated on a fresh installation of DietPi? | Yes

#### Expected behaviour:

Dietpi-Software should be installing Mosquitto

#### Actual behaviour:

It is throwing an error:

```
[Failed] The apt cache may be corrupt, apt mirror offline, or you have held broken packages. DietPi-Software will now exit.

 [Info] Apt Log:
E: Unable to correct problems, you have held broken packages.

 [Info] Please try:
 [Info] - Fixing apt package errors with 'apt-get install -f'
 [Info] - Change apt mirror: 'dietpi-config' > 'Networking: NAS/MISC' > 'Apt Mirror'
 [Info] - Clear and refresh apt 'apt-get update'
 [Info] - If problems persist, please create a Git ticket so DietPi can investigate the issue:
 - https://github.com/Fourdee/DietPi/issues
```

This on a completely fresh DietPi Install on a brand new SD Card

#### Steps to reproduce:

Go to Dietpi-Software, select Mosquitto to install

Fails

#### Did you submit a ```dietpi-bugreport```?

No

#### Extra details:

MichaIng, 2017-12-27:
Fixed with v160, see here how to fix it now: https://github.com/Fourdee/DietPi/issues/1306#issuecomment-352167350

Will close the issue, reopen, if the error persists.

closed 2017-12-27 by MichaIng

techwithjake, 2017-12-27:
Thanks for letting me know. Tried looking but I guess I didn't look hard enough.

I found it odd though I could install it via apt-get just fine.

MichaIng, 2017-12-28:
@techdabbler 
Yeah that's true, with Stretch the official debian package also made a huge version jump, thou it is not as current as mosquittos own repo (that we use): 4.10 vs. 4.14 and as we know, usually the versions are frozen on debian stable repo, thus will fall behind more over time.

It is by the way also known/mentioned at mosquittos own guthub page: https://github.com/eclipse/mosquitto/issues/529#issuecomment-324885355
So we can hope that they will resolve it there soon also.