MichaIng/DietPi #1347: OrangePI PC image does not boot

glogiotatidis, 2017-12-28:
## Creating a bug report/issue:
OrangePI PC image does not boot. Serial console reports "** Unrecognized filesystem type ** ext2"

#### Required Information:
- DietPi Version | ```cat /DietPi/dietpi/.version```? v145
- SBC Device (EG: RPi 3)? OrangePi PC v1.2
- Power supply used (EG: 5v 1A RAVpower)? 5v 2A
- SD card used: Kingston Class 10 and Sandisk Ultra Class 10

#### Actual behaviour:
Does not boot

#### Extra details:
- Managed to boot with the Orange Pi One image.
- Armbian for OrangePi PC works fine.

Fourdee, 2017-12-29:
@glogiotatidis 

Thanks for the report 👍 We'll try to replicate.

Fourdee, 2018-01-14:
I'll create the updated v6.0 today, test, then link here.

Fourdee, 2018-01-14:
OPi PC image re-created:
http://dietpi.com/downloads/testing/v6.0/DietPi_OPiPC-ARMv7-Stretch.7z

Tested and boots fine. Please use the image above.

Marking as resolved.

closed 2018-01-14 by Fourdee

glogiotatidis, 2018-01-24:
tested as well and boots. Thanks for the quick fix!