MichaIng/DietPi #1351: Compile mpd with pulseaudio support

stefanprobst, 2017-12-29:
The mpd version in DietPi stretch (0.20.11-1) is compiled without pulseaudio support:
`mpd --version

Output plugins:
 shout null fifo pipe alsa httpd recorder`

Would it be possible to offer mpd with pulseaudio suport? The official mpd version 0.19.21-1 in the raspbian stretch archive is compiled with these output plugins:
`shout null fifo pipe alsa roar ao oss openal pulse jack httpd recorder`

Thanks.

Fourdee, 2017-12-29:
@stefanprobst 

Hi,

Thanks for the request. 

We will never support or endorse pulseaudio throughout DietPi.
The main reasons being bulky, unnecessary, waste of CPU usage. ALSA does it better.

Please don't take the above personally, just me expressing my sheer hatred for pulseaudio.

closed 2017-12-29 by Fourdee

stefanprobst, 2017-12-29:
Ok, thanks for the info. For me, pulseaudio became interesting with the new `avoid-resampling` option, and I like how easy it is to set up network streaming.