MichaIng/DietPi #1352: DietPi-Software | Uninstall terminate, if package is not installed+not present in repo

MichaIng, 2017-12-29:
Just faced this on uninstalling ffmpeg. Sadly lost the exact error, but is reproducible:
```
root@DietPi:~# G_AGP libx264 libmp3lame libfdk-aac
 [Info] APT removal for: libx264 libmp3lame libfdk-aac
Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package libx264
E: Unable to locate package libmp3lame
E: Unable to locate package libfdk-aac

 [Error] G_AGP: libx264 libmp3lame libfdk-aac
 [Info] exit_code = 100
 [Info] HW_MODEL:20 | HW_ARCH:10 | DISTRO:4
 [Info] Logfile contents:
Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package libx264
E: Unable to locate package libmp3lame
E: Unable to locate package libfdk-aac
 [Info] Unable to continue, the program will now terminate.
```
- Error is thrown, as the packages do only get installed on RPi together with own built ffmpeg. These libraries are not installed on non-RPi devices and do not exist in apt repositories, which seems to be the reason why an error is thrown, thus at least new global error handler terminates uninstallation.
```
root@DietPi:~# G_AGP ffmpeg
 [Info] APT removal for: ffmpeg
Reading package lists...
Building dependency tree...
Reading state information...
Package 'ffmpeg' is not installed, so not removed
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
 [Ok] G_AGP: ffmpeg
```
- As we can see, if package is part of repo, then no error is thrown, just information.
```
root@DietPi:~# G_RUN_CMD dpkg -P libx264 libmp3lame libfdk-aac
 [Info] Running command: dpkg -P libx264 libmp3lame libfdk-aac
 [Info] Please wait...
 [Ok] dpkg -P libx264 libmp3lame libfdk-aac
```
- `dpkg -P` seems to work in cases, where packages were also installed by `dpkg -i`. Errors are shown on command, but do not exit with error code:
```
root@DietPi:~# dpkg -P libx264 libmp3lame libfdk-aac
dpkg: warning: ignoring request to remove libx264 which isn't installed
dpkg: warning: ignoring request to remove libmp3lame which isn't installed
dpkg: warning: ignoring request to remove libfdk-aac which isn't installed
```

How should be solve it? Use dpkg -P for all packages, clearly installed by dpkg -i, or somehow ignore this kind of error? Of course `(( $G_HW_MODEL < 10 )) && ...` would also work in this specific case.

Just for quick find location:
https://github.com/Fourdee/DietPi/blob/testing/dietpi/dietpi-software#L14354

Fourdee, 2017-12-30:
@MichaIng 

1st is to not exit on ```G_AGP``` error, allowing notification, but continue:
```
export G_ERROR_HANDLER_ONERROR_EXIT=0
G_AGP libx264 libmp3lame libfdk-aac
```

2nd, as you mention, use the ```$G_HW_MODEL``` id for the additional package removals. But check those packages are installed in the code.

3rd, as you mentioned ```G_RUN_CMD dpkg -P libx264 libmp3lame libfdk-aac```, although, this only removes single packages, not deps.

I'd probably go with options 1 and 2?

MichaIng, 2017-12-30:
@Fourdee 
About 1st:
- Easy here, but generally I like the idea of generally terminate on every (unknown) error, as who knows what could go wrong on some wrong/missing/whatever library or I don't know.
- But then we need to handle every known error before.

2nd:
- Jep makes sense here, so adopt the hw model check from installation. Will create PR about this.

3rd:
- Also A_AGP/apt-get purge does not remove dependencies, or did I missunderstand? Autoremove is called at the end of uninstall script.
- If packages are installed via `dpkg -i`, I guess dependencies cannot be pulled automatically through APT? If so, they should also be able to autoremove, after dpkg -P.

Finally I would go with 2 + 3, as they include exactly the checks/methods we use during installation, which sounds clean to me. If then some error occurs, it would be good for us to know.

_________________

Ahh and just the next idea:
- Giving user the choice to terminate the script or not.
- Perhaps users don't want to wait for some fix until they are able to finish and installation/uninstallation, especially as an error in one piece of software terminates all further installations and let's the already installed ones stay marked as not installed (and the other way round on uninstallation).
- To round it up, we could, on error during dietpi-software, mark the related one as "installed/uninstalled with error" or something, maybe an additional .installed stage of "-1". This would need to give an additional hint to the G_ERROR_HANDLER or doing this before.
**- Simpler and cleaner approach: Rework of software installation states, to enable marks as not_installed/to_install/installed_unconfigured/to_configure/installed_configured/to_uninstall and directly do the marks at beginning/end of the related part of the script. With this it would be able to know exactly where dietpi-software stopped and all finished steps are recorded/software is marked exactly until the stage they are. If one stays marked as "to_...", then it is kind of error state, as the e.g. configuration started but stopped in between.**

Fourdee, 2017-12-30:
@MichaIng 

> but generally I like the idea of generally terminate on every (unknown) error

Agree 👍 

> Also A_AGP/apt-get purge does not remove dependencies, or did I missunderstand?

G_AGP removes and purges the package
G_AGA will then autoremove and purge unneeded packages (eg: lib's used from the package we removed with G_AGP)

Default apt is to not purge (data/conf etc) stays in place: https://askubuntu.com/questions/231562/what-is-the-difference-between-apt-get-purge-and-apt-get-remove/231568

Unless its in ```dietpi-software``` which runs ```G_AGA``` during last stages. Add  ```G_AGA``` after ```G_AGP package```.

> Giving user the choice to terminate the script or not.

Agree, we'll need to test this, as some failed commands may not be recoverable/dangerous to system?. But yep, thumps up 👍 

> Perhaps users don't want to wait for some fix until they are able to finish and installation/uninstallation, especially as an error in one piece of software terminates all further installations and [...]
Cleaner approach: Rework of software installation states, to enable marks as not_installed/to_install/installed_unconfigured/to_configure/installed_configured/to_uninstall and directly do the marks at beginning/end of the related part of the script. With this it would be able to know exactly where dietpi-software stopped and all finished steps are recorded/software is marked exactly until the stage they are. If one stays marked as "to_...", then it is kind of error state, as the e.g. configuration started but stopped in between.

Improved indexing, well worth implementing and ideal time for core version update( https://github.com/Fourdee/DietPi/issues/1355), agree 👍

MichaIng, 2017-12-30:
@Fourdee 

About APT:
- Ah yeah, `apt-get remove package` equals `dpkg -r package`, leaving configuration files of `package`.
- `apt-get purge package` equals `dpkg -P package` also removing the configuration files of `package`.
- `apt-get autoremove --purge` will purge all auto marked packages with no dependants, at least the dependencies of `package`, no matter whether it was removed/purged by dpkg or apt-get command.
- Just tested `G_AGI vim` (with 3 dependencies) && `dpkg -P vim` && `G_AGA` and and everything went just the same as with `G_AGP`. Even `apt-mark showmanual` in between seems to be directly aware of removed/purged packages via dpkg.
- Thus for local installed packages via `dpkg -i` it should make no difference, even be more error prove, to also purge them via `dpkg -P` afterwards.

MichaIng, 2017-12-31:
PR is ready: https://github.com/Fourdee/DietPi/pull/1359

Just searched for dpkg -P and apt-get purge a bid, and technically there seems absolutely no difference. The strength of APT is, there everybody agrees, the automated installation of dependencies. On uninstalling packages, it also shows in case that there are obsolete dependencies that can be autoremoved. But we anyway hide these messages and G_AGA will be done anyway afterwards and works just the same.
As mentioned inside PR, `*` can't be used to name several packages at once. Thus we need to name all separately or handle '*' inside G_AGP (via dpkg --get-selections) before giving the string to dpkg -P.

MichaIng, 2018-01-15:
@Fourdee 
I did not yet make a fine uninstallation try for all affected software, but did try the basic method on all distro versions.

Fourdee, 2018-01-16:
@MichaIng 

Tested the main ones fine with most package changes (mpd/plex).

- Plex
```
#### Required Information:
- DietPi Version | v6.0
- SBC Device     | oDroid C2 (aarch64) (index=12)
- Distro         | jessie (index=3)
- Command        | G_AGP: plexmediaserver-installer
- Error Handler  | G_ERROR_HANDLER

#### Additional Information (if applicable):
- Software title | DietPi-Software

#### Expected behaviour:
<!-- What SHOULD be happening? -->

#### Actual behaviour:
<!-- What IS happening? -->

#### Steps to reproduce:
<!-- Explain how to reproduce the issue -->

#### Additional logs:
```
Log file contents:
Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package plexmediaserver-installer
```
I'll revert back to ```plexmediaserver*``` for G_AGP.

Fourdee, 2018-01-16:
Rest look fine, i'll mark as closed.

closed 2018-01-16 by Fourdee

MichaIng, 2018-01-16:
@Fourdee 
Interesting, as we clearly install plexmediaserver-installer: https://github.com/Fourdee/DietPi/blob/7ae462367b3e7e7bf6580a9da707c4092176215d/dietpi/dietpi-software#L5877
But good, as "*" works with apt-get, compared to dpkg -P 😄.