MichaIng/DietPi #1354: Info | Software requests reset

Fourdee, 2017-12-30:
Hi all,

Just for info:
- I've closed all outdated software requests that we have not be able to pick up yet, or, complete.
- If the software request gains enough interest, and, we are able to pick up the software requests, we'll reopen.

This is to allow us to focus and support our current software titles, ensuring stability and expected DietPi experience.

Invictaz, 2017-12-30:
@Fourdee Can you help me set up a Travis / compile robot? Then I can compile some software for the Pi version of DietPi and help you make some software requests I have work.

k-plan, 2017-12-31:
I think, we should leave this information on front Issues page, so everyone will read it.
After [DietPi v6.0 | Pre-release](https://github.com/Fourdee/DietPi/issues/1355) is closed, I will close this here as well.
But have lock conversation.

Thanks.

closed 2018-01-28 by Fourdee