MichaIng/DietPi #1355: DietPi v6.0 | Pre-release FAQ

Fourdee, 2017-12-30:
## DietPi v6.0

### A new major core version of DietPi will be our next release (previously v160).

#### When will it be released?
We could say (like most) "when its ready", however, we estimate 1-2 weeks at the least. Upto 1 month at the most.
- We apologise for the delay in this release, however, we need to ensure the new major core version meets expectations from our users.

#### What can we expect:
All DietPi images will be re-done from scratch using our new ```PREP_SYSTEM_FOR_DIETPI.sh``` (https://github.com/Fourdee/DietPi/blob/testing/PREP_SYSTEM_FOR_DIETPI.sh) system
- The new image prep system was co-created with @MichaIng, who has helped make this possible.
- Image sizes will be reduced.
- All images will be identical (package wise, minus any HW/ARCH/DISTRO specific changes)
- All images will be upto date

#### Vastly improved error handling:
We don't like to see users experience issues with DietPi, however, they can happen. Sometimes its our fault, sometimes not, but either way we want to investigate:
- Our new error control system will identify and report issues to the user, in a consistent and human readable debug-able format. https://github.com/Fourdee/DietPi/issues/1311#issuecomment-353716344
- The user can then simply copy and paste the pre-created Git Ticket post, into a Git Ticket for us to assist/investigate. https://github.com/Fourdee/DietPi/issues/1311#issuecomment-354544907

#### Use of global variables and commands
The backbone of DietPi has received vast improvements to simplify and optimize the sourcecode.
- Users will be able to benefit from a more stable system experience, that does not rely on duplicate code throughout DietPi scripts.
- We've added global commands that start with ```G_```, which can simplify many tedious Linux command processes https://github.com/Fourdee/DietPi/issues/1311#issuecomment-352999485, eg:
-- ```G_AGI package```, will install a package with minimal prompts, fuss and questions
-- ```G_AGP package``` will remove and purge the data (purge deletes the data from disk freeing up space, the default apt method is to leave it in place)
-- ```G_AGA``` will autoremove and purge data (purge deletes the data from disk freeing up space, the default apt method is to leave it in place)

#### Why will I not be able to update my existing installation of v159 (or lower)?
One of DietPi core features is the ability to update on demand. We love this feature, however, sometimes we simply need a full reset of existing installations to continue our support with the latest features and code.
- The new version of DietPi will be available with the latest images.
- Patching existing systems into the new version is not possible.
- Due to the major update, we need to ensure all our users have the same experience. We have vastly improved our image creation process, and, it will ensure future patches are consistent against the baseline images.

#### Will my existing v159 (or lower) installation still work?
Yes, however, the system will not be able to update and we will not be able to provide support.
- Please install the latest DietPi image to ensure future support and updates.

---

Reference in @MichaIng @k-plan. I hope you both agree this will allow us to completely reset DietPi baseline and focus on the future.

Fourdee, 2017-12-30:
Notes:
- Patch_File will be reset
- Images will be created, prior to v6.0's release
- Allo images will be re-created

MichaIng, 2017-12-30:
@Fourdee 
Nice summary,
however do you think it is necessary to completely block update from v159? So far I didn't face major issues during update and as usual the some, appearing, can be handled.
- If you think, it is necessary/the benefit outweighs the work for existing users, then I will start to create a Nextcloud/ownCloud migration script, more less based on this: https://docs.nextcloud.com/server/12/admin_manual/maintenance/migrating.html

However dropping Jessie support can be going hand in hand with this, as this anyway would need installing new image.. As I see on Odroid forum Meverics thread, the issue with Kodi on Odroid C2 is still present, so I guess we would need to pause offering this for C2 until Meveric made some progress there, to enable us moving on to Stretch.
But ah, I just see that there are many other devices still on Jessie. But I guess, as long as they base on default Debian ARM images, there should not occur major issues from our side?

Fourdee, 2017-12-31:
@MichaIng 

> however do you think it is necessary to completely block update from v159? So far I didn't face major issues during update and as usual the some, appearing, can be handled.

Yes. 
The main reason is to reset and standardize each image we offer, and ensure the user is running the same.
We can then ensure patches are more stable, as the previous images (and the process we used to create them) have changed vastly since v120 > v159, with minimal standardization.

> If you think, it is necessary/the benefit outweighs the work for existing users, then I will start to create a Nextcloud/ownCloud migration script, more less based on this: 

Go for it 👍 

> But ah, I just see that there are many other devices still on Jessie. But I guess, as long as they base on default Debian ARM images, there should not occur major issues from our side?

We wont be dropping Jessie, due to Odroid GPU/kodi packages limited to that distro. Other images (ARMbian etc), we'll re-do with Stretch.

MichaIng, 2017-12-31:
@Fourdee 
Do we want to offer a last v160 update, where we just backport some urgent bugfixes, like the LD_LIBRARY_PATH error?

Invictaz, 2017-12-31:
If someone can help me set up a Doozer or Travis build robot I can compile some software for DietPi that is now requested but not filled in.

Fourdee, 2018-01-01:
@MichaIng 

> Do we want to offer a last v160 update, where we just backport some urgent bugfixes, like the LD_LIBRARY_PATH error?

Personally, as we offer solutions on the git tickets for these issues, I don't really want to focus on fixes for v159.
Main reason being, we'd also need to dedicate time for testing and then supporting the release if we make any mistakes, or, other issues appear.

I'd prefer we simply focus on v6.0, give it all our attention and ensure its stable from the get go.

WolfganP, 2018-01-01:
Sounds like a good plan. As no auto update will be provided, I suggest creating an special document on how to migrate to the new version for those with custom setups (ie running an special backup command that doesn't port unnecessary features, and then restore on v160 forwards), to entice users not to remain on v159 just because the manual overload of migrating to the new platform.

MichaIng, 2018-01-01:
@WolfganP 
> to entice users not to remain on v159 just because the manual overload of migrating to the new platform.

This were also my concerns, but providing a full migration script, easily breaks the advantage of the reset and makes a lot or work, without anyone being sure if the outcome is what was expected on the individual machine.
As far as I can think of now, this would require migration scripts individually for each software we provide. I will provide one for Nextcloud+ownCloud, but I guess for other software(data+configs) it would need someone who has enough knowledge about the related software and is willing to coding.

MichaIng, 2018-01-02:
@Fourdee 
To hold the other growing v6.0 related threads at bay, I created the new `v6.0` label, to indicate which issues are (or seem to be) related to v6.0 development.

Not to muddle up with v6.0 milestone, which indicates, what we want to solve with v6.0 (but is an issue from prior releases).

Fourdee, 2018-01-14:
Note:
@MichaIng @k-plan 
v6.1 will be reserved solely for bug fixes, if required. To ensure we can patch any issues our users may experience with the vast changes in v6.0

MichaIng, 2018-01-20:
@Fourdee @k-plan 
ownCloud/Nextcloud migration script is ready for testing: https://github.com/Fourdee/DietPi/issues/1413
Automatic recovery on fresh ownCloud/Nextcloud installation will follow. Will review and test this again first.

blogabe, 2018-01-23:
How's the timing looking for v6.0?  Since dietpi-update won't automatically upgrade to 6.0, should we check the main dietpi.com for the release update?

Fourdee, 2018-01-24:
@blogabe 

Due date (release date) for v6.0 is Sunday 28th Jan: https://github.com/Fourdee/DietPi/milestone/64

Should be fee-sable, minus any RL stuff that crops up in-between 😃

MichaIng, 2018-01-24:
[image]

Congratulations, you are allowed to buy something from us 😆.

k-plan, 2018-01-24:
@Fourdee 

> Surprising email arrived today, looks legit lmao :rofl:

first I believe it is a joke.

> Your Best Linux Distributions Recognition Preview is now available.
> As Published In: Makeuseof.com, November 2017.

But:

> **The Best Linux Operating Systems**
> Christian Cawley November 24, 2017
> 
> **6 Lightweight Operating Systems for Your Raspberry Pi**
> Christian Cawley November 29, 2017
> 
> © 2018 MakeUseOf. All Rights Reserved.

https://www.makeuseof.com/?s=Dietpi

[The Best Linux Operating Systems](https://www.makeuseof.com/tag/best-linux-distributions/#raspberry)

[6 Lightweight Operating Systems for Your Raspberry Pi](https://www.makeuseof.com/tag/lightweight-operating-systems-raspberry-pi/)

MichaIng, 2018-01-24:
@k-plan 
Nice, should have taken a closer look.

But the offer to buy some wall plaque there should still be taken as a joke 😄.

tatoosh, 2018-01-28:
Can i use G_AGI with installiert id to install Software without dietpi-software menu?

Fourdee, 2018-01-28:
@tatoosh 

> Can i use G_AGI with installiert id to install Software without dietpi-software menu?

```G_AGI``` is for APT only installations. To install software in dietpi-software without the menu:

Get a full list of software titles
```
dietpi-software list
```

Find the ```id``` of the title you want, eg: to install nextcloud:
```
dietpi-software install 114
```

closed 2018-01-28 by Fourdee

blogabe, 2018-01-28:
I'm still not clear on how to upgrade from 159 to 6.0... is it simply following the instructions at http://dietpi.com/phpbb/viewtopic.php?f=8&t=9#p9 or is there another topic that specifically talks about the migration to 6.0?

If it is following the above instructions, will this reset any of the configurations I've made, for example, the host name?  I'm using DietPi on Allo DigiOne is that helps provide more context.

Fourdee, 2018-01-29:
@blogabe 

> I'm still not clear on how to upgrade from 159 to 6.0... is it simply following the instructions at http://dietpi.com/phpbb/viewtopic.php?f=8&t=9#p9

In short, a fresh installation of the latest DietPi v6.0 image is required, if you want to update. The following guide will walk you through the steps:
http://dietpi.com/phpbb/viewtopic.php?f=8&t=9#p9

k-plan, 2018-01-29:
@Fourdee 

BTW:
- Can you please arrange a new sub forum for DietPi v6.0 in http://dietpi.com/phpbb/viewforum.php?f=5 
 for only v6.0 relevant threads? to get some more clearness over there?
- If possible, a global announcement would be very nice, something [like this](http://dietpi.com/phpbb/viewtopic.php?f=9&t=2735&p=10490#p10490)?

Yes, I know .... reading is difficult for the most ... but anyway.

Thanks in advance.

k-plan, 2018-01-29:
@blogabe

> ... how to upgrade from 159 to 6.0

Sorry but not simple upgrade is possible. You have to start with a fresh image.

But if you are on DietPi v159 with your device, all is working fine and you are happy with it, all looking good, stay on it.

Base system behind DietPi is Debian. Debian Jessie has a [LTS until 2020-04.](https://de.wikipedia.org/wiki/Debian#Ver%C3%B6ffentlichungszyklus) As well for every package which comes via `apt-get` from debian repository. 

Only things you will miss are DietPi updates and DietPi support from now on.

Some software and applications you have installed may have their own live cycle and support cycle.
If a software project stops or died, where is noting we can do.

If you are a `"must-have-always-last-updates-and-feature" junky`, you have to switch to DietPi v6.0.

blogabe, 2018-01-29:
In the process of switching to 6 using the fresh image.  Since I have historically used `dietpi-upgrade` I'm unsure if following the fresh image steps will totally reset the environment to default settings or if it will respect my changes.

Fourdee, 2018-01-29:
@blogabe 

>  I'm unsure if following the fresh image steps will totally reset the environment to default settings or if it will respect my changes.

Its a full reset, fresh installation. Your existing installation settings, would need to be reapplied to the new installation manually.

k-plan, 2018-01-29:
@blogabe 

Hmm ... it's seems to be very difficult to understand.

Go/switch/chance from DietPi v159  ==>> DietPi v6.0 is simple impossible.
No update, no upgrade, no dist-upgrade.

You will loose all software installations and all settings in any way.

So, copy your `/mnt/dietpi_userdata/` to a safe location and start to write a fresh image on a sd card.
Boot this sd card in you device and a fresh/new/first/initial installation will start, without additional software.
After you run trough first setup, you can install whatever you like or need via `dietpi-software`.
So you get DietPi v6.0 and with this, you will get updates, like before with `dietpi-update`

If you don't want this, stay on your current installation.

From now on DietPi v1xx will stay for ever on his last version 159.
You can do as much `dietpi-update` as you like. Have a look:

```
 ───────────────────────────────────────
 DietPi     | 17:19 | Mon 29/01/18
 ───────────────────────────────────────
 V159       | NanoPi Neo (armv7l)
 ───────────────────────────────────────
 IP Address | 192.168.100.126
 ───────────────────────────────────────

root@NanoPi-Neo:~# dietpi-update
 [Info] Checking RootFS R/W access. Please wait...
 [Ok] RootFS R/W access.

 DietPi-Update
─────────────────────────────────────────────────────
 Mode: Checking for DietPi updates
 Please wait...

 [Info] Checking Mirror : http://dietpi.com/downloads/dietpi-update_mirror/master/server_version
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100     6  100     6    0     0     62      0 --:--:-- --:--:-- --:--:--    63
 [Ok] Using update server: http://dietpi.com/downloads/dietpi-update_mirror/master/server_version

 [Ok] No updates required, your DietPi installation is up to date.

 [Info] Current Version : 159
 [Info] Server Version  : 159
root@NanoPi-Neo:~#
```
---------------------

@Fourdee 
puh ... I get the feeling, version numbering `v6.0` was not such a great idea.
We would have better change to `DietPi-Reloaded`, `DiePi Xperience`, `DietPi-X` :smiley:  :see_no_evil:

Fourdee, 2018-01-29:
@k-plan 

> http://dietpi.com/downloads/dietpi-update_mirror/master/server_version

I've updated this, user will now receive a message explaining new image is required.

k-plan, 2018-01-29:
@Fourdee 

> I've updated this, user will now receive a message explaining new image is required.

Nice!  :+1:  :smiley: 

[image]

dee31797, 2018-01-29:
Hi,
First off big fan of the OS.  Been rock solid for me for a while.

I'd like to quadruple check because deleting my whole server to update to the latest supported release is not an option for me.  Other than the previously mentioned "/mnt/dietpi_userdata/" is there another way to copy some files or folders to maintain current operation after reinstall?  Or can you list the new packages that I can install individually?  If migrating between supported and unsupported versions is outside of the project scope I understand and I thank you for the great ride, was fun while it lasted.

Fourdee, 2018-01-29:
@dee31797 

You do **not** need to update if you are happy with your current installation, or, uncomfortable in moving forward with us on this journey. 
However, we cannot support the older v159 versions with latest code.

Fourdee, 2018-01-29:
Thoughts on recent events:
- Charge for DietPi, granting the funds/staff to allow core version upgrade path expectations of the user, such as MS win 95 > 10
- Drop updates, provide new images only which refresh install each time. Like most other projects.
- Continue with user complaints of "minor inconvenience and under appreciation of free software", ignorant to the fact that software may require a reset every 2 years to bring the project forward, and, benefit the users in the long run.

:angry:

dee31797, 2018-01-29:
I am happy with the current install and I thank you and the Dietpi team @Fourdee.  I didn't have to pay a dime for using this project which I think is worth much much more. I can't show my appreciation for that enough. However, running this OS as a server, I can't stay on the version that's being marked as obsolete.  At the same time I can't start it over.  I would be ignorant to think windows 95 can be migrated to windows 10 since the versions are decades apart, but as you know this update does not compare to that very extreme case at all.

blogabe, 2018-01-30:
Whoa whoa whoa :)  More than happy with DietPi and the all the work you guys have put in.  No doubt.  Was just looking for clarification since I received an already flashed DietPi install from Allo and only used the built in upgrade command.  Re-flashing is new to me and I wasn't sure about the ramifications.  Certainly I would have preferred a way to migrate certain services I enabled, but starting from fresh isn't the end of the world.

Fourdee, 2018-01-30:
@dee31797 

Thanks for your feedback. We'll take it on-board.

MichaIng, 2018-01-30:
@all 
Hehe, after reading first about the idea of dropping update support for this version, I didn't expect anything different, than many users complaining, or at least feeling left behind/stressed by the needed effort to reflash the DietPi image. I do myself 😉.

But yeah, indeed this step enabled us/massively simplified to do some deep improvements, rearrangements, file and software changes. To track and assure all these via incremental patching would have been a pain and could have easily break in some cases, if manual changes were done.

So I can feel with you guys, make the best of it, take the chance to do e.g. a Debian version jump, try some new software versions and feel happy afterwards with a clean fresh DietPi, which further development benefits of this step.

dee31797, 2018-01-30:
Hi @MichaIng 
Sorry if I sounded complainy, I really was asking if there was an alternate way to upgrade.  Run a lot of OS's and was a bit baffled.  You mentioned "Debian version jump", the most recent example of that going from Jessie to Stretch its very easy to do an in-place upgrade.  I couldn't image wiping my Debian server every two years it's just not done.

Fourdee, 2018-01-30:
>  the most recent example of that going from Jessie to Stretch its very easy to do an in-place upgrade.

Fresh install, yep.

How i see it, on any system with software installed, regardless of DietPi:
- sshd old config needs a new one, yes or no? yes = cant login with root, no = service fails to run
- Failed to upgrade Sysinit to SystemD, due to package X being a dep of package Y with package Z being downloaded from user added customrepo.com/ubuntu and having a version of 5.1 but requires 5.1.0.1. Broken.
- segmentation fault, out of memory as you have manually disabled swapfile, PSU dropped to 3.5v for 100ms, SDcard failing. Broken.
- Owncloud APT upgrade, data loss due to mysql conversion failure during mysql Jessie installed with mariadb Stretch expected, during this upgrade. Next run of Owncloud will reset user data as sql db is reset after being incorrectly upgraded.

End result, fresh install and wasted user time on a hope that APT (and all its available packages) succeeds  without requiring intervention, and, assuming user has made no changes to any conf file that could accelerate its failure.

So, in our own way, our v6.0 image and forced reset, prevents situations like the above.

Fourdee, 2018-01-30:
@dee31797 

Please do not take this post personally https://github.com/Fourdee/DietPi/issues/1355#issuecomment-361427742
Not in any way aimed at you, we've had a few complaints, all of which add up and caught me at the wrong time. A time in which I needed to vent in order to try and comprehend the situation.

Very unprofessional/unacceptable from my end and I apologise.

dee31797, 2018-01-30:
No worries, we just started dating on the internet, don't know each other well enough to get personal ;) kidding

When I mentioned 

> the most recent example of that going from Jessie to Stretch its very easy to do an in-place upgrade

 I was taking about Debian main (or however it's said). If you run straight Debian Jessie or any modern server really, you can update yesterday's version to today's version, like from Jessie to Stretch.  Then from Stretch to whatever and so on.  No love lost if dietpi is going in a different direction do whats best for the project I just wanted to ask about possible upgrade methods before I wiped this thing.

borpin, 2018-01-30:
$0.02 worth;

No upgrade path is disappointing, but sometimes necessary.  It is the old adage of 'throw away a prototype'.

I have been swithering over rebuilding my systems; Stretch was one driver but this is the final one.  I think DietPi is an excellent Distro and this change will, I think move it forward big time.

Anyone with a 'I can't rebuild my system' argument is, as you pointed out daft.  Everyone has to rebuild at some point and often it is beneficial.  Document what you do and then you can recreate it :)

Keep going and rubber ear the complainers.  Emphasise that updating the underlying OS is still possible.

dee31797, 2018-01-30:
Hi @borpin I don't agree with .02 worth, at least 10 times that! Kidding Dietpi has been a priceless OS for me.
Not rebuilding isn't silly, unless of course your goal is to rebuild, it's just a different perspective.  Perspectives are created based on your background and environment. Is Dietpi's perspective wrong, of course not noone's is, just different.  Me being an old enterprise geezer, you take good backups, 
maintain up time, roll your OSs.  I've done Win2008 to 2012 to 2016, or Ubuntu in a shorter duration but similar manner. I have done some rebuilds with virtual machines, keep the old one up while I move from to a new one, but for me that's usually a platform change like debian to centos.  To wipe centos and then install centos, well I'll say it doesn't fit my perspective. But I do agree with your comment I can rebuild, and I have!  I had to switch platforms since this rebuild made my backups kinda worthless, created a large amount of downtime and I'm still not 100% (main services are running that I can't live without).  I love the os so much I'm going to keep it bookmark, and I'll probably still use it for one of my RPi zero Ws, now that I have a better understanding of the project perspective I know the proper use case going forward.

borpin, 2018-01-30:
Just one other thought - Just tried to update an install currently on v158 and it baulked.  Does this mean that anyone on an older version cannot even update to the last version - v159?

I'd be inclined to have a v160 to remove the Christmas tree and update the banner message.

k-plan, 2018-01-30:
@borpin 

:christmas_tree:  - Remover:

http://dietpi.com/phpbb/viewtopic.php?f=9&t=2689#p10362

```
### DietPi XMAS tree remover ###
cp /DietPi/dietpi/dietpi-banner /DietPi/dietpi/dietpi-banner.XMAS-tree.bak
wget https://raw.githubusercontent.com/Fourdee/DietPi/521f611a69b9cc41ca7ac1f07ccc18182c5a4eb0/dietpi/dietpi-banner -O /DietPi/dietpi/dietpi-banner
chmod +x /DietPi/dietpi/dietpi-banner
clear
/DietPi/dietpi/login
echo -e '[\e[32m  OK  \e[0m] Backup XMAS tree and housekeeping.\n'
Done!
```
or as script: ` XMAS_TREE_REMOVER_FOR_DIETPI.sh`

[XMAS_TREE_REMOVER_FOR_DIETPI.zip](https://github.com/Fourdee/DietPi/files/1678777/XMAS_TREE_REMOVER_FOR_DIETPI.zip)

k-plan, 2018-01-30:
@borpin 

> Just tried to update an install currently on v158 ...

Hmm ... must be a RPi Jessie? Correct?

Do you want/need/search a Jessie DietPi image for RPi? 
Something like: `DietPi_v6.0_RPi-ARMv6-Jessie.img` :question:

borpin, 2018-01-31:
@k-plan I think there should be an option for folk who do not want to migrate to V6 to at least be able to move to v159 - the last, and greatest, of that series!  In this case, for me, I'm not going to move.  The whole OS is on an HDD so rebuilding that is not an option at this time.  It works fine as a data gatherer for my weather station (hence the HDD as there are frequent writes).

blogabe, 2018-02-01:
@Fourdee Allo image is spot on and works.  I had downloaded the generic RPi image from the download page, not the specific Allo Web GUI image.  Reflashed and everything is as it should be.  Letting you know here since I don't have an account at computer audiophile and I see that thread has many questions about the upgrade.  Either way, for those of us using dietpi w/ Allo, reflashing and following the instructions at http://dietpi.com/phpbb/viewtopic.php?f=8&t=2317&p=8869#p8869 will work.

MGWGIT, 2018-02-17:
Why in v.6 I didn't see a storage with configuration files such as dietpi.txt after flashing an image?
Windows suggest only to format this drive...

k-plan, 2018-02-17:
@MGWGIT 

> Why in v.6 I didn't see a storage with configuration files such as dietpi.txt after flashing an image?

Which image?

ARMbian based images, build using ARMbian build tools?

https://github.com/Fourdee/DietPi/issues/1385#issue-287129015

Please have a look here:

https://github.com/Fourdee/DietPi/issues/1465#issuecomment-362912009

https://github.com/Fourdee/DietPi/issues/1423#issue-291560616

https://github.com/Fourdee/DietPi/issues/1385#issuecomment-360261137

MGWGIT, 2018-02-17:
@k-plan thanks, I think there are because Windows can't read EXT4...
I try to configure image for Orange pi zero.

k-plan, 2018-02-17:
> thanks, I think there are because Windows can't read EXT4...

Yes, that's correct. 
Read ext4 file systems form Windows is not such a big problem. 
There are some free file system drivers.
But we don't see a practicable way to edit and save file on ext4 file systems direct from Windows.

eunito, 2018-02-21:
Means I have to reinstall all from "ground 0"?
So many things... :/

JimLee1996, 2018-03-04:
Is NanoPi Neo (armv7l) out of the supported list?

There is no download link at dietpi.com/downloads.

:-(

mzeichner, 2018-03-11:
Is there anyplace (link to) that I can fins detailed instructions on how O can update an rpi runninn version 158 to 160?