MichaIng/DietPi #1357: DietPi-Software | Tonido/General: Fix installation/s, as G_CHECK_URL now breaks dietpi-software

MichaIng, 2017-12-30:
Just found this, as I wanted to fix uninstallations for a similar reasons.

G_CHECK_URL now terminates whole dietpi-software, thus on the one hand, Error_NoConnection_NoInstall is not needed anymore, on the other hand, I found at least one case, where, based on device, the check gets an empty string, thus has to fail, before other devices are handled.

The check has to move into device/distro check statement now. I will fix in this case, but I just checked for `dpkg -i` installations so far.

MichaIng, 2017-12-30:
Ah sorry, I was mistaken, the URL check in that case do occur after all supported hw architectures were checked.

Still, we should go through the installations and at least remove the `Error_NoConnection_NoInstall ` by times 😉. I adjust issue name to reflect this task as part of dietpi-globals.

Fourdee, 2018-01-02:
Ok, so the reason of leaving ```Error_NoConnection_NoInstall``` in:
- Allows us to add the software title/name to the error ```		export G_ERROR_HANDLER_ONERROR_USERMSG="Unable to Install ${aSOFTWARE_WHIP_NAME[$INSTALLING_INDEX]}: ${aSOFTWARE_WHIP_DESC[$INSTALLING_INDEX]}\n\n$INSTALL_URL_ADDRESS is offline and/or unreachable." ```

I'll see if we can change/remove it.

MichaIng, 2018-01-02:
@Fourdee 
At the moment in case of failure, the script would end directly with G_CHECK_URL and not going further into the if-then-else statement, if I'm not wrong?

But yes, we could add the G_ERROR_HANDLER_ONERROR_USERMSG at the beginning of installation+configuration+purging? Or adjust the title to reflect the exact software that failed.

Fourdee, 2018-01-03:
@MichaIng 

> At the moment in case of failure, the script would end directly with G_CHECK_URL and not going further into the if-then-else statement, if I'm not wrong?

Yep, correct 👍 

> But yes, we could add the G_ERROR_HANDLER_ONERROR_USERMSG at the beginning of installation+configuration+purging? Or adjust the title to reflect the exact software that failed.

Yep, was thinking about this aswell, a small function would be ideal that we can call, prior to installing each item. The function would set ```G_ERROR_HANDLER_ONERROR_USERMSG``` to contain the software title.
I'll take a look.

Fourdee, 2018-01-03:
@MichaIng 

Believe this is now resolved, will mark as closed. Please reopen if required.

closed 2018-01-03 by Fourdee

MichaIng, 2018-01-03:
@Fourdee 
Great work, looks well cleaned up.