MichaIng/DietPi #1358: issue with plex installation

pgferr, 2017-12-30:
## Creating a bug report/issue:
[dietpi-software_apt.log](https://github.com/Fourdee/DietPi/files/1594861/dietpi-software_apt.log)
[image]

#### Required Information:
- DietPi Version: DietPi_OdroidC2-armv8-(Stretch).7z
- SBC Device: OdroidC2
- Power supply used (EG: 5v 1A RAVpower)?
- SD card used:Sandisk ultra
- Distro (EG: Jessie) | ```uname -a```?

#### Additional Information (if applicable):
- Software title?
- Can this issue be replicated on a fresh installation of DietPi?

#### Expected behaviour:

#### Actual behaviour:

#### Steps to reproduce:

#### Did you submit a ```dietpi-bugreport```?

#### Extra details:

MichaIng, 2017-12-31:
Just to support investigation:
- I guess you tried `apt --fix-broken install` as suggested by the log?
- https://github.com/Fourdee/DietPi/blob/testing/dietpi/dietpi-software#L6402
- http://dev2day.de/pms/dists/stretch/main/binary-armhf/Packages
- For now I don't see where the dependency issue could come from. `libc6` and `curl` should be both clearly installed. But I don't know which then again needs `:armhf` packages.

pgferr, 2017-12-31:
i think that i have solved:

export LD_LIBRARY_PATH=/lib/arm-linux-gnueabihf:/usr/lib:/usr/local/lib:/opt/vc/lib
apt-get install libc6:armhf

MichaIng, 2017-12-31:
@pgferr 
Ah so it was related to the LD_LIBRARY_PATH issue, as mentioned here: https://github.com/Fourdee/DietPi/issues/1329

I will mark as closed, reopen on demand.

closed 2017-12-31 by MichaIng