MichaIng/DietPi #1360: Shairport Sync ERROR

medapayne, 2017-12-31:
Hello i have a problem when i try to install "Shairport Sync"
I install without problem SqueezeBox and SqueezeLite but for Shairport Sync don't work
I try 3 fresh install but i have a same problem.

I hope you can solved the problem for me

## Creating a bug report/issue:

Apt-get has failed: │
│ - Apt string = openssl libsoxr0 libavahi-client3 libtool │
│ libconfig9 libpopt0 libdaemon0 --no-install-recommends │
│ - Error code = 100 │
│ - Software Title = Shairport Sync: airplay audio player with │
│ multiroom sync │
│ - Apt log = /etc/dietpi/logs/dietpi-software_apt.log │
│ │
│ Please report this issue to DietPi (INCLUDING THE APT LOG) so it │
│ can be investigated. │
│ │
│ DietPi-Software will now exit. 

#### Required Information:
- DietPi Version : 
- SBC Device : Orange Pi Zero 512MB
- Power supply used : 5V
- SD card used: Sandisk ultra)
- Distro : IDK

#### Additional Information (if applicable):
- Software title : Shairport Sync
- Can this issue be replicated on a fresh installation of DietPi: same problem after 3 fresh install

MichaIng, 2017-12-31:
@medapayne 
Could you try `unset LD_LIBRARY_PATH` before starting Shairport Sync installation? We accidentally produced an error with v159, that affects different APT installations.

In case add the content of your `/etc/dietpi/logs/dietpi-software_apt.log` as mentioned within the error, to allow a closer look.

medapayne, 2018-01-01:
Hello again 

this is the /etc/dietpi/logs/dietpi-software_apt.log : 

root@SalledeBain:/etc/dietpi/logs# nano dietpi-software_apt.log
  GNU nano 2.2.6         File: dietpi-software_apt.log

Preconfiguring packages ...
(Reading database ... ^M(Reading database ... 5%^M(Reading database ... 10%^M(R$
Preparing to unpack .../libc6_2.19-18+deb8u10_armhf.deb ...

LD_LIBRARY_PATH contains the traditional /lib directory,
but not the multiarch directory /lib/arm-linux-gnueabihf.
It is not safe to upgrade the C library in this situation;
please remove the /lib/directory from LD_LIBRARY_PATH and
try again.

dpkg: error processing archive /var/cache/apt/archives/libc6_2.19-18+deb8u10_ar$
 subprocess new pre-installation script returned error exit status 1
Errors were encountered while processing:
 /var/cache/apt/archives/libc6_2.19-18+deb8u10_armhf.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

I hope you can solved the problem

MichaIng, 2018-01-01:
@medapayne 
> LD_LIBRARY_PATH contains the traditional /lib directory,
> but not the multiarch directory /lib/arm-linux-gnueabihf.
> It is not safe to upgrade the C library in this situation;
> please remove the /lib/directory from LD_LIBRARY_PATH and
> try again.

As I thought: do `unset LD_LIBRARY_PATH` and try again, in case a reboot is also helpful in between.

medapayne, 2018-01-01:
ok i try now...

MichaIng, 2018-01-01:
@medapayne 
Yes, did APT part of installation run through now?

medapayne, 2018-01-01:
all is GOOD

[image]

Thank you
For my experience can you explain me what's its use "unset LD_LIBRARY_PATH" ?

k-plan, 2018-01-01:
> all is GOOD
>Thank you

So I will close.
Please reopen, if you need. Thanks

closed 2018-01-01 by k-plan

medapayne, 2018-01-02:
THX for all guys...

DavidPower, 2018-01-20:
I'm encountering this exact problem.

The suggested work-around/fix isn't working for me. I've tried it several times and in several permutations.

Also... Despite the shairport install failing, my /etc/dietpi/logs/dietpi-software_apt.log file is always empty.

Can you offer any advice on how I might get around this?

Thanks,
dp

Fourdee, 2018-01-20:
@DavidPower 

Hi,

Please try the updated RC v6.0 image for your device (due to release next week):
http://dietpi.com/downloads/testing/v6.0/

If problems persist with the installation, you should receive an error prompt which you can copy+paste or screenshot and attach here.

DavidPower, 2018-01-20:
Dan,

 

The exact error message I receive is this:

 

http://ftp.us.debian.org/debian/pool/main/o/openssl/libssl1.0.0_1.0.1t-1+deb8u6_armhf.deb is offline and/or unreachable.

 

I’ve confirmed that specific Debian package doesn’t exist.

 

Does the v6.0 image address this issue or is there a manual change I can make (to avoid pulling my Pi Zero W board out of its tight, little case!) to get this working?

 

Thanks,

dp

 

http://dietpi.com/downloads/testing/v6.0/

Fourdee, 2018-01-21:
@DavidPower 

> Does the v6.0 image address this issue

Yes, fixed in v6.0, we've hosted the required libssl package on dietpi.com
ref: https://github.com/Fourdee/DietPi/issues/1303