MichaIng/DietPi #1361: Having sensible defaults for dietpi.txt parameters

TerryE, 2017-12-31:
- DietPi Version 159
- SBC / Distro: RPi 3 / Jessie)

Dan, this is a minor point so please feel free to close it or action as you wish.  

Your `dietpi-config` can die if a required parameter is missing from the `dietpi.txt` file.  I came across this because my practice is to remove all the redundant embellishments  from my ini and config files becaue I find clean files easier to read.  In my case I don't set up a DietPi proxy so I deleted the entire proxy section including the line
```
proxy_enabled=0
```
but this then caused the config script to fail at line 3081 on the test `(( $PROXY_ENABLED == 1 ))` because the variable `PROXY_ENABLED` was null.  This initialised to a default **-1** at line 5224, but is reset by `Load_Proxy_Vars()` at lines 5231:5239 and if the config file doesn't include the `proxy_enabled` parameter, this resets the default to null.   The standard pattern for avoiding this is to insert 
```
PROXY_ENABLED=${PROXY_ENABLED:--1}
```
after line 5233 and ditto for any other numeric parameters then the numeric evaluations won't fail.

Fourdee, 2017-12-31:
@TerryE 

v6.0, ```dietpi.txt``` entries are checked during ```dietpi-update``` (and therefore, first run of device).
https://github.com/Fourdee/DietPi/issues/1292#issuecomment-350901663

After that point:
As ```dietpi.txt``` contains the default variable values (even if not required), they should never be deleted by the user.

TerryE, 2017-12-31:
@Fourdee  Dan, the usecase that I was running was an initial install of Stretch where I tailored the dietpi.txt file to do a minimal hands-free install which I then followed by using SSH to use dietpi-launcher to load the rest, and your comment is that this error path is avoided as of V6.0.  I agree that this isn't worth a backport to legacy versions, so I'll close this.

closed 2017-12-31 by TerryE