MichaIng/DietPi #1363: Squuezelite & USBridge ALSA device

badboygolf16v, 2017-12-31:
I have an Oppo BDP-105 as USB DAC being fed by an Allo USBridge. The OS is DietPi v 157 with Squeezelite v1.8.7-1007 installed.

The problem I'm having is that when the Oppo is restarted and the USBridge is not restarted, the ALSA USB DAC device doesn't reappear so Squeezelite reports this error:

```
ALSA _snd_pcm_hw_open:1713 Invalid value for card
[16:31:01.040296] output_thread:614 waiting for device hw:CARD=DAC,DEV=0 to return
```

A powercycle restart of the USBridge resolves the issue.

`reboot` and `systemctl restart squeezelite.service` do not work and hang the system.

Fourdee, 2018-01-01:
Hi,

@badboygolf16v 

Many thanks for the report.

> The problem I'm having is that when the Oppo is restarted and the USBridge is not restarted, the ALSA USB DAC device doesn't reappear so Squeezelite reports this error:

Unfortunately we see this with a few DAC's, that fail to handle USB power/reset correctly. 
It would appear the OPPO does not correctly reset the USB connection and its state during a restart. This is a limitation with the DAC itself, and must be fixed by the manufacture (ideally via firmware update).

I'll mark this as closed, please reopen if required.

badboygolf16v, 2018-01-02:
Hi @Fourdee 

The same scenario when running an Rpi3 as a USB transport works perfectly, so I don't think it's the case that there is a problem with the Oppo.

If I install piCorePlayer on an RPi3 and set the Squeezelite audio output to the Oppo as a USB DAC, I can restart the Oppo and Squeezelite will continue to work perfectly.

Any suggestions?

Fourdee, 2018-01-03:
@badboygolf16v 

Intresting, could be kernel related on Sparky.

Please try installing the latest kernel to see if it resolves the issue (copy+paste all into term):
```
wget https://raw.githubusercontent.com/sparky-sbc/sparky-test/master/dragon_fly_check/uImage -O /boot/uImage
wget https://raw.githubusercontent.com/sparky-sbc/sparky-test/master/dragon_fly_check/3.10.38.bz2 -O package.tar
tar xvf package.tar -C /lib/modules/
rm package.tar

cat << _EOF_ > /DietPi/uEnv.txt
uenvcmd=setenv os_type linux;
bootargs=earlyprintk clk_ignore_unused selinux=0 scandelay console=tty0 loglevel=1 real_rootflag=rw root=/dev/mmcblk0p2 rootwait init=/lib/systemd/systemd aotg.urb_fix=1 aotg.aotg1_speed=0
_EOF_
```

Failing that, try changing ```aotg.aotg1_speed=1``` in ```/DietPi/uEnv.txt``` for USB compatibility.

badboygolf16v, 2018-01-03:
Thanks for info. It didn't make any difference to the behaviour.

Should I contact Allo directly?

Fourdee, 2018-01-04:
@badboygolf16v 

> Should I contact Allo directly?

Yep, probably the next path to take 👍

badboygolf16v, 2018-01-08:
No joy contacting Allo via their website, I've posted on Computer Audiophile:
[https://www.computeraudiophile.com/forums/topic/32132-allo-sparky-usbridge/?do=findComment&comment=763299](https://www.computeraudiophile.com/forums/topic/32132-allo-sparky-usbridge/?do=findComment&comment=763299)

sparkysbc, 2018-01-09:
Possibility :  while playing Music you can't poweroff USB dac , in this case Squeezelite locked with USB port. once locked with USB port need to restart the setup. 
stop the playing music on player (LMS )  or disable squeezelite on Dietpi webGUI , then switch off USB dac.
please verify this and confirm.

badboygolf16v, 2018-01-13:
Changing aotg.urb_fix=0 from 1 resolves this issue.

Fourdee, 2018-01-14:
Confirmed v159 required for default + patched ```aotg.urb_fix=1```:
https://github.com/Fourdee/DietPi/blob/master/dietpi/patch_file#L2907-L2921

Will mark as closed.

closed 2018-01-14 by Fourdee