MichaIng/DietPi #1366: qBittorrent login error

UnfairGamer, 2018-01-01:
I am using DietPi version 158(Stretch-based) on a RPI 3. 

PROBLEM: 
Cant log in to qBittorrent 

EXPLANATION/BACKGROUND INFORMATION:

I started using DietPi a very few days ago and today i installed qbittorrent for use via the web interface. I installed the software from the dietpi-software and it installed without an issue. I can connect to the device (192.168.0.30:1340 for qbit...), and the Web UI Login page shows up. What on earth is the password on this thing?! I have tried everything i can, and checked github and other sources and everyone says the uname is root and the password is dietpi. I get the wrong uname/password error on the page when trying this. I have checked the ~/.config/qBittorrent/qBittorrent.conf file and it says `WebUI\Password_ha1=@ByteArray(5e434b0004be6970cf8878fcf929baba)`.

I have not managed to decode this, and i have also tried replacing it with another value(looks like md5 to me?), but the value resets to this upon reboot. I have removed and uninstalled the software via dietpi-software and the problem persists.

I have also tried logging in with the root password(that i have changed). Is there any way to sign in or to change the password? I dont really have the option to do a fresh os install, since i have spent many hours setting up owncloud on this pi and sd card. If nesseccary, i can perform tests on another sd card.

Fourdee, 2018-01-02:
@UnfairGamer 

Thanks for the report 👍 

We'll try to replicate.

Fourdee, 2018-01-02:
Confirmed bug, default login details fail:
http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=70#p2272

Fourdee, 2018-01-02:
This will be fixed in our next release.

To fix now:
- After installing via dietpi-software
- stop services ```dietpi-services stop```
- edit the config file ```nano /root/.config/qBittorrent/qBittorrent.conf```
- Scroll down to bottom and change the following line ```WebUI\Password_ha1=@ByteArray(90c322f4dcabce6e66e8ea8aff218775)```
- Save changes CTRL+X, press Y.
- Restart services ```dietpi-services start```

Fourdee, 2018-01-02:
md5hash of GLOBAL_PW will be used to generate. Tested install fine. Resolved.

closed 2018-01-02 by Fourdee

brumouta, 2018-02-05:
I'm still unable to login to qbittorrent webui, using image downloaded yesterday for odroid C2

tnnuyri, 2018-02-10:
I have the same problem! and already tried to change how and said ... I have the XU4 odroid, but already on the raspberry pi 3 does not happen!! Does anyone know how to solve the problem?

jhbball2002, 2018-02-13:
Confirmed this is still an issue using an odroid-xu4.    Tested with the latest release, v6.1.   Can't login with the suggested username passwords;  the work around above does not solve this issue.

Fourdee, 2018-02-13:
@jhbball2002 

Hi,

This fix is outdated, please see following link for workaround:
https://github.com/Fourdee/DietPi/issues/1499#issuecomment-364769906