MichaIng/DietPi #1371: DietPi-Notify | echo, printf and [Processing] messages

MichaIng, 2018-01-03:
I know it is not an important topic, but I was motivated to experiment a bid around with possibilities to write text into the terminal.

The initial reason was, that I am not too happy with the current behaviour of our [Processing] messages:
- The idea was, that these messages only show up on ongoing process, to inform users, if some process takes longer. A simple animation (moving dot in between the brackets) would be also nice, to prove a not hang system, but that's another topic. Afterwards these messages should be overwritten be hopefully [Ok], otherwise [Error], or another [Info], if the process was simply to trivial/not worth to mention some result.
- Current solution is, to move cursor to beginning of line after [Processing] messages, to allow following messages to overwrite it. But if the following message is shorter, some junk will stay.
- We can force some line clearing before our own notifications (current solution) to avoid this, but if any other message from outside our control appears, this will be not the case.
- If we move the cursor at beginning of our notifications instead of end of [Processing] message, 3rd party messages would not overwrite, but printed onto the same line, which also doesn't look nice.
- If you would keep a newline after [Processing], then our notifications would need to know, that the prior line was a [Processing], then move up and clear that prior line. But this would need some kind of global flag. More about this later.

First of all I looked around for other possibilities to read/write the terminal, but just found `printf` als alternative to `echo`. I always thought it is a way less basic, move overhead command, but tests on my systems gave interesting results:

<details>
<summary>`echo` vs `printf` without newline</summary>

```
root@DietPi:~# time for i in {1..999999}; do echo -n "$i" > /dev/null; done

real    0m13.193s
user    0m10.588s
sys     0m2.528s
root@DietPi:~# time for i in {1..999999}; do echo -n "$i" > /dev/null; done

real    0m13.254s
user    0m10.404s
sys     0m2.764s
root@DietPi:~# time for i in {1..999999}; do printf "$i" > /dev/null; done

real    0m12.515s
user    0m9.916s
sys     0m2.528s
root@DietPi:~# time for i in {1..999999}; do printf "$i" > /dev/null; done

real    0m12.574s
user    0m9.924s
sys     0m2.580s
```
</details>

<details>
<summary>`echo` vs `printf` with newline</summary>

```
root@DietPi:~# time for i in {1..999999}; do echo "$i" > /dev/null; done

real    0m12.804s
user    0m10.312s
sys     0m2.424s
root@DietPi:~# time for i in {1..999999}; do echo "$i" > /dev/null; done

real    0m12.600s
user    0m10.020s
sys     0m2.496s
root@DietPi:~# time for i in {1..999999}; do printf "$i\n" > /dev/null; done

real    0m12.964s
user    0m10.292s
sys     0m2.588s
root@DietPi:~# time for i in {1..999999}; do printf "$i\n" > /dev/null; done

real    0m12.920s
user    0m10.316s
sys     0m2.520s
```
</details>

<details>
<summary>`/bin/echo` vs `/usr/bin/printf`</summary>

```
root@DietPi:~# time for i in {1..**9999**}; do /bin/echo -n 'test' > /dev/null; done

real    0m54.816s
user    0m5.184s
sys     0m21.152s
root@DietPi:~# time for i in {1..**9999**}; do /bin/echo -n 'test' > /dev/null; done

real    0m58.056s
user    0m6.560s
sys     0m22.784s
root@DietPi:~# time for i in {1..9999}; do /usr/bin/printf 'test' > /dev/null; done

real    0m59.054s
user    0m6.908s
sys     0m23.152s
root@DietPi:~# time for i in {1..9999}; do /usr/bin/printf 'test' > /dev/null; done

real    0m57.879s
user    0m6.788s
sys     0m21.996s
```
</details>

Results:
- There was not much difference in performance between echo and printf during my tests.
- It didn't matter, if written to /dev/null, on screen, double quotes static string or dynamic with changing variable, the relative results were everywhere more less the same.
- Never use the binaries, they are waaaaaaay slower than the bash builtin commands (note the maximum
 count) 😆.

=> So finally it doesn't matter which one we use, I was just wondering, as printf by default interprets escapes and has more format features.

As mentioned, I would like to allow every random terminal print to overwrite (and clear) prior [Processing] lines. Thus I experimented with `hooks` of echo and `printf` commands:

<details>
<summary>/etc/bash.bashrc</summary>

```
...
export G_PROCESS_FLAG=0

echo()
{
        (( $G_PROCESS_FLAG == 1 )) && builtin echo -en "\r\33[K" && G_PROCESS_FLAG=0
        builtin echo "$@"
}

export -f echo

printf()
{
        (( $G_PROCESS_FLAG == 1 )) && builtin printf "\r\33[K" && G_PROCESS_FLAG=0
        builtin printf "$@"
}

export -f printf
```
</details>

- Processing messages then set `G_PROCESS_FLAG=1` to allow every echo and printf to clear and overwrite them.
- This leads to about doubled processing time:
<details>
<summary>hooked `echo` vs `builtin echo`</summary>

```
root@DietPi:~# time for i in {1..999999}; do builtin echo "$i" > /dev/null; done

real    0m14.811s
user    0m11.096s
sys     0m3.628s
root@DietPi:~# time for i in {1..999999}; do builtin echo "$i" > /dev/null; done

real    0m14.138s
user    0m11.036s
sys     0m3.016s
root@DietPi:~# time for i in {1..999999}; do echo "$i" > /dev/null; done

real    0m28.475s
user    0m24.856s
sys     0m3.540s
root@DietPi:~# time for i in {1..999999}; do echo "$i" > /dev/null; done

real    0m28.696s
user    0m24.976s
sys     0m3.644s
```
</details>

What I never thought: Printing on the screen or to > /dev/null does not make much difference:
<details>
<summary>echo to terminal/screen vs echo to /dev/null</summary>

```
root@DietPi:~# time for i in {1..999999}; do echo -n 'test' ; done
...
...
...
esttesttesttesttesttesttesttesttesttesttesttesttesttesttesttesttesttesttesttesttesttesttesttesttesttesttesttesttesttesttesttesttesttesttesttesttest
real    0m16.126s
user    0m11.172s
sys     0m2.464s
root@DietPi:~# time for i in {1..999999}; do echo -n 'test' > /dev/null ; done

real    0m14.061s
user    0m11.504s
sys     0m2.484s
```
</details>

Just tested, works well to move those hooks into `dietpi-globals`, without export etc. Then everything is in one place. Finally I moved all our adjustments from `/etc/bash.bashrc` to dietpi-globals and I don't see any reason why we should not do it like this. dietpi-globals anyway need to be sourced for DietPi to run 😃. I can also move occ/ncc commands into globals.

Back to topic, after creating the echo+printf hooks, only a small amount of screen appearance does not correctly clean [Processing] lines. These are some APT command outputs, the ncc/occ command outputs, and if I didn't test wrong, the reason for this is, that actually all dietpi-globals commands are not working in combination with [sudo -u](url):
```
root@DietPi:~# sudo -u root G_DIETPI-NOTIFY 2 'test'
sudo: G_DIETPI-NOTIFY: command not found
```
So the question are:
- Do we want hooks to allow most terminal outputs overwriting our [Processing] messages? 
- If yes, do we want those, we cannot handle, to at least partly overwrite out [Processing] or attach them to the end of the line? Both looks not very nice, but I don't see any good solution.

MichaIng, 2018-01-06:
Further experimenting:
- I created an animation for the processing messages.
- Basically a background process is started, that constantly overprints the first characters to simulate a moving dot within the brackets. A bid like `dietpi-funtime`, but as background precess to allow the script going on, and as unlimited animation, until kill PID is done.
- Then within dietpi-globals I adjusted the mentioned wrappers for echo+printf to kill the background process, if existent and clear the line before actual echo/printf is going on.
- Works nice in most cases, however as mentioned there may be other terminal output commands, that would need a wrapper, and especially other users (also by sudo -u calls) do not use dietpi-globals.
- I tried many things, especially `/etc/profile.d/`, to export the wrappers to other users, but that does not work at least for `sudo -u www-data`.
- `/usr/local/bin/echo` would work, but then of course www-data does not have permissions to kill other users processes 😅.
- Result is:
[image]
- Especially APT output is affected.

So finally the animation is quite nice, but to prevent the shown, if we use it, we should suppress any other output, until our [Ok]/[Error]/[Info] stops the animation precess. If so, then we also do not need the wrapper functions, but simply include e.g. a clear_animation() into Print_Ok/Failure/Info.
If we want to keep apt-get output, we can simply avoid `G_DIETPI-NOTIFY -2` here, and use `+2` instead for [Info]. Would be absolutely not problem, as apt-get output comes constantly, showing that some process is going on 😉.

Fourdee, 2018-01-06:
@MichaIng 

> Should I provide PR, that you can see the result? But should be seen as low priority, as we need to focus on finishing automation script of course.

Yep agree 👍 

Send PR when your ready, i'll look/test when I can.

MichaIng, 2018-01-08:
PR merged, thus mark as closed. Further integration of new `G_DIETPI-NOTIFY` can be done, but no focus, as it's just cosmetic.

closed 2018-01-08 by MichaIng