MichaIng/DietPi #1372: DietPi-Software | Plex Media Server fails to install on Odroid C2

ThomvanDijk, 2018-01-03:
I Installed a fresh image on Odroid C2. After that I did a clean Plex installation and got the following error afterwards:

Apt-get has failed:
 - Apt string = binutils:armhf plexmediaserver-installer:armhf
 - Error code = 100
 - Software Title = Plex Media Server: web interface media
streaming server
 - Apt log = /etc/dietpi/logs/dietpi-software_apt.log

DietPi-Software will now exit.

#### Required Information:
- DietPi v159
- SBC Device: Odroid C2
- Power supply used: 5v 2A
- SD card used: Adata 32 GB
- Distro: Linux DietPi 3.14.79+

#### Additional Information (if applicable):
- Software title: Plex Media Server
- Can this issue be replicated on a fresh installation of DietPi: Yes

#### Steps to reproduce:
1) Flash image on sd card.
2) Select Plex as software to install.
3) Install.

#### dietpi-bugreport:
Reference CODE: 50a50ef2-5021-4e4d-9082-32b8408dc6ba-0 

#### Extra details:
Apt log:

Preconfiguring packages ...
(Reading database ... 16193 files and directories currently installed.)
Preparing to unpack .../libc6_2.19-18+deb8u10_arm64.deb ...

LD_LIBRARY_PATH contains the traditional /lib directory,
but not the multiarch directory /lib/aarch64-linux-gnu.
It is not safe to upgrade the C library in this situation;
please remove the /lib/directory from LD_LIBRARY_PATH and
try again.

dpkg: error processing archive /var/cache/apt/archives/libc6_2.19-18+deb8u10_arm64.deb (--unpack):
 subprocess new pre-installation script returned error exit status 1
Selecting previously unselected package libc6:armhf.
Preparing to unpack .../libc6_2.19-18+deb8u10_armhf.deb ...
De-configuring libc6:arm64 (2.19-18+deb8u7) ...

LD_LIBRARY_PATH contains the traditional /lib directory,
but not the multiarch directory /lib/arm-linux-gnueabihf.
It is not safe to upgrade the C library in this situation;
please remove the /lib/directory from LD_LIBRARY_PATH and
try again.

dpkg: error processing archive /var/cache/apt/archives/libc6_2.19-18+deb8u10_armhf.deb (--unpack):
 subprocess new pre-installation script returned error exit status 1
Errors were encountered while processing:
 /var/cache/apt/archives/libc6_2.19-18+deb8u10_arm64.deb
 /var/cache/apt/archives/libc6_2.19-18+deb8u10_armhf.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

Fourdee, 2018-01-04:
@ThomvanDijk 

Thanks for the report 👍 

This is a known issue, resolved for v6.0.

In the mean time, please copy and paste the following command, to remove the LD_LIBRARY_PATH export:
```
sed -i '/export LD_LIBRARY_PATH=\/lib:\/usr\/lib:\/usr\/local\/lib:\/opt\/vc\/lib/d' /etc/bash.bashrc
```

Reboot system, and re-try the installation:
````
apt-get install -f
dietpi-software install 42
```

ThomvanDijk, 2018-01-04:
Thanks the issue is resolved by your fix!

Fourdee, 2018-01-05:
@ThomvanDijk 

Excellent, i'll mark this as closed.

closed 2018-01-05 by Fourdee