MichaIng/DietPi #1374: issue with home assistant installation

pgferr, 2018-01-05:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version:  V159       | BananaPi Pro (armv7l)
- SBC Device: BananaPi Pro (armv7l)
- Power supply used (EG: 5v 1A RAVpower)?
- SD card used (64gb: Sandisk ultra)
- Distro (EG: Jessie) | ```uname -a```?

#### Additional Information (if applicable):
- Software title: home assistant
- Can this issue be replicated on a fresh installation of DietPi?

my terminal:
 DietPi     | 19:08 | Fri 05/01/18
 ───────────────────────────────────────
 V159       | BananaPi Pro (armv7l)
 ───────────────────────────────────────
 IP Address | 192.168.1.4
 ───────────────────────────────────────

 Welcome to DietPi-Software

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Update & upgrade APT
 Please wait...

Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  ca-certificates curl debconf debconf-i18n debconf-utils
  debian-archive-keyring gpgv initramfs-tools libc6 libcurl3 libcurl3-gnutls
  libdb5.3 libdns-export100 libexpat1 libffi6 libgcrypt20 libgnutls-deb0-28
  libgnutls-openssl27 libgssapi-krb5-2 libicu52 libirs-export91
  libisc-export95 libisccfg-export90 libk5crypto3 libkrb5-3 libkrb5support0
  libldap-2.4-2 libncurses5 libncursesw5 libpam-modules libpam-modules-bin
  libpam-systemd libpam0g librtmp1 libssl1.0.0 libsystemd0 libtasn1-6
  libtinfo5 libudev1 linux-image-sun7i linux-u-boot-bananapipro-default
  locales multiarch-support openssl passwd sensible-utils sudo systemd
  systemd-sysv tzdata udev unzip wget wpasupplicant
54 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 43.1 MB of archives.
After this operation, 10.2 kB of additional disk space will be used.
Get:1 http://security.debian.org/ jessie/updates/main sensible-utils all 0.0.9+deb8u1 [10.9 kB]
Get:2 http://ftp.debian.org/debian/ jessie/main libc6 armhf 2.19-18+deb8u10 [3,965 kB]
Get:3 http://apt.armbian.com/ jessie/main linux-image-sun7i armhf 5.35 [14.0 MB]
Get:4 http://ftp.debian.org/debian/ jessie/main libtinfo5 armhf 5.9+20140913-1+deb8u2 [271 kB]
Get:5 http://ftp.debian.org/debian/ jessie/main libncursesw5 armhf 5.9+20140913-1+deb8u2 [103 kB]
Get:6 http://ftp.debian.org/debian/ jessie/main libncurses5 armhf 5.9+20140913-1+deb8u2 [78.5 kB]
Get:7 http://ftp.debian.org/debian/ jessie/main libudev1 armhf 215-17+deb8u7 [53.1 kB]
Get:8 http://ftp.debian.org/debian/ jessie/main udev armhf 215-17+deb8u7 [853 kB]
Get:9 http://ftp.debian.org/debian/ jessie/main initramfs-tools all 0.120+deb8u3 [96.7 kB]
Get:10 http://ftp.debian.org/debian/ jessie/main libsystemd0 armhf 215-17+deb8u7 [83.1 kB]
Get:11 http://ftp.debian.org/debian/ jessie/main systemd armhf 215-17+deb8u7 [2,316 kB]
Get:12 http://ftp.debian.org/debian/ jessie/main systemd-sysv armhf 215-17+deb8u7 [36.7 kB]
Get:13 http://ftp.debian.org/debian/ jessie/main libpam-systemd armhf 215-17+deb8u7 [115 kB]
Get:14 http://ftp.debian.org/debian/ jessie/main debconf-utils all 1.5.56+deb8u1 [57.6 kB]
Get:15 http://ftp.debian.org/debian/ jessie/main debconf-i18n all 1.5.56+deb8u1 [206 kB]
Get:16 http://ftp.debian.org/debian/ jessie/main debconf all 1.5.56+deb8u1 [167 kB]
Get:17 http://ftp.debian.org/debian/ jessie/main libpam0g armhf 1.1.8-3.1+deb8u2+b1 [121 kB]
Get:18 http://ftp.debian.org/debian/ jessie/main libpam-modules-bin armhf 1.1.8-3.1+deb8u2+b1 [101 kB]
Get:19 http://ftp.debian.org/debian/ jessie/main libpam-modules armhf 1.1.8-3.1+deb8u2+b1 [290 kB]
Get:20 http://ftp.debian.org/debian/ jessie/main libdb5.3 armhf 5.3.28-9+deb8u1 [601 kB]
Get:21 http://ftp.debian.org/debian/ jessie/main libgcrypt20 armhf 1.6.3-2+deb8u4 [354 kB]
Get:22 http://ftp.debian.org/debian/ jessie/main libssl1.0.0 armhf 1.0.1t-1+deb8u7 [867 kB]
Get:23 http://ftp.debian.org/debian/ jessie/main libgnutls-openssl27 armhf 3.3.8-6+deb8u7 [140 kB]
Get:24 http://ftp.debian.org/debian/ jessie/main libtasn1-6 armhf 4.2-3+deb8u3 [43.3 kB]
[... 95 lines cut ...]
 [Failed] The apt cache may be corrupt, apt mirror offline, or you have held broken packages. DietPi-Software will now exit.

 [Info] Apt Log:
Preconfiguring packages ...
(Reading database ... 73987 files and directories currently installed.)
Preparing to unpack .../libc6_2.19-18+deb8u10_armhf.deb ...

LD_LIBRARY_PATH contains the traditional /lib directory,
but not the multiarch directory /lib/arm-linux-gnueabihf.
It is not safe to upgrade the C library in this situation;
please remove the /lib/directory from LD_LIBRARY_PATH and
try again.

dpkg: error processing archive /var/cache/apt/archives/libc6_2.19-18+deb8u10_armhf.deb (--unpack):
 subprocess new pre-installation script returned error exit status 1
Errors were encountered while processing:
 /var/cache/apt/archives/libc6_2.19-18+deb8u10_armhf.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

 [Info] Please try:
 [Info] - Fixing apt package errors with 'apt-get install -f'
 [Info] - Change apt mirror: 'dietpi-config' > 'Networking: NAS/MISC' > 'Apt Mirror'
 [Info] - Clear and refresh apt 'apt-get update'
 [Info] - If problems persist, please create a Git ticket so DietPi can investigate the issue:
 - https://github.com/Fourdee/DietPi/issues

k-plan, 2018-01-05:
@pgferr 

> please remove the /lib/directory from LD_LIBRARY_PATH and try again.

Can you please try:

```
echo $LD_LIBRARY_PATH

unset LD_LIBRARY_PATH

echo $LD_LIBRARY_PATH

ldconfig -v

```

Can you please as well  have a look here:

https://github.com/Fourdee/DietPi/issues/1329

https://github.com/Fourdee/DietPi/issues/1360

idfxken, 2018-01-06:
Had the same issue, just as some others, see: http://dietpi.com/phpbb/viewtopic.php?f=11&t=2632
Problem was fixed with: 

```
export LD_LIBRARY_PATH=/lib/arm-linux-gnueabihf:/usr/lib:/usr/local/lib:/opt/vc/lib
ldconfig
```

Fourdee, 2018-02-22:
This is now resolved for v6.x, was a previous bug our end with ```LD_LIBRARY_PATH``` exporting.

Marking as closed.

closed 2018-02-22 by Fourdee