MichaIng/DietPi #1375: DietPi-Software | Pi-hole: update process

k-plan, 2018-01-05:
> http://dietpi.com/phpbb/viewtopic.php?f=9&t=2631#p10195

This did not help: http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=20#p174

- if it is ONLY a FTL version update, 
```
:~# pihole -v
  Pi-hole version is v3.2 (Latest: v3.2)
  AdminLTE version is v3.2 (Latest: v3.2)
  FTL version is vDev-437af07 (Latest: v2.13.1)
```
you can simple start `pihole -up` without doing anything else. (tested -work)

- if it is a pi-hole core and/or webinterface update
```
:~# pihole -v
  Pi-hole version is v3.2 (Latest: v3.2.1)
  AdminLTE version is v3.2 (Latest: v3.2.1)
  FTL version is vDev-437af07 (Latest: v2.13.1)
```

you have to do:

```
##  have a look, what is to  update
 pihole -v

## if only FTL version have to be updated
# pihole -up
## finished

## if it is a core/AdminLTE version update or more - Backup lighttpd.conf
mv /etc/lighttpd/lighttpd.conf /etc/lighttpd/lighttpd.conf.bak

## https://discourse.pi-hole.net/t/why-does-the-pi-hole-installer-produce-the-error-unable-to-complete-update-contact-pi-hole/4159
cd /etc/.pihole;git reset --hard
cd /var/www/html/admin;git reset --hard
cd $HOME

## now update pi-hole
pihole -up

## Restore lighttpd.conf
cp /etc/lighttpd/lighttpd.conf.bak /etc/lighttpd/lighttpd.conf

## Restart all running services
dietpi-services restart

## finished
```

Is it possible to get something like this in a script (eg. with a nice menu) and   if `cat /DietPi/dietpi/.installed | grep 93` =2 , make it accessible via  eg. `dietpi-pihole-update`  for a user frindly update process without hassle?

Yes, if Pi-hole will change something in future, we get problems. But this is not new and  Fourdee has delete his own pi-hole repo, this will happend from time to time.

https://github.com/Fourdee/DietPi/issues/1313

Fourdee, 2018-02-02:
@k-plan 

> Is it possible to get something like this in a script (eg. with a nice menu) and if cat /DietPi/dietpi/.installed | grep 93 =2 , make it accessible via eg. dietpi-pihole-update for a user frindly update process without hassle?

We could, however, it relies on the user applying our update method, and, assuming PiHole does not change the update process.

The current solution is for a fresh install, uninstall then install PiHole.

mmm, what we could do is replace ```/etc/lighttpd/lighttpd.conf``` during ```dietpi-services start``` with our default,
+ ```sed -i '/^server.document-root/c\server.document-root        = "/var/www"' /etc/lighttpd/lighttpd.conf```

But if we make any changes to the ```dietpi-software``` configuration of ```/etc/lighttpd/lighttpd.conf``` it may break the workaround.

The ideal solution would be for PiHole to not overwrite ```/etc/lighttpd/lighttpd.conf``` in our case, but we cant really force that on them, unless they want to support DietPi as a install for their users.

k-plan, 2018-02-02:
@Fourdee 

> We could, however, it relies on the user applying our update method, and, assuming PiHole does not change the update process.

How can I promise this?  Never ever ...  😞 

> The current solution is for a fresh install, uninstall then install PiHole.

Rebooting, all inital dialog stuff, reconfigure, please no.
Sorry to say, but that is not a solution. 👎 

> unless they want to support DietPi as a install for their users.

Really? 😄   Don't think this will ever happen.  :rofl: 

> All this, assuming the /etc/lighttpd/lighttpd.conf overwrite is the only issue preventing our users successfully updating via pihole update?

I can't say. I'm not familiar with this whole `git Client` `git clone`.

This tinker workaround coming from researching and I have tested a little bit.
Works for me at the moment and I be happy with it.

But okay, leave it as it is.

Fourdee, 2018-02-02:
@k-plan 

> http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=20#p174
Updating PiHole:
PiHole overwrites the lighttpd.conf during updates, so please backup your lighttpd.conf before updating:
Backup lighttpd.conf
mv /etc/lighttpd/lighttpd.conf /etc/lighttpd/lighttpd.conf.bak
PiHole can then be updated by running pihole -up
Restore lighttpd.conf
CODE: SELECT ALL
mv /etc/lighttpd/lighttpd.conf.bak /etc/lighttpd/lighttpd.conf

The online doc references an update procedure, i'll try to verify this tomorrow.

k-plan, 2018-02-02:
Tested and did not work ... (by my self)

Pi-hole v3.1 => Pi-hole v3.2

Fourdee, 2018-02-02:
@k-plan 

Ok, this is one of most popular installs. 
Lets find out exactly whats needed to update and be compatible, then we can discuss possible implementations.

Note to self:
- Global for ```pihole``` ?

k-plan, 2018-02-02:
@Fourdee 

👍 

> then we can discuss possible

Sorry, I don't want to discuss with you. You can do or not do what ever you like. 😄 
It's okay for me.

Long day, let's  pack up. Good night.

MichaIng, 2018-04-09:
@k-plan @Fourdee 
Ah found it, seems it has become even more complicated as Lighttpd is installed now by PiHole install script, which breaks other web servers in case?

I see the idea of recovering our lighttpd config was already there. Perhaps it is possible to use pihole-FTL.service.d/dietpi.conf to include ExecStartPre and ExecStopPost to backup and recover the current lighttpd config: https://github.com/Fourdee/DietPi/issues/1623#issuecomment-379768454
If every update includes the services stop and start, this should work and avoids any issue with lighttpd.conf changed by dietpi-software or manually by user.

I would like to test, but does anyone know how to manually lower PiHole version strings to allow updates? Adjusting `/etc/pihole/localversions` does not have any effect 🤔.

k-plan, 2018-04-09:
@MichaIng 

> I would like to test, but does anyone know how to manually lower PiHole version strings to allow updates? Adjusting /etc/pihole/localversions does not have any effect thinking

https://discourse.pi-hole.net/t/how-do-i-revert-to-a-previous-version-of-pi-hole/7168

or

```
# To Revert to master:
pihole checkout master
```

BTW:

- you should have a look onto upcoming new FTLDNS™ Pi-hole’s Faster Than Light Domaon Name Service

https://pi-hole.net/2018/02/22/coming-soon-ftldns-pi-holes-own-dns-dhcp-server/

https://pi-hole.net/2018/03/24/help-us-beta-test-ftldns/

https://github.com/pi-hole/FTL/pull/238

[image]

MichaIng, 2018-04-10:
@k-plan 
Ah good news, this might solve the Stretch incompatibility due to dnsmasq version, we will see.
Good that the service name did not chance, thus the same hooks can be used.
So far I have this:
```
root@DietPi:/var/www/pihole# cat /etc/systemd/system/pihole-FTL.service.d/dietpi.conf
[Service]
ExecStartPre=/bin/bash -c '/var/lib/dietpi/dietpi-software/services/pihole-startpre.sh'
ExecStopPost=/bin/bash -c '/var/lib/dietpi/dietpi-software/services/pihole-stoppost.sh'
root@DietPi:/var/www/pihole# cat /var/lib/dietpi/dietpi-software/services/pihole-stoppost.sh
#!/bin/bash
# As PiHole overwrites our lighttpd.conf, we need to do a backup in case, using pihole-FTL ExecStopPost.

# Do lighttpd.conf backup, if installed via dietpi-software:
grep -q "aSOFTWARE_INSTALL_STATE\[84\]=2" /DietPi/dietpi/.installed && [ ! -f /mnt/dietpi_userdata/lighttpd.conf.bak ] && cp /etc/lighttpd/lighttpd.conf /mnt/dietpi_userdata/lighttpd.conf.bak

exit 0
root@DietPi:/var/www/pihole# cat /var/lib/dietpi/dietpi-software/services/pihole-startpre.sh
#!/bin/bash
# As PiHole installs lighttpd, dhcpcd5 and overwrites our lighttpd.conf,
# we need to purge those packages in case or recover our lighttpd.conf.
# To assure this on installation, reconfiguration and update, we do this on pihole-FTL ExecStartPre.

dpkg --get-selections > /tmp/dpkg-selections

# Purge dhcpcd5, if our default isc-dhcp-client is installed:
grep -q 'dhcpcd5' /tmp/dpkg-selections && grep -q 'isc-dhcp-client' /tmp/dpkg-selections && DEBIAN_FRONTEND=noninteractive apt-get purge -y dhcpcd5

# Purge lighttpd, if another webserver is (to be) installed via dietpi-software:
grep -q "aSOFTWARE_INSTALL_STATE\[8[35]\]=[12]" /DietPi/dietpi/.installed && grep -q 'lighttpd' /tmp/dpkg-selections && DEBIAN_FRONTEND=noninteractive apt-get purge -y lighttpd

# Recover lighttpd.conf, if backup was made:
grep -q "aSOFTWARE_INSTALL_STATE\[84\]=2" /DietPi/dietpi/.installed && mv /mnt/dietpi_userdata/lighttpd.conf.bak /etc/lighttpd/lighttpd.conf

exit 0
root@DietPi:/var/www/pihole# systemctl status pihole-FTL.service
● pihole-FTL.service - LSB: pihole-FTL daemon
   Loaded: loaded (/etc/init.d/pihole-FTL; generated; vendor preset: enabled)
  Drop-In: /etc/systemd/system/pihole-FTL.service.d
           └─dietpi.conf
   Active: active (running) since Tue 2018-04-10 17:40:34 CEST; 16min ago
     Docs: man:systemd-sysv-generator(8)
   CGroup: /system.slice/pihole-FTL.service
           └─18558 /usr/bin/pihole-FTL

Apr 10 17:40:34 DietPi su[18552]: Successful su for pihole by root
Apr 10 17:40:34 DietPi su[18552]: + ??? root:pihole
Apr 10 17:40:34 DietPi su[18552]: pam_unix(su:session): session opened for user pihole by (uid=0)
Apr 10 17:40:34 DietPi pihole-FTL[18511]: FTL started!
Apr 10 17:40:34 DietPi systemd[1]: Started LSB: pihole-FTL daemon.
Apr 10 17:40:34 DietPi dnsmasq[18558]: started, version pi-hole-2.79 cachesize 150
Apr 10 17:40:34 DietPi dnsmasq[18558]: compile time options: IPv6 GNU-getopt no-MMU no-DBus no-i18n no-IDN DHCP DHCPv6 no-scripts TFTP no-conntrack ipset auth DNSSEC loop-detect inotify
Apr 10 17:40:34 DietPi dnsmasq[18558]: reading /etc/resolv.conf
Apr 10 17:40:34 DietPi dnsmasq[18558]: ignoring nameserver 127.0.0.1 - local interface
Apr 10 17:40:34 DietPi dnsmasq[18558]: read /etc/hosts - 2 addresses
```
- Now forcing reinstall:
```
root@DietPi:/var/www/pihole# pihole -r
...
  [i] Checking for lighttpd (will be installed)
...
  [✓] Starting lighttpd service
  [✓] Enabling lighttpd service to start on reboot
...
  [✓] Pi-hole blocking is Enabled

  [i] The install log is located at: /etc/pihole/install.log
  Update Complete!
...
[... 16 lines cut ...]
Apr 10 18:01:18 DietPi systemd[1]: lighttpd.service: Unit entered failed state.
Apr 10 18:01:18 DietPi systemd[1]: lighttpd.service: Failed with result 'exit-code'.
Apr 10 18:01:18 DietPi systemd[1]: lighttpd.service: Service hold-off time over, scheduling restart.
Apr 10 18:01:18 DietPi systemd[1]: Stopped Lighttpd Daemon.
Apr 10 18:01:18 DietPi systemd[1]: Starting Lighttpd Daemon...
Apr 10 18:01:18 DietPi systemd[1]: Started Lighttpd Daemon.
Apr 10 18:01:18 DietPi bash[23188]: 0 upgraded, 0 newly installed, 1 to remove and 0 not upgraded.
Apr 10 18:01:18 DietPi bash[23188]: After this operation, 974 kB disk space will be freed.
Apr 10 18:01:18 DietPi lighttpd[23294]: 2018-04-10 18:01:18: (network.c.473) listen failed:  Address already in use
Apr 10 18:01:18 DietPi systemd[1]: lighttpd.service: Main process exited, code=exited, status=255/n/a
Apr 10 18:01:18 DietPi systemd[1]: lighttpd.service: Unit entered failed state.
Apr 10 18:01:18 DietPi systemd[1]: lighttpd.service: Failed with result 'exit-code'.
Apr 10 18:01:18 DietPi bash[23188]: [614B blob data]
Apr 10 18:01:18 DietPi bash[23188]: Removing lighttpd (1.4.45-1) ...
Apr 10 18:01:18 DietPi systemd[1]: Reloading.
Apr 10 18:01:19 DietPi systemd[1]: Stopped Lighttpd Daemon.
Apr 10 18:01:19 DietPi systemd[1]: Reloading.
Apr 10 18:01:19 DietPi systemd[1]: Reloading.
Apr 10 18:01:19 DietPi bash[23188]: [614B blob data]
Apr 10 18:01:19 DietPi bash[23188]: Purging configuration files for lighttpd (1.4.45-1) ...
...
```
- As can be seen, lighttpd fails due to port already used by apache.
- But PiHole installation went through without error. But I also already saw errors there and I am not sure, if the installation/update goes through fully in this case.
- We could stop/start apache/nginx with pihole-FTL to assure Lighttpd is working correctly during update/install/reconfigure, but I don't like this done "silently", where users might want to stop pihole but keep webserver running...

MichaIng, 2018-04-10:
Close in favour of https://github.com/Fourdee/DietPi/issues/1696

closed 2018-04-10 by MichaIng