MichaIng/DietPi #1378: General | Revert to stock RPi kernel

Fourdee, 2018-01-07:
#### I've taken the decision to revert our DietPi RPi kernel to stock RPi kernel.

We simply do not have the man power and available time, to ensure this is updated regularly. Time we can use to focus on supporting bug fixes and improvements for DietPi.

Currently the only difference in our kernel was:
- DSD pop prevention on initial playback.

In future, all RPi kernel patch requests should be made directly to the RPi foundation:
https://github.com/raspberrypi/linux
This will ensure all RPi users will benefit from the requested fix, if accepted by the kernel maintainers.

I apologise if this effects audiophile users, however, I hope we have your understanding for this change.

Fourdee, 2018-01-07:
Reference other tickets, which will be closed:
- https://github.com/Fourdee/DietPi/issues/1274
- https://github.com/Fourdee/DietPi/issues/1200
- https://github.com/Fourdee/DietPi/issues/1078

closed 2018-01-07 by Fourdee

MichaIng, 2018-01-07:
@Fourdee 
Makes totally sense to me. This also allows apt kernel updates and rpi-updates without (accidentally) overwriting our custom kernel.
We just have to decide if we want to stick to the APT kernel package, or ship a newer release or master branch from: https://github.com/raspberrypi/firmware/tree/master

Fourdee, 2018-01-07:
@MichaIng 

👍 

> We just have to decide if we want to stick to the APT kernel package, or ship a newer release or master branch from: https://github.com/raspberrypi/firmware/tree/master

I've gone with the APT packages for now (we can assume they are "stable"). I've seen a few issues with ```rpi-update``` in the past, where the latest kernel was not stable and broken some features (cant remember exactly mind, was a few years ago lol)

```rpi-update``` can be done in ```dietpi-config```. Lets make this end user choice?

MichaIng, 2018-01-07:
@Fourdee 
> Lets make this end user choice?

👍 

So this reams we can stop holding RPi specific bootloader+kernel packages, right? I mean still, general issue with this system of RPi is, that if you choose rpi-update, the next APT release will overwrite it. This was great in the past, when I regularly ran `apt update && apt dist-upgrade && rpi-update` and kernels did overwrite each other, in case two times at once 😆.