MichaIng/DietPi #1379: Selecting 'Daemon + Drift' as ntp method does not set date

borpin, 2018-01-07:
## Creating a bug report/issue:
relates to #1208 - not fixed.

#### Required Information:
- RaspPi current image tested on Pi Zero and B+

#### Additional Information (if applicable):
- Can this issue be replicated on a fresh installation of DietPi? Yes

#### Expected behaviour:
Time/date set on first boot when option 4 selected in Dietpi.txt for date/time method

#### Actual behaviour:
Time date is not set so further tasks are aborted as date is too far out.

#### Steps to reproduce:
Flash new image, change date/time option to option 3.

If option 2 is chose no issues.

#### Did you submit a ```dietpi-bugreport```?
No

#### Extra details:
Possibly due to the method used to set the date/time initially fails as difference is too great.

Fourdee, 2018-01-08:
@borpin 

Thanks for the report 👍 

I'll try to replicate with large time differences.

Fourdee, 2018-01-08:
Think I see the issue here:
- The ```dietpi.txt``` entry for ```CONFIG_NTP_MODE``` (v6.0 name), is not an automation entry. Does not get applied during 1st run setup.

The setting must be changed in ```dietpi-config```, once system is running, to ensure ntp is uninstalled.

We currently force check for NTP syncs (not systemd), this is required to ensure DietPi runs successfully during 1st run updates (where the time difference would be large).

Features we could implement:
- Support ```CONFIG_NTP_MODE``` as an automation option, apply during 1st run, however, only if::
- We add support for force checking status/waiting of ```timedatectl``` in ```/DietPi/dietpi/func/run_ntpd status```
-- Downside, ```timedatectl``` can take upto 30 seconds to sync, ```ntp``` takes less than 5 seconds in most cases.

Fourdee, 2018-01-08:
Should be good to go, will test when creating updated images.

Fourdee, 2018-01-14:
Confirmed working with v6.0 RPi Stretch image.

Completed.

closed 2018-01-14 by Fourdee