MichaIng/DietPi #1382: cpu command does not show cpu utilization

kargakis, 2018-01-07:
First of all, great work! Using dietpi has been a breeze so far.

I expected to see cpu utilization when running `cpu`. I know that it is hard to give an accurate result but I would like to know whether it's something you have pondered.

Fourdee, 2018-01-08:
@kargakis 

getting CPU information from bash, without additional software packages is:
- Extremely slow, due to ```ps -aux``` list and calculations to work out the usage
- Inaccurate

Due to this, we left it out of the cpu command itself.

In v6.0, we've created a global command to obtain cpu usage:
```
root@DietPi:~# G_
G_AGA                      G_CHECK_ROOT_USER
G_AG_CHECK_INSTALL_PREREQ  G_CHECK_URL
G_AGDUG                    G_DIETPI-NOTIFY
G_AGF                      G_ERROR_HANDLER
G_AGI                      G_ERROR_HANDLER_RESET
G_AGP                      G_FILE_EXISTS
G_AGUG                     G_OBTAIN_CPU_TEMP
G_AGUP                     G_OBTAIN_CPU_USAGE
G_CHECK_ROOTFS_RW          G_RUN_CMD

root@DietPi:~# G_OBTAIN_CPU_USAGE
1.2
```

If accurate CPU usage and vast stats are required, i'd highly recommend NetData:
http://dietpi.com/phpbb/viewtopic.php?f=8&t=5&start=60#p1611

Fourdee, 2018-01-08:
I'll mark this as question answered, please reopen if required.

closed 2018-01-08 by Fourdee