MichaIng/DietPi #1384: DietPi-Process_Tool | Non-verbose mode

MichaIng, 2018-01-09:
During applying dietpi-config settings, dietpi-software runs etc., as services are often stopped and started, `dietpi-process_tool` often consumes the whole screen, displacing perhaps other useful information.

Maybe we can use some "non-verbose" mode to just show one "success" or "failed" message instead of every assigned value for every process.

Just running the script as G_RUN_CMD also works, we could just assign a less plain command name:
```
 DietPi-Services
─────────────────────────────────────────────────────
 Mode: start

[  OK  ] start : cron
[  OK  ] start : lighttpd
[  OK  ] start : php7.0-fpm
[  OK  ] start : mysql
[  OK  ] start : redis-server

[  OK  ] /DietPi/dietpi/dietpi-process_tool 1
[  OK  ] ncc maintenance:mode --off
```

MichaIng, 2018-01-25:
Simple "solution" offer: https://github.com/Fourdee/DietPi/pull/1424

Fourdee, 2018-02-18:
Marking as closed in favor of:
https://github.com/Fourdee/DietPi/pull/1424#issuecomment-366525898

Please reopen if required.

closed 2018-02-18 by Fourdee