MichaIng/DietPi #1385: v6.0 | Recreate all images

Fourdee, 2018-01-09:
Info TBA:

Fourdee, 2018-01-09:
Reference:
- https://github.com/Fourdee/DietPi/issues/1285
- https://github.com/Fourdee/DietPi/issues/1268

Fourdee, 2018-01-09:
Notes:
- Manually check ```/etc/bash.bashrc``` and clean (eg: ```LANG=``` options if added)

k-plan, 2018-01-10:
@Fourdee 

Sorry, but I am really unsure, if it make sense, to help you building images from my site.

PREP is to new and there are surely several bits and bobs you will find out while building images. So fixes are a lot easier for you and mainly faster altogether.

Fourdee, 2018-01-11:
ASUS TB: http://dlcdnet.asus.com/pub/ASUS/mb/Linux/Tinker_Board_2GB/20171115-tinker-board-linaro-stretch-alip-v2.0.4.zip?_ga=2.111431302.324915190.1515670093-379854191.1515670093
```
[FAILED] DietPi-PREP_SYSTEM_FOR_DIETPI: G_AGA
[ INFO ] exit_code = 100
[ INFO ] VERSION:v. | HW_MODEL:100 | HW_ARCH:2 | DISTRO:4
[ INFO ] Log file contents:
Removing libwww-perl (6.15-1) ...
Removing libhttp-negotiate-perl (6.00-2) ...
Removing libfile-listing-perl (6.04-1) ...
Removing libwww-robotrules-perl (6.01-1) ...
Removing libnet-http-perl (6.12-1) ...
Removing libhtml-tree-perl (5.03-2) ...
Removing libhtml-parser-perl (3.72-3) ...
Removing libhtml-tagset-perl (3.20-3) ...
Removing libhttp-cookies-perl (6.01-1) ...
Removing libio-socket-ssl-perl (2.044-1) ...
Removing libnet-ssleay-perl (1.80-1) ...
Removing netbase (5.4) ...
Removing perl-openssl-defaults:armhf (3) ...
Removing python3 (3.5.3-1) ...
Removing dh-python (2.20170125) ...
Removing libhttp-message-perl (6.11-1) ...
Removing libencode-locale-perl (1.05-1) ...
Removing liburi-perl (1.71-1) ...
Removing libhttp-date-perl (6.02-1) ...
Removing libio-html-perl (1.001-1) ...
Removing liblwp-mediatypes-perl (6.02-1) ...
Removing libpython3-stdlib:armhf (3.5.3-1) ...
Removing python3.5 (3.5.3-1) ...
Removing libpython3.5-stdlib:armhf (3.5.3-1) ...
Removing libmpdec2:armhf (2.4.2-1) ...
Removing python3-minimal (3.5.3-1) ...
Removing python3.5-minimal (3.5.3-1) ...
Removing libpython3.5-minimal:armhf (3.5.3-1) ...
Removing libtimedate-perl (2.3000-2) ...
Removing mime-support (3.60) ...
Removing perl (5.24.1-3+deb9u2) ...
Removing libperl5.24:armhf (5.24.1-3+deb9u2) ...
Removing libgdbm3:armhf (1.8.3-14) ...
Removing perl-modules-5.24 (5.24.1-3+deb9u2) ...
dpkg: libip4tc0:armhf: dependency problems, but removing anyway as you requested:
 systemd depends on libip4tc0 (>= 1.6.0+snapshot20161117).

Removing libip4tc0:armhf (1.6.0+snapshot20161117-6) ...
dpkg: libseccomp2:armhf: dependency problems, but removing anyway as you requested:
 systemd depends on libseccomp2 (>= 2.3.1); however:
  Package libseccomp2:armhf is to be removed.

Removing libseccomp2:armhf (2.3.1-2.1) ...
dpkg: libcryptsetup4:armhf: dependency problems, but removing anyway as you requested:
 systemd depends on libcryptsetup4 (>= 2:1.4.3).

Removing libcryptsetup4:armhf (2:1.7.3-4) ...
Errors were encountered while processing:
 systemd
```

Fourdee, 2018-01-11:
@MichaIng 
Just finished PREP:
```
root@DietPi:~# dpkg --get-selections | grep ^init
init-system-helpers                             install
initramfs-tools                                 install
initramfs-tools-core                            install
initscripts                                     install

###
root@DietPi:~# dpkg --get-selections | grep ^sys
systemd                                         install
systemd-sysv                                    install
sysv-rc                                         install
sysvinit-utils                                  install

```

MichaIng, 2018-01-11:
@k-plan 
Okay, no `init` there: https://packages.debian.org/en/stretch/init
Jep, then we should just always install `systemd-sysv`, which also ensures base `systemd`. We could then actually also purge `init`, which is just a meta package to ensure one of systemd or sysvinit is installed.

Fourdee, 2018-01-11:
Pre-Prep:
```
root@tinkerboard:~# dpkg --get-selections | grep ^init
init-system-helpers                             install
initramfs-tools                                 install
initramfs-tools-core                            install
initscripts                                     install
root@tinkerboard:~# dpkg --get-selections | grep ^sys
system-tools-backends                           install
systemd                                         install
systemd-sysv                                    install
sysv-rc                                         install
sysvinit-utils                                  install
```

k-plan, 2018-01-11:
I don't want to disturb your image creation process.
But I want to better mention this here. 
I have a quick look into your fresh build images. Nice work. 👍 
Think, we have to do some housekeeping after this (delete  some unwanted/useless/violate files).

Suggestion:
- build all new images now until it is finshed
- then we can delete all unwanted/useless/violate files altogether at once (mount and delete/edit)
- drawback: must  create a new hash file 😢  , so I would suggest not to create a `hash.txt` before last finalize (woebegone experience)

But this is only fine tuning. We can as well leave it as it is.

MichaIng, 2018-01-11:
@Fourdee 
Just saw the other issues you faced:
- I also faced the stuck processing animation. First thought this was because of `G_DIETPI-NOTIFY 3` first also created animation, where it is not assured that same user with rights to stop process tries it. But I also faced this after removing animation there, mostly on `dietpi-led_control` but also on e.g. `dietpi-set_cpu`.
- I just played around, first I thought, that every non-root user has no access to `dietpi-globals`...
```
root@VM-Stretch:~# sudo -u dietpi touch /var/run/test.pid
touch: cannot touch '/var/run/test.pid': Permission denied
root@VM-Stretch:~# sudo -u dietpi G_DIETPI-NOTIFY -2 testing
sudo: G_DIETPI-NOTIFY: command not found
root@VM-Stretch:~# sudo -u dietpi source /DietPi/dietpi/func/dietpi-globals
sudo: source: command not found
root@VM-Stretch:~# sudo -u dietpi /DietPi/dietpi/func/dietpi-globals
root@VM-Stretch:~# sudo -u dietpi G_DIETPI-NOTIFY -2 testing
sudo: G_DIETPI-NOTIFY: command not found 
```
- But if you actually log in, G_DIETPI-NOTIFY can be used, just there is no access to `/var/run` = `/run`
```
dietpi@VM-Stretch:~$ G_DIETPI-NOTIFY 2 testing
[ INFO ] testing
dietpi@VM-Stretch:~$ G_DIETPI-NOTIFY -2 testing
         testing-bash: /var/run/dietpi-process.pid: Permission denied
```
```
root@VM-Stretch:~# ls -al /var/run
lrwxrwxrwx 1 root root 4 Jan 10 21:01 /var/run -> /run
root@VM-Stretch:~# ls -al /run
total 24
drwxr-xr-x 15 root root  500 Jan 11 18:14 .
drwxr-xr-x 22 root root 4096 Jan 10 22:51 ..
-rw-------  1 root root    0 Jan 11 15:28 agetty.reload
drwxr-xr-x  2 root root   80 Jan 11 15:28 console-setup
-rw-r--r--  1 root root    4 Jan 11 15:28 crond.pid
----------  1 root root    0 Jan 11 15:28 crond.reboot
drwxr-xr-x  2 root root   60 Jan 11 15:28 dbus
-rw-r--r--  1 root root    4 Jan 11 15:28 dhclient.eth0.pid
-rw-r--r--  1 root root    4 Jan 11 15:28 dropbear.pid
lrwxrwxrwx  1 root root   25 Jan 11 15:28 initctl -> /run/systemd/initctl/fifo
drwxr-xr-x  2 root root   40 Jan 11 18:07 initramfs
drwxrwxrwt  3 root root   60 Jan 11 15:28 lock
drwxr-xr-x  3 root root   60 Jan 11 18:07 log
-rw-r--r--  1 root root    0 Jan 11 15:28 motd.dynamic
drwxr-xr-x  2 root root   60 Jan 11 18:07 mount
drwxr-xr-x  2 root root  120 Jan 11 15:28 network
-rw-r--r--  1 root root 2229 Jan 11 15:28 ntp.conf.dhcp
drwxr-xr-x  3 root root  100 Jan 11 15:28 resolvconf
drwxr-xr-x  2 root root   40 Jan 11 15:28 sendsigs.omit.d
lrwxrwxrwx  1 root root    8 Jan 11 15:28 shm -> /dev/shm
drwxr-xr-x 17 root root  400 Jan 11 15:28 systemd
drwxr-xr-x  2 root root   60 Jan 11 18:07 tmpfiles.d
drwxr-xr-x  7 root root  160 Jan 11 15:28 udev
drwxr-xr-x  2 root root   40 Jan 11 15:28 user
-rw-rw-r--  1 root utmp 1536 Jan 11 18:14 utmp
```
- So if any non-root user starts the animation, no pid file will be created, thus more animations can be started.
- So we could create the file somewhere, where all users have full read+write access with `chmod 777` or limit the creation to root user/just start if, after pid file is created (touch file && start process && pid > file).

- On my testing VM I so far added an animation break, if no pid file exists: `[ -f /var/run/dietpi-process.pid ] && builtin printf "\r${aprocess_string[i]} " || break` But this doesn't solve issues, where e.g. a script, executed by root, breaks, before animation is stopped and then one uses e.g. dietpi user for security reasons, which has no rights to remove pid file and/or kill animation process.

=> Creating pid file on 777 location with chmod 777 would be actually best. Not sure if this possible just inside root? `touch /dietpi-process.pid && chmod 777 /dietpi-process.pid && <start animation> && write pid to file`

MichaIng, 2018-01-12:
Accidentally merged commit into existing PR branch: https://github.com/Fourdee/DietPi/pull/1386
+ After investigation, (main) reason for stuck process animation was, that `/var/run` seemed to be cleared ~2 times during boot. Thus no pid information remained => nobody can kill the animation.
+ Added file check into animation script, now breaks if file does not exists anymore, never faced the issue anymore so far.
+ Proactive: Sadly it seems to be impossible that user `dietpi` removes a file from user `root` no matter what permissions he has on the file or the parent directory: Both 777 does not work, also not on own created directory within `/`. So for now I don't see any possibility where a non-root user can ever stop the animation (besides sudo), if started by root. Thus we still need to ensure, that every animation is handled well and in the end the same script always does some error handling in case, to clean animation.

MichaIng, 2018-01-13:
@Fourdee 
While creating VirtualBox image, I recognized following warning during APT installations when doing via dropbear (actually already face this since I can think):

> dropbear: WARNING: Invalid authorized_keys file, remote unlocking of cryptroot via SSH won't work!

Made some research and found, that `dropbear-initramfs` causes the error, which is only necessary to unlock full encrypted drives and expects the key file for this: https://packages.debian.org/stretch/dropbear-initramfs

Instead of installing `dropbear` (dummy package including `dropbear-initramfs`), for images we just need `dropbear-run`: https://packages.debian.org/de/stretch/dropbear-run
They made the separation some time ago, as users faced problem with the fix initramfs integration.
Everyone who uses encrypted drives will anyway configure the system + SSH client to use key authentication and stuff and then in case need to install the additional package.

Fourdee, 2018-01-14:
NanoPi Neo (default kernel):
[image]

We'll need to use mainline kernel to create Stretch images.

Fourdee, 2018-01-14:
Considering:
If we can automate the detection of current kernel packages in PREP, I'll create only the popular OPi/BPi/NanoPi images, the less used ones can be user created with PREP.

Fourdee, 2018-01-14:
ARMbian PREP images failing to boot:
```
Retry time exceeded; starting again
BOOTP broadcast 1
BOOTP broadcast 2
BOOTP broadcast 3
BOOTP broadcast 4
BOOTP broadcast 5
DHCP client bound to address 192.168.0.30 (4270 ms)
Using ethernet@1c30000 device
TFTP from server 0.0.0.0; our IP address is 192.168.0.30; sending through gateway 192.168.0.1
Filename 'boot.scr.uimg'.
Load address: 0x42000000
Loading: T T T T T T T T T
```

Fourdee, 2018-01-14:
NanoPi NEO2:
```
qemu: uncaught target signal 11 (Segmentation fault) - core dumped
                                                                  /root/armbian/build/lib/debootstrap-ng.sh: line 204: 78878 Segmentation fault      LC_ALL=C LANG=C chroot $SDCARD /bin/bash -c "DEBIAN_FRONTEND=noninteractive apt-get -y -q $apt_extra $apt_extra_progress --no-install-recommends install $PACKAGE_LIST"
                                                                            [ error ] ERROR in function create_rootfs_cache [ debootstrap-ng.sh:208 ]
                                                                     [ error ] Installation of Armbian packages failed
                                       [ o.k. ] Process terminated
                                                                   [ error ] ERROR in function unmount_on_exit [ image-helpers.sh:59 ]
                                                      [ error ] debootstrap-ng was interrupted
               [ o.k. ] Process terminated
                                           root@vmcross16:~/armbian/build#
```

k-plan, 2018-01-14:
@MichaIng 

> Instead of installing dropbear (dummy package including dropbear-initramfs), for images we just need dropbear-run: https://packages.debian.org/de/stretch/dropbear-run

Nice finding, but this will only work with Debian Stretch, Buster and sid. 
Your given link show this.
It will break Debian Jessie image building via `PREP_SYSTEM_FOR_DIETPI.sh`.
Jessie needed https://packages.debian.org/de/jessie/dropbear
Sorry.

@Fourdee 

Play a little bit with Odroid C1 Jessie image creation.
Something going wrong with [dietpi-obtain_hw_model](https://github.com/Fourdee/DietPi/blob/testing/dietpi/dietpi-obtain_hw_model#L562-L601) `G_HW_MODEL_DESCRIPTION` for Odroid C1.

Temporary fixed with:

```
                       #oDroid C2 via /etc/.dietpi_hw_model
                        elif (( $G_HW_MODEL == 12 )); then

                                G_HW_MODEL_DESCRIPTION='oDroid C2'
                                IMAGE_ADDITIONAL_CREDITS='Meveric'

                        #oDroid XU3/4 via /etc/.dietpi_hw_model
                        elif (( $G_HW_MODEL == 11 )); then

                                G_HW_MODEL_DESCRIPTION='oDroid XU3/4'
                                IMAGE_ADDITIONAL_CREDITS='Meveric'

                        #oDroid C1 via /etc/.dietpi_hw_model
                        elif (( $G_HW_MODEL == 10 )); then

                                G_HW_MODEL_DESCRIPTION='oDroid C1'
                                IMAGE_ADDITIONAL_CREDITS='Meveric'

                        fi
```

But I don't know why this is not working:

```
                #oDroid C1
                elif (( $(cat /proc/cpuinfo | grep -ci -m1 'ODROIDC') )); then

                        G_HW_MODEL_DESCRIPTION='oDroid C1'
                        G_HW_MODEL=10
                        IMAGE_ADDITIONAL_CREDITS='Meveric'

                fi

```

Test:

```
root@DietPi:~# cat /proc/cpuinfo | grep -ci -m1 'ODROIDC'
1
```

And yes, your Meveric mirror is down at the moment:

```
~# apt-get update
...
Err http://fuzon.co.uk all/c1 Translation-en_GB
  Unable to connect to fuzon.co.uk:http:
...
```

MichaIng, 2018-01-15:
@k-plan 
Thx for hint, yes the package separation is not yet available on Jessie, thus I added a G_DISTRO specific package section: https://github.com/Fourdee/DietPi/pull/1394

Fourdee, 2018-01-15:
@k-plan 

> root@DietPi:~# cat /proc/cpuinfo | grep -ci -m1 'ODROIDC'
1

Bizarre. I'll send commit to hard code the index as per your fix:
https://github.com/Fourdee/DietPi/issues/1385#issuecomment-357539747

We only really need the detection for RPi's, and, new prep system always sets the G_HW_MODEL file which is read by https://github.com/Fourdee/DietPi/blob/testing/dietpi/dietpi-obtain_hw_model#L562-L601

Glad you tried this one, I cant find my C1 anywhere 😢

k-plan, 2018-01-15:
Bizarre on Odroid C1:

```
root@DietPi:~# cat /etc/apt/sources.list
deb http://ftp.debian.org/debian/ jessie main contrib non-free
deb http://ftp.debian.org/debian/ jessie-updates main contrib non-free
deb http://security.debian.org jessie/updates main contrib non-free
deb http://ftp.debian.org/debian/ jessie-backports main contrib non-free

root@DietPi:~# /DietPi/dietpi/func/dietpi-set_software apt-mirror http://debian.netcologne.de/debian/
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (0)
[  OK  ] Root access verified.
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (0)
[ INFO ] Checking RootFS R/W access. Please wait...
[  OK  ] RootFS R/W access.

 DietPi-Set_Software
─────────────────────────────────────────────────────
 Mode: apt-mirror (http://debian.netcologne.de/debian/)

[  OK  ] apt-mirror http://debian.netcologne.de/debian/ | Completed

root@DietPi:~# cat /etc/apt/sources.list
deb http://deb.debian.org/debian/ jessie main contrib non-free
deb http://deb.debian.org/debian/ jessie-updates main contrib non-free
deb http://deb.debian.org/debian-security/ jessie/updates main contrib non-free
deb http://deb.debian.org/debian/ jessie-backports main contrib non-free

root@DietPi:~# locale
locale: Cannot set LC_CTYPE to default locale: No such file or directory
locale: Cannot set LC_MESSAGES to default locale: No such file or directory
locale: Cannot set LC_ALL to default locale: No such file or directory
LANG=en_GB.UTF-8
LANGUAGE=0:0
LC_CTYPE="0"
LC_NUMERIC="0"
LC_TIME="0"
LC_COLLATE="0"
LC_MONETARY="0"
LC_MESSAGES="0"
LC_PAPER="0"
LC_NAME="0"
LC_ADDRESS="0"
LC_TELEPHONE="0"
LC_MEASUREMENT="0"
LC_IDENTIFICATION="0"
LC_ALL=0
```

k-plan, 2018-01-15:
Odroid C1 problem with `dietpi-config` => `Performance Options`

`/DietPi/dietpi/dietpi-config: line 1560: * 25 : syntax error: operand expected (error token is "* 25 ")`

```
root@DietPi:~# cat /DietPi/dietpi.txt | grep CONFIG_CPU_GOVERNOR
CONFIG_CPU_GOVERNOR=ondemand

root@DietPi:~# cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor
ondemand

root@DietPi:~# cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_available_governors
hotplug interactive conservative ondemand performance

```

[image]

To get the new value `CONFIG_CPU_GOVERNOR=ondemand` in https://github.com/Fourdee/DietPi/blob/testing/dietpi.txt#L121-L123

you have to modify:

https://github.com/Fourdee/DietPi/blob/testing/dietpi/dietpi-config#L1868-L1873

Fourdee, 2018-01-16:
@k-plan 

Thanks for testing/feedback 👍 

> /etc/apt/preferences.d/deb-multimedia-pin-99

can be removed, but no harm in leaving it in place.

> /etc/apt/sources.list.d/meveric-*

Require for Meveric's repo (GPU/Kodi/Kernel packages)

All other items can be left in place.

> /DietPi/dietpi/func/dietpi-set_software apt-mirror http://debian.netcologne.de/debian/

I'll try to replicate.

> /DietPi/dietpi/dietpi-config: line 1560: * 25 : syntax error: operand expected (error token is "* 25 ")

Fixed for commit (below). Was due to missed renaming of dietpi.txt vars (user error my end 😉 ).

k-plan, 2018-01-16:
@Fourdee 

> Resolved: 1d90070

Testing on Odroid C1+:

--------------

#### Required Information:
- DietPi Version | v6.-1
- SBC Device     | Odroid C1 (armv7l) (index=10)
- Distro         | jessie (index=3)
- Command        | G_AGUP
- Error Handler  | G_ERROR_HANDLER

#### Additional Information (if applicable):
- Software title | DietPi-Software

#### Expected behaviour:

- inital setup

#### Actual behaviour:

`[FAILED] Unable to continue, the program will now terminate.`

#### Steps to reproduce:

1. `./PREP_SYSTEM_FOR_DIETPI.sh`

2. `reboot`

3. `[FAILED] Unable to continue, the program will now terminate.`

#### Additional logs:
```
Log file contents:
Get:10 http://ftp.debian.org jessie-updates/contrib armhf Packages [32 B]
Get:11 http://ftp.debian.org jessie-updates/contrib Translation-en [14 B]
Get:12 http://fuzon.co.uk jessie/main armhf Packages [74.1 kB]
Get:13 http://ftp.debian.org jessie-updates/main Translation-en [14.9 kB]
Get:14 http://ftp.debian.org jessie-updates/non-free Translation-en [496 B]
Get:15 http://fuzon.co.uk jessie/backports armhf Packages [76.5 kB]
Ign http://fuzon.co.uk all/c1 Translation-en_GB
Ign http://fuzon.co.uk all/c1 Translation-en
Ign http://fuzon.co.uk all/main Translation-en_GB
Ign http://fuzon.co.uk all/main Translation-en
Ign http://fuzon.co.uk all/testing Translation-en_GB
Ign http://fuzon.co.uk all/testing Translation-en
Get:16 http://ftp.debian.org jessie-backports/contrib Translation-en [7,266 B]
Get:17 http://ftp.debian.org jessie-backports/main Translation-en [629 kB]
Ign http://fuzon.co.uk jessie/backports Translation-en_GB
Ign http://fuzon.co.uk jessie/backports Translation-en
Ign http://fuzon.co.uk jessie/main Translation-en_GB
Ign http://fuzon.co.uk jessie/main Translation-en
Get:18 http://ftp.debian.org jessie-backports/non-free Translation-en [29.1 kB]
Get:19 http://ftp.debian.org jessie-updates/main armhf Packages [20.1 kB]
Get:20 http://ftp.debian.org jessie-updates/non-free armhf Packages [516 B]
Get:21 http://ftp.debian.org jessie/main armhf Packages [6,644 kB]
Get:22 http://ftp.debian.org jessie/contrib armhf Packages [37.8 kB]
Get:23 http://ftp.debian.org jessie/non-free armhf Packages [62.5 kB]
Get:24 http://ftp.debian.org jessie/contrib Translation-en [38.3 kB]
Get:25 http://ftp.debian.org jessie/main Translation-en [4,583 kB]
Get:26 http://ftp.debian.org jessie/non-free Translation-en [72.1 kB]
Err http://ftp.debian.org jessie/updates/main armhf Packages
  404  Not Found
Err http://ftp.debian.org jessie/updates/contrib armhf Packages
  404  Not Found
Err http://ftp.debian.org jessie/updates/non-free armhf Packages
  404  Not Found
Ign http://ftp.debian.org jessie/updates/contrib Translation-en_GB
Get:27 http://ftp.debian.org jessie-backports/main armhf Packages [875 kB]
Get:28 http://ftp.debian.org jessie-backports/contrib armhf Packages [7,944 B]
Get:29 http://ftp.debian.org jessie-backports/non-free armhf Packages [6,692 B]
Ign http://ftp.debian.org jessie/updates/contrib Translation-en
Ign http://ftp.debian.org jessie/updates/main Translation-en_GB
Ign http://ftp.debian.org jessie/updates/main Translation-en
Ign http://ftp.debian.org jessie/updates/non-free Translation-en_GB
Ign http://ftp.debian.org jessie/updates/non-free Translation-en
Fetched 13.7 MB in 14s (967 kB/s)
W: Failed to fetch http://ftp.debian.org/debian-security/dists/jessie/updates/main/binary-armhf/Packages  404  Not Found

W: Failed to fetch http://ftp.debian.org/debian-security/dists/jessie/updates/contrib/binary-armhf/Packages  404  Not Found

W: Failed to fetch http://ftp.debian.org/debian-security/dists/jessie/updates/non-free/binary-armhf/Packages  404  Not Found

E: Some index files failed to download. They have been ignored, or old ones used instead.
```

-------------

```
root@DietPi:~# cat /DietPi/dietpi.txt | grep CONFIG_APT_DEBIAN_MIRROR
CONFIG_APT_DEBIAN_MIRROR=http://ftp.debian.org/debian/
```

```
root@DietPi:~# cat /etc/apt/sources.list
deb http://ftp.debian.org/debian/ jessie main contrib non-free
deb http://ftp.debian.org/debian/ jessie-updates main contrib non-free
deb http://ftp.debian.org/debian-security/ jessie/updates main contrib non-free
deb http://ftp.debian.org/debian/ jessie-backports main contrib non-free
```

Fourdee, 2018-01-16:
@k-plan 

I've hard coded the debian-security repo: https://github.com/Fourdee/DietPi/commit/50875c7a4b4e1eee0d68783e9d1923052d03083e

Seems not all mirrors provide the debian-security repo.

Fourdee, 2018-01-16:
OPi Win:
- No ethernet/USB
- HDMI output only.

MichaIng, 2018-01-16:
@Fourdee @ k-plan
Yeah, in different mirrors the security repo is available on different URLs, not always via /debian-security/. I suggest we keep it hard coded on deb.debian.org. I will fix it with current PR as I already adjusted the default there.

MichaIng, 2018-01-16:
@Fourdee 
Created new PR for this: https://github.com/Fourdee/DietPi/pull/1399
+ Let's stay on deb.debian.org (mirrordirector) by default and hardcoded for debian-security repo, as said.
+ Revert just for mirrordirector to http on Jessie. Generally https works, it just shows some warnings in combination with deb.debian.org. So we can generally allow https there (which will in every case overwritten right now).

k-plan, 2018-01-16:
Hmm ... testing `dietpi-config` => `Language/Regional Options` => `Locale`

- select three different languages or more, eg.
 `en_GB.UTF-8`
`en_US.UTF-8`
`de_DE.UTF-8`

- select a new default eg. `de_DE.UTF-8`

- say ` <Yes> ` to reboot

- look what's happend after reboot

- try to change locale via `dietpi-config` to a different default
```
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
        LANGUAGE = "de_DE:de",
        LC_ALL = "de_DE.UTF-8",
        LANG = "en_GB.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to a fallback locale ("en_GB.UTF-8").
locale: Cannot set LC_CTYPE to default locale: No such file or directory
locale: Cannot set LC_MESSAGES to default locale: No such file or directory
locale: Cannot set LC_ALL to default locale: No such file or directory
/usr/bin/locale: Cannot set LC_CTYPE to default locale: No such file or directory
/usr/bin/locale: Cannot set LC_MESSAGES to default locale: No such file or directory
/usr/bin/locale: Cannot set LC_ALL to default locale: No such file or directory
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (de_DE.UTF-8)

```

```
root@DietPi:~# locale
locale: Cannot set LC_CTYPE to default locale: No such file or directory
locale: Cannot set LC_MESSAGES to default locale: No such file or directory
locale: Cannot set LC_ALL to default locale: No such file or directory
LANG=en_GB.UTF-8
LANGUAGE=de_DE:de
LC_CTYPE="de_DE.UTF-8"
LC_NUMERIC="de_DE.UTF-8"
LC_TIME="de_DE.UTF-8"
LC_COLLATE="de_DE.UTF-8"
LC_MONETARY="de_DE.UTF-8"
LC_MESSAGES="de_DE.UTF-8"
LC_PAPER="de_DE.UTF-8"
LC_NAME="de_DE.UTF-8"
LC_ADDRESS="de_DE.UTF-8"
LC_TELEPHONE="de_DE.UTF-8"
LC_MEASUREMENT="de_DE.UTF-8"
LC_IDENTIFICATION="de_DE.UTF-8"
LC_ALL=de_DE.UTF-8

```

MichaIng, 2018-01-16:
@k-plan 
Couldn't replicate on first try (VM Jessie). Chose the same three locales via dietpi-config and `de_DE.UTF-8` as new default. Locale generation and dpkg reconfiguration finished without error, as well as everything after reboot:
```
root@DietPi:~# locale
LANG=en_GB.UTF-8
LANGUAGE=de_DE:de
LC_CTYPE="de_DE.UTF-8"
LC_NUMERIC="de_DE.UTF-8"
LC_TIME="de_DE.UTF-8"
LC_COLLATE="de_DE.UTF-8"
LC_MONETARY="de_DE.UTF-8"
LC_MESSAGES="de_DE.UTF-8"
LC_PAPER="de_DE.UTF-8"
LC_NAME="de_DE.UTF-8"
LC_ADDRESS="de_DE.UTF-8"
LC_TELEPHONE="de_DE.UTF-8"
LC_MEASUREMENT="de_DE.UTF-8"
LC_IDENTIFICATION="de_DE.UTF-8"
LC_ALL=de_DE.UTF-8
```

But when I want to switch default language again to another (en_US.UTF-8), then suddenly I face same error.

Somehow in `/etc/locale.gen` only two languages are commented out, even that three were generated. While after using `dpkg-reconfigure locales` and doing just the same (choosing 3 languages and switching default) afterwards `/etc/locale.gen` has 3 uncommented entries.

Fourdee, 2018-01-17:
```
Move LANG=en_GB.UTF-8 out of dietpi-globals (into all scripts separately again), as we source it on every login which might affect also desktop/software languages? We can assure on the other hand that dietpi-globals itself is parsed correctly, if we set LANG=en_GB.UTF-8 in the dedicated scripts before sourcing it. The language will then only be set until script ending. Just at boot/dietpi-login script, we might need to test, if everything behaves as expected. I am not sure how all this is handled before user login.
```

We may also need to consider that ```en_GB.UTF-8``` is the required locale, and, DietPi only supports ```en_GB.UTF-8```. However, this really is a last resort if we cant find a solution.

I'll need to do some digging into this, once i've finished the v6.0 image creations.

Fourdee, 2018-01-18:
@k-plan 

> I can upload my DietPi_v6.0_OdroidC1-ARMv7-Jessie.img, if you want.

Legend 👍 Please if you can. 

Locales + CPU, we can patch during day one v6.0 update.

k-plan, 2018-01-18:
@Fourdee 

> Legend 👍 Please if you can.

Uploaded without `hash.txt`. Please have a look and check if it is okay.

### What doesn't work:

- automated install of software via `dietpi.txt` eg: `AUTO_SETUP_INSTALL_SOFTWARE_ID=3`

- WiFi: set `wificountrycode` via `dietpi.txt` or `dietpi-config`
[image]
```
DietPi-Set_Hardware
─────────────────────────────────────────────────────
 Mode: wificountrycode (gb)

[  OK  ] wificountrycode GB | Completed

 DietPi-Services
─────────────────────────────────────────────────────
 Mode: stop

[  OK  ] stop : cron

 DietPi-Set_Hardware
─────────────────────────────────────────────────────
 Mode: wifimodules (enable)

[ INFO ] Checking for pre-req APT packages: crda firmware-atheros firmware-brcm80211 firmware-ralink iw rfkill wireless-tools wpasupplicant
[ INFO ] (crda) | Currently installed
[ INFO ] (firmware-atheros) | Currently installed
[ INFO ] (firmware-brcm80211) | Currently installed
[ INFO ] (firmware-ralink) | Currently installed
[ INFO ] (iw) | Currently installed
[ INFO ] (rfkill) | Currently installed
[ INFO ] (wireless-tools) | Currently installed
[ INFO ] (wpasupplicant) | Currently installed
[ INFO ] Pre-req APT packages are currently already installed
[ INFO ] Please wait, enabling WiFi Modules...
[  OK  ] wifimodules enable | Completed
[ INFO ] Dropping connections, please wait...
[ INFO ] Restarting networking services, please wait...
Error for wireless request "Set Power Management" (8B2C) :
    SET failed on device wlan0 ; Operation not permitted.
ioctl[SIOCSIWAP]: Operation not permitted
ioctl[SIOCSIWENCODEEXT]: Invalid argument
ioctl[SIOCSIWENCODEEXT]: Invalid argument
Internet Systems Consortium DHCP Client 4.3.1
Copyright 2004-2014 Internet Systems Consortium.
All rights reserved.
For info, please visit https://www.isc.org/software/dhcp/

Listening on LPF/wlan0/74:da:38:73:2c:aa
Sending on   LPF/wlan0/74:da:38:73:2c:aa
Sending on   Socket/fallback
DHCPDISCOVER on wlan0 to 255.255.255.255 port 67 interval 8
....
```
[image]

Fourdee, 2018-01-19:
Issues:
- DietPi-Software no longer prompts user to setup connection, if it fails (due to error handler)

Fourdee, 2018-01-20:
Notes:
- Redone all images with updated DietPi Sourcecode (due to commits above)
- Rock64 redone, due to corrupt image (faulty PSU)

Just needs uploading.

Fourdee, 2018-01-20:
@MichaIng 

Q:
I tried exporting the updated VirtualBox image (after i've updated DP source code), but it generates 300MB .7z file lol.

```zerofree -v /dev/sda1``` was not supported from my linux boot iso, what method do you use, if any?

MichaIng, 2018-01-20:
@Fourdee 
I used `zerofree` from another VirtualBox DietPi (as I am running three: Jessie + Stretch + Buster), cross mounting the fresh images hdd. I think you first need to free the hdd from within global tools / virtual media manager.

And I manually purged the realtek firmware, as it is not needed for any of the VirtualBox provided Ethernet cards. I just recognize, that this is still not yet done by current preparation script. Will fix it. Actually I think we can leave out `firmware-linux-nonfree` package for VMs, didn't face any issue as I already purge them on my VMs since long. This will further reduce image size.

MichaIng, 2018-01-21:
@Fourdee
You mean Windows 10 host? Might be due to the networking drivers for host-only and bridged network that VirtualBox installs for easier setup of their guest network. But interesting that vmware then does not load at all... Maybe it would be enough to temporary disable the drivers within Windows adapter settings of main network card.
I didn't use vmware so far, bit good to know.

k-plan, 2018-01-21:
@MichaIng 

Hmm, login screen on VirtualBox  looks a little bit ugly?

[image]

With SSH it looks okay:

[image]

Fourdee, 2018-01-21:
@k-plan 

Not just the VM's. Native PC and a few other SBC's (outside SSH)

Fourdee, 2018-01-21:
@MichaIng 

>  I think you first need to free the hdd from within global tools / virtual media manager.

Tried that, simply removes the disk/device from VM. Out of time, so 300MB 7z it is ☹️

k-plan, 2018-01-21:
@Fourdee 

I have a look at/in your last build images for Odroid, NanoPi and OrangePi.

Personally I dislike the "_one-ext4-partition-layout_" (ARMbian based devices).
Why? Because of you loose some of the great basic  DietPi features:  Simplified | For everyone
Without a FAT partition, which contain `dietpi.txt`, you will loose the option for preconfigure installation and dietpi-automation. You always need a linux system or a paid ext filesystem driver package to do this.
Like so, many user questions and suport will be generated. And most of documentation is now outdated, because some of the devices will keep support this great feature and most of them not any more.

"_Official FriendlyARM image_" is more a OpenMediaVault (OMV) image, based on [ARMbian build system modification powered by Mr. tk](https://forum.armbian.com/topic/3953-preview-generate-omv-images-for-sbc-with-armbian/), which is [his new project](https://forum.openmediavault.org/index.php/Board/31-Announcements/).
With our build image eg. for NanoPi Neo2, based on this, I have the problem, what with my Mac OSX, the FAT partition is not accessible. Tested with a Win7 VM it will works, I can see and edit the files. Same with `nanopi-neo2_debian-nas-jessie_4.14.0_20171208.img`.
Strange, because when is use `OMV_3_0_90_Nanopineo2_4.13.10.img`, I can see and edit the FAT partition.

I personally would like to discontinue all ARMbian based device support.
Sorry. Just my 2 cent.

MichaIng, 2018-01-21:
@Fourdee
That's true, freeing removes the image first from VM. But it stays at the VMs folder. So you just need to add it again to the virtual controller/remount it to the drive.
I see if I find time to recreate the image tonight.

@k-plan
Yeah I agree, that this is a nice feature loss for ARMbian based devices. There are also some free ext4 drivers for Windows, but they are not working to reliable. Tried it once and ended with a knoppix live USB stick, respectively a Linux VM.

If it is too difficult to change ARMbian image back to two partitions, we can maybe find/offer an easy reliable way for users to get ext4 access.

Another alternative would be to allow editing the dietpi.txt on first boot? But this needs direct terminal access or at least network (SSH) working. Also does not allow to preconfigure once for several machines.

But I would better stay with offering support for this devices without preconfiguring ability than completely drop them.

MichaIng, 2018-01-21:
@k-plan
About none SSH boot: https://github.com/Fourdee/DietPi/issues/1376
This is already since longer. Timestamps and rc.local added there automatically
We could at least make all our notifications (info/ok/failure) overwrite existing lines from the start. But this could in case overwrite interesting/important output also.

k-plan, 2018-01-21:
@MichaIng 

> About none SSH boot:

Ah ... sorry, my fault. I imagine you have meanwhile fixed it.
Leave it as it is, it's only optical imperfect.

And sorry, VMs are not on my real focus. 
Only notice, it is different then before.

[image]

k-plan, 2018-01-21:
@MichaIng 

> Yeah I agree, that this is a nice feature loss for ARMbian based devices.

👍  If so, I can directly stay with ARMbian. They have as well copy some nice feature from DietPi like `armbian_first_run.txt` and [`armbian-config`](https://docs.armbian.com/User-Guide_Armbian-Config/)  😳 

> There are also some free ext4 drivers for Windows, but they are not working to reliable. Tried it once and ended with a knoppix live USB stick, respectively a Linux VM.

Try more then once and ended up with [Paragon ExtFS for Windows](https://www.paragon-software.com/de/home/extfs-windows-pro/download.html) and NTFS/ExtFS for Mac.

> 

> we can maybe find/offer an easy reliable way for users to get ext4 access.

You need the hardware for donig this and change it every time you create a new/update image.
Not so easy and much work. And some dev on armbian dislike DietPi and and the way we are doning things, so ... 🙊  🙈 

Yes, for me, not need to enter all file in `/boot`. They can stay their on ext4 partition. Had as well some advantage. But a accessible FAT with `dietpi.txt` is responsibility.

> But this needs direct terminal access ...

Puhh, some devices have no HDMI output and headless is cool. Particularly with ssh, if you can set your needed network configuration before fist boot.
Dealing with UART/Serial is not a solution, because of additional hardware, most userer didn't have.

> But I would better stay with offering support for this devices without preconfiguring ability than completely drop them

Good, but not good for support. I hear the exclamation from now on, particular while discontinue v159 updates at the same time ...

Lieber ein Ende mit Schrecken als ein Schrecken ohne Ende.
Ist das so richtig übersetzt?

> It's better to make a painful break than draw out the agony.

Aber ich würde mich sehr für die Nutzer freuen, wenn wir eine gangbare Lösung finden könnten.

MichaIng, 2018-01-21:
@k-plan 
Actually in that cases, creating a separate _tiny_ fat partition just for dietpi.txt would be maybe the simplest solution. This could be even removed on first run, after copying the file to /DietPi. The latter would be anyway necessary if we do not want to rewrite a bunch of scripts to look on different locations for dietpi.txt.
@Fourdee 
Do you think that would be possible? I am not too deep in in early boot scripts, in case including fs changes.

k-plan, 2018-01-21:
@MichaIng 

great Idea! 👍  ❤️ 

Würde auch noch einige andere Probleme direkt mit erschlagen.
Einziger Nachteil den ich so direkt sehe, wenn man sich vertippt, darf man direkt noch einmal von vorne anfangen. Aber das übt und ist auf alle Fälle ein Fortschritt zum momentanen Zustand.

Fourdee, 2018-01-22:
@MichaIng 

> Do you think that would be possible? I am not too deep in in early boot scripts, in case including fs changes.

We cant place the extra partition after the rootFS partition, would prevent expansion of rootFS. 

We could try shifting the rootFS partition to P2, then create a 64MB FAT16/32 P1 at start after 4MiB for ARMbian. We'd need to check all the kernel command line options in ```/boot```, ensuring UUID is used etc.
- Move all boot files to P1
- Then edit ```/etc/fstab``` with ```/boot``` mount

Lengthy/tedious process if applied to all ARMbian images, and, that is assuming its successful.

Fourdee, 2018-01-22:
@k-plan 

> Try more then once and ended up with Paragon ExtFS for Windows and NTFS/ExtFS for Mac.

If these options work (i've downloaded Paragon to test)? We could provide this information in 1st run guide for ARMbian images.

MichaIng, 2018-01-22:
@Fourdee
And if we, based on HW_MODEL/2nd partition size/content(dietpi.txt), just remove the tiny 2nd partition before first run resizing rootfs? A tiny free space would not hurt. I am afraid of generally changing all the partitions for other devices, if I understand your idea right, as this was/is always a bid fragile.

Fourdee, 2018-01-22:
@MichaIng 

>  just remove the tiny 2nd partition before first run resizing rootfs?

Could work 👍 
- prior to rootFS resize
- mount p1/p2
- move ```dietpi.txt``` to ```/boot```
- unmount p1/p2
- del p2
- rootfs resize

Fourdee, 2018-01-22:
@zador-blood-stained @ThomasKaiser @ igorpecovnik

**Question:**

Are the following options, fully supported in ARMbian build tools?
- ```BOOTFS_TYPE```
- ````BOOTSIZE```

https://github.com/armbian/build/blob/e546ac684d0fe0c0f47ab032aa867062e37da4e6/lib/configuration.sh#L224-L225

Reason for query:
- No mention of ````BOOTSIZE``` in the online doc
- Its been mentioned in the past as a depreciated feature: (https://github.com/armbian/build/commit/e3c0988d3e793bdf0218cd24394302a84eb28ee6#diff-0a30c16b7ceda04b74ad5a1dc0a9e2a3R57) (https://github.com/armbian/build/issues/660#issuecomment-297428579)

zador-blood-stained, 2018-01-22:
> Are the following options, fully supported in ARMbian build tools?
BOOTFS_TYPE
BOOTSIZE

No, and by that answer I mean that it *may* work (or rather *should* work) in most cases, but

- FAT /boot partition is not tested in general so it may be broken on some platforms
- FAT /boot partition support was not added to some newer platforms so it may not be supported on all boards
- there are known issues with the `nand-sata-install` script on multi-partition images

so it is a semi-supported build script option that may not work since currently we favor single-partition images (except for experimental OMV builds).

> Its been mentioned in the past as a depreciated feature

FAT boot partition support in general and using `BOOTSIZE` to switch to a FAT partition is deprecated, but except for mentioned bugs and limitations building 2-partition images is supported.

MichaIng, 2018-01-22:
@Fourdee @zador-blood-stained
FAT partition support should be no problem,  as DietPi by default adds dosfstools package. So we should be able to create the tiny FAT partition after full distro/package upgrade was done during preparation script, and then it should be able to mount during first run. All kernel/boot files can/should stay on 1st root partition to not mess something up there.

zador-blood-stained, 2018-01-22:
> FAT partition support should be no problem, as DietPi by default adds dosfstools package.

True unless you are talking about the first partition on the system device mounted to /boot.

This was discussed already, but there are several problems:

- dpkg breaks when updating files on FAT filesystem so without special workarounds the system will break on the kernel upgrade

- you can't make symlinks on FAT so again the system will break after the kernel upgrade without workarounds for `ln -s` call in the kernel upgrade logic

- u-boot boot script loading sequence and the boot script itself should be able to work for loading files from FAT as older u-boot has different commands - `fatload` and `ext2load`/`ext4load`

Fourdee, 2018-01-24:
Native PC BIOS, 1st attempt failed. invalid filesystem after image ready (faulty SSD 😢 ).

k-plan, 2018-01-24:
@Fourdee 

> If these options work (i've downloaded Paragon to test)? We could provide this information in 1st run guide for ARMbian images.

Quick check on Windows 7 running native on hardware.

:100:  fail

- Paragon can mount, read and write ext4 partitions on SD card, but to unmount you need administrator rights! :-1: 

- AND Paragon` extfsservice.exe` block access for image writing software, eg for Rufus, to format and write SD cards, if Paragon is installed. No chance to stop this services without administrator rights and taskmanager!  :-1: 

 Forget it - uninstalled -  finish

Fourdee, 2018-01-24:
@k-plan 

@MichaIng method (extra 1MB fat partition) is the solution here, however, if i'am being honest, time limited at the moment. It may be after v6.0 is released that we can achieve it.

notes:
```
if [ partitionlabel = 'dietpi_extra' ]; then

del part

fi

expand
```

Fourdee, 2018-01-25:
Sparky Allo install, hangs system. Hardware issue:
```
[ 1076.024195] read phy reg-5
[ 1200.875145] INFO: task kworker/1:2:102 blocked for more than 120 seconds.
[ 1200.875155] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[ 1200.875162] kworker/1:2     D c07bfe48     0   102      2 0x00000000
[ 1200.875187] Workqueue: events_power_efficient close_delayed_work
[ 1200.875213] [<c07bfe48>] (__schedule+0x390/0x7cc) from [<c07c065c>] (schedule_preempt_disabled+0x14/0x20)
[ 1200.884908] [<c07c065c>] (schedule_preempt_disabled+0x14/0x20) from [<c07bef44>] (__mutex_lock_slowpath+0x174/0x368)
[ 1200.895631] [<c07bef44>] (__mutex_lock_slowpath+0x174/0x368) from [<c07bf144>] (mutex_lock+0xc/0x24)
[ 1200.904913] [<c07bf144>] (mutex_lock+0xc/0x24) from [<c056ebdc>] (close_delayed_work+0x20/0xc8)
[ 1200.913756] [<c056ebdc>] (close_delayed_work+0x20/0xc8) from [<c00470dc>] (process_one_work+0x134/0x414)
[ 1200.923380] [<c00470dc>] (process_one_work+0x134/0x414) from [<c00477b8>] (worker_thread+0x138/0x37c)
[ 1200.932730] [<c00477b8>] (worker_thread+0x138/0x37c) from [<c004d394>] (kthread+0xa4/0xb0)
[ 1200.941113] [<c004d394>] (kthread+0xa4/0xb0) from [<c000f3d8>] (ret_from_fork+0x14/0x3c)
[ 1210.845796] read phy reg-4
[ 1210.846145] read phy reg-0
[ 1210.846487] read phy reg-5
```

Fourdee, 2018-01-25:
AmiBerry image test did not install, automation issue?

Fourdee, 2018-01-26:
Allo RPi image test:
- Hangs 1.3 during boot after smsc
- bad image read?
- fs-resize issue?

[4 more replies]

closed 2018-01-28 by Fourdee