MichaIng/DietPi #1387: After short idle sound from Kodi stops working

xDavo, 2018-01-11:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version: 159
- SBC Device: Odroid - XU4
- Power supply used: Vigan, 5V, 3.5A
- SD card used: Sandisk Ultra
- Distro: Jessie

#### Additional Information (if applicable):
A while after Kodi releases audio channel (according to Kodi
settings in System -> Settings -> System -> Audio output -> Keep
audio device alive) Kodi is unable to connect to PulseAudio again 

- Software title: Kodi
- Can this issue be replicated on a fresh installation of DietPi: YES

Issue reproduced few times using automated installation:
AUTO_DietpiSoftware_Install_ID=31
AUTO_DietpiSoftware_SSHServerIndex=-2
AUTO_AutoStartTarget=1

- tested with both audio over HDMI and audio over USB (using Axagon ADA-15 USB - HQ MINI AUDIO)

#### Expected behaviour:
Sound should work at all times in Kodi

#### Actual behaviour:
Sound doesn't work at all after a short while in Kodi

#### Steps to reproduce:
Preconditions: 
- Install via Automated installation: 
AUTO_DietpiSoftware_Install_ID=31
AUTO_DietpiSoftware_SSHServerIndex=-2
AUTO_AutoStartTarget=1

- after first boot run dietpi-config and change default soundcard
in 	Audio options to usb-dac

Step 1: Boot up the device and wait until Kodi appears
Step 2: Use a source of sound (eg. install Youtube addon)
Step 3: Play a random source of sound (eg. youtube video)
Result: Sound works correctly
Step 4: Stop video and let Kodi idle for a while (2-5 minutes)
Step 5: play the source of sound again

Result: FAILED - sound doesn't work anymore

#### Did you submit a ```dietpi-bugreport```?
Reference CODE: c67ca81e-3042-456c-af0a-127222a9df27-0

#### Extra details:
kodi log:

11:22:25 T:3007005616   ERROR: PulseAudio: Failed to connect context
11:22:25 T:3007005616  NOTICE: PulseAudio might not be running. Context was not created.

#### Other tested scenarios:
1. Sound over HDMI
Result: FAILED - result doesn't differ from original scenario

2. speaker test
Preconditions: reboot device and idle until audio stops working

Step 1: run speaker-test -c2

Result: PASSED - white noise sound correctly alternates between left
and right channel.

3. alternative USB port (with audio over USB)
Result: FAILED - result doesn't differ from original scenario

4. running PulseAudio manually from command line
Preconditions: reboot device and idle until audio stops working
Step 1: connect via SSH and run pulseaudio
Result: Sound works again until Kodi releases audio, then pulseaudio process dies
pulse log:

( 113.726|   0.001) I: [pulseaudio] main.c: Daemon terminated.

5. Kodi settings Keep audio device alive to Always
Result: PASSED - Kodi never releases audio channel and sound works for tested period
of 1 hour

6. Other source of audio - raspotify
Preconditions: reboot device and idle until audio stops working
Step 1: play music over spotify
Result: PASSED - spotify is playing correctly

xDavo, 2018-01-11:
Run another scenario: 
##### 7. Autostart to LXDE and run Kodi manually
Preconditions: 
- install LXDE
- set autostart to desktop

Step 1: Run Kodi from LXDE desktop
Step 2: Use a source of sound to test audio then let Kodi idle for a while (30 minut tested)
Step 3: Use a source of sound again

Result: PASSED - Sound works even after prolonged period of idle. 

I'm not a developer but maybe there is a connection to this: https://github.com/mk-fg/pulseaudio-mixer-cli/issues/17

MichaIng, 2018-01-28:
@xDavo 
Seems (our) Kodi does not handle energy savings of PulseAudio correctly. Could you try this: https://github.com/Thinstation/thinstation/issues/33#issuecomment-35364115
Just to reduce the issue source to PulseAudio idle timer.

Fourdee, 2018-02-02:
I'll mark as closed, no response/outdated. If the issue still exists, please reopen.

closed 2018-02-02 by Fourdee