MichaIng/DietPi #1388: Public Key issue?

pedrom34, 2018-01-12:
## Creating a bug report/issue:

#### Required Information:
- DietPi Version | 159  
- SBC Device | RPi 2 Model B (armv71)  
- Power supply used : 5v 2.5A  
- SD card used : Sandisk ultra 32Gb  
- Distro (EG: Jessie) | Linux DietPi 4.9.62-v7+ #2 SMP Fri Nov 17 23:52:26 GMT 2017 armv7l GNU/Linux  

#### Additional Information (if applicable):
Fresh install. Some dietpi software installed.  
[COPS](https://github.com/seblucas/cops) installed according to the official HowTo, and working.
On the use of apt-get update  

#### Expected behaviour:
update

#### Actual behaviour:
W: GPG error: http://download.mono-project.com/repo/debian stretch InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY A6A19B38D3D831EF  
W: The repository 'http://download.mono-project.com/repo/debian stretch InRelease' is not signed.  
N: Data from such a repository can't be authenticated and is therefore potentially dangerous to use.  
N: See apt-secure(8) manpage for repository creation and user configuration details.  

#### Steps to reproduce:
just apt-get update

Fourdee, 2018-01-12:
@pedrom34 

Thanks for the report 👍 

We'll check the code, however, i'll flag this is low priority due to it being a warning, which will not effect the operation of repo.

pedrom34, 2018-01-12:
Thanks. :relaxed:

Fourdee, 2018-01-12:
@pedrom34 

Lets check mono is installed via ```dietpi-software```, please paste results:
```
dietpi-software list | grep mono
```

If it reports ```=2```, try re-adding the key:
```
apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys 3FA7E0328081BFF6A14DA29AA6A19B38D3D831EF
apt-get update
```

pedrom34, 2018-01-12:
`id 150 | =2 | mono: runtime libraries and repo |`

So I did:  
`apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys 3FA7E0328081BFF6A14DA29AA6A19B38D3D831EF`  

and got:  
`Executing: /tmp/apt-key-gpghome.xr14d7xmGq/gpg.1.sh --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys 3FA7E0328081BFF6A14DA29AA6A19B38D3D831EF
gpg: failed to start the dirmngr '/usr/bin/dirmngr': No such file or directory
gpg: connecting dirmngr at '/run/user/0/gnupg/d.t8z1iyjmdadjsz5893iwrw5n/S.dirmngr' failed: No such file or directory
gpg: keyserver receive failed: No dirmngr`

MichaIng, 2018-01-19:
@pedrom34 
- First of all there seems to be a special Raspbian repo: https://download.mono-project.com/repo/debian/dists/raspbianstretch/
- And it is available via https, as you can see.
- I would suggest to add it to souces as: `deb https://download.mono-project.com/repo/debian raspbianstretch main`
- Just found for verification: http://www.mono-project.com/download/#download-lin-raspbian
- Just to sort out: The package `gnupg` are installed right? And as I see `dirmngr` is asked for, which is recommendation (so by default installed together with) of `gnupg`, so please also verify that it's installed.
- To access the repo via https, also `apt-transport-https` and `ca-certificates` are needed.

pedrom34, 2018-01-20:
Hi @Michalng

gnupg was installed, dirmngr was not. With:  

`apt-get install dirmngr`
`apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys 3FA7E0328081BFF6A14DA29AA6A19B38D3D831EF`
`echo "deb http://download.mono-project.com/repo/debian raspbianstretch main" | sudo tee /etc/apt/sources.list.d/mono-official.list`
`apt-get update`

problem is solved. But why dirmngr wasn't installed?

Fourdee, 2018-01-20:
Notes:
- https://packages.debian.org/stretch/dirmngr

Tried this on XU4 Jessie, seems fine without ```dirmngr```:
```
root@DietPi:~# dpkg -l | grep dirmngr
root@DietPi:~#

root@DietPi:~# apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys 3FA7E0328081BFF6A14DA29AA6A19B38D3D831EF
Executing: gpg --ignore-time-conflict --no-options --no-default-keyring --homedir /tmp/tmp.kPhYCtoJvW --no-auto-check-trustdb --trust-model always --keyring /etc/apt/trusted.gpg --primary-keyring /etc/apt/trusted.gpg --keyring /etc/apt/trusted.gpg.d/debian-archive-jessie-automatic.gpg --keyring /etc/apt/trusted.gpg.d/debian-archive-jessie-security-automatic.gpg --keyring /etc/apt/trusted.gpg.d/debian-archive-jessie-stable.gpg --keyring /etc/apt/trusted.gpg.d/debian-archive-stretch-automatic.gpg --keyring /etc/apt/trusted.gpg.d/debian-archive-stretch-security-automatic.gpg --keyring /etc/apt/trusted.gpg.d/debian-archive-stretch-stable.gpg --keyring /etc/apt/trusted.gpg.d/debian-archive-wheezy-automatic.gpg --keyring /etc/apt/trusted.gpg.d/debian-archive-wheezy-stable.gpg --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys 3FA7E0328081BFF6A14DA29AA6A19B38D3D831EF
gpg: requesting key D3D831EF from hkp server keyserver.ubuntu.com
gpg: key D3D831EF: public key "Xamarin Public Jenkins (auto-signing) <releng@xamarin.com>" imported
gpg: Total number processed: 1
gpg:               imported: 1  (RSA: 1)
```

Fourdee, 2018-02-28:
General | Resolved multiple issues with failed GNU key management during APT installs from non-standard repos. **Dirmngr is now installed on all DietPi systems by default**: https://github.com/Fourdee/DietPi/issues/1388

Marking as completed.

closed 2018-02-28 by Fourdee

clarky0000, 2018-05-01:
Im having this issue while trying to install Sonarr. I'm running Dietpi version 159 also. I see you say 
 Dirmngr is now installed on all DietPi systems by default
Can I update my existing version? If so how? Or do I have to rebuild from a new image?
Thanks in advance
clarky

pedrom34, 2018-05-01:
You have go from scratch to switch from v159 to v6.X.

Fourdee, 2018-05-01:
@clarky0000 

Please see the v6.x FAQ for more information:
https://github.com/Fourdee/DietPi/issues/1355