MichaIng/DietPi #1391: kodi issue:     "Package 'kodi-odroid:armhf' has no installation candidate"

pgferr, 2018-01-13:
## Creating a bug report/issue:
 i cannot install kodi. The installation don't go on and i have this message:Package 'kodi-odroid:armhf' has no installation candidate

#### Required Information:
- DietPi Version: 159
- SBC Device: odroid c2
- Power supply used (EG: 5v 1A RAVpower)?
- SD card used (EG: Sandisk ultra)?
- Distro: debian stretch | Linux DietPi 3.14.79+ #1 SMP PREEMPT Sun Jul 23 00:04:44 CEST 2017 aarch64 GNU/Linux

#### Additional Information (if applicable):
[image]
[image]

#### Expected behaviour:

#### Actual behaviour:

#### Steps to reproduce:

#### Did you submit a ```dietpi-bugreport```?

#### Extra details:

Fourdee, 2018-01-13:
@pgferr 

Thanks for the report 👍 

Kodi is currently not available for Debian Stretch on the C2:
https://github.com/Fourdee/DietPi/issues/1032#issue-238395824

You'll need to use the Jessie image for Kodi/GPU usage:
dietpi.com/download

Fourdee, 2018-02-02:
As we no longer offer Stretch images for Odroid, due to limitations with available Meveric repos, where the Kodi binaries are installed from . I'll mark this as closed.

closed 2018-02-02 by Fourdee