MichaIng/DietPi #1395: Update reporting older version as newer

hydrouk, 2018-01-15:
dietpi-update    = Run now to update DietPi (from V159 to V157).

I doubt this is normal and i run updates as instructed.

hydrouk, 2018-01-15:
I rebooted and it disappeared but still weird

k-plan, 2018-01-15:
Please have a look at:

https://github.com/Fourdee/DietPi/issues/1258

DasClarky, 2018-01-15:
Thanks k-plan! Didn't notice that was already an issue

k-plan, 2018-01-15:
To fix it:

```
rm /DietPi/dietpi/.update_available

rm /boot/dietpi/.update_available

dietpi-update 1

```

k-plan, 2018-01-15:
@Fourdee 

leave it open, so you can have a look, if additional fix is require:

https://github.com/Fourdee/DietPi/commit/78ca55697fbba380fcb78fa4c650ce3eee455381#diff-c613a85da508fb885b67c34f9661e243R137

Please close, if it is fixed. Thanks.

Fourdee, 2018-01-16:
@k-plan 

Should be fixed for v159. Requires a script run of dietpi-update which auto runs during boot, or when user runs ```dietpi-update```.

We did have a possible issue with dietpi-ramdisk not saving to disk during shutdown, this is resolved for v6.0.

I'll mark this as closed for now, unable to replicate it in testing. We can reopen if v6.0 users report the same issue.

closed 2018-01-16 by Fourdee